romasku/tuya-zigbee-switch #3: Devices remain as "EndDevice" after flashing OTA

klu16, 2024-12-26:
Hey! As in the description, two TS0012 which were flashed OTA remained as "EndDevice".
Only the first one that I flashed via UART and soldering method, changed its type to Router.

Example via OTA:
[image]

via UART:
[image]

romasku, 2024-12-26:
Hi,
Can you please try to re-interview it by pressing this "i" button:
[image]

klu16, 2024-12-27:
> Hi, Can you please try to re-interview it by pressing this "i" button: ![Screenshot From 2024-12-27 01-17-41](https://private-user-images.githubusercontent.com/11975516/398813479-8664fd67-5947-4988-9dcf-a7f2817ab1e5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzUyODI2NzQsIm5iZiI6MTczNTI4MjM3NCwicGF0aCI6Ii8xMTk3NTUxNi8zOTg4MTM0NzktODY2NGZkNjctNTk0Ny00OTg4LTlkY2YtYTdmMjgxN2FiMWU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjI3VDA2NTI1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcxZDc3MmUxZDlhZGNjYzQxOTE1Njg3NDUzYzVmMzA4YTc0MmRjMThjMzA5MTE3ZThkMTY0Y2JmMzQxY2FmYWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NwZLOfdAxdHMGgmap0U7vIi8yMJGKA9HfG-9uJH16EQ)

It worked, now all are as Routers! Sorry, I don't know Z2M that much 😅
Thank you!!

closed 2024-12-27 by klu16