romasku/tuya-zigbee-switch #4: Add support for TS0011 (1-gang without neutral)

klu16, 2024-12-26:
Hi. I know that this firmware was primarily developed to fix simultaneous press of 2 switches to turn on both relays, BUT the response time makes a HUGE difference, e.g when you use motion sensors to turn on your home lights. With that 0,5 second delay you can already make 2-3 steps before the light turns on, so it can be a bit annoying sometimes... 

Therefore I would like to ask if you could consider making a support also for 1-gang switch 🙏

https://www.zigbee2mqtt.io/devices/TS0011_switch_module.html#tuya-ts0011_switch_module

Cheers! 🍻

romasku, 2024-12-26:
Unfortunately, I do not have this device. I may order it to adapt firmware, but it will take time to arrive.

klu16, 2024-12-27:
I have some. So if you need anything and I can help somehow, just let me know :) And don't worry, take your time! 🍺

romasku, 2025-01-18:
Hi, I added support for TS0011. Please check, and as always, please be cautious before updating, I do not want my bugs to disturb your home life too much:)

klu16, 2025-01-19:
Hi again :)

It does not matter what I do, no switches appear in the OTA tab in Z2M :( I added 3 files and restarted Z2M, as per the instruction.

This is my Z2M folder:

[image]

This is my configuration.yaml file:

[image]

klu16, 2025-01-20:
In the meantime, I upgraded Z2M to v2.0 which I've been planning for a while.

So, I created the external_converters folder and put all files there:

[image]

restarted Z2M, but when I check for updates, I got:

[image]

It looks like it wants to look in a different directory of /config/zigbee2mqtt/... for index_router.json
but in Z2M there is no /config.
Am I missing something here?

klu16, 2025-01-20:
Hmm... I flashed it via OTA, but the light started blinking every few seconds and now it seems bricked.

Could you upload full firmware for TS0011 to the flash via wire guide (there is only for TS0012 there)? I will attempt to recover it.

klu16, 2025-01-20:
I flashed both EndDevice and Router firmware versions and unfortunately, the lights are blinking every few seconds on both variants while off. When I turn the light on, it's OK.

It did not happen on stock of course.

romasku, 2025-01-20:
I'm sorry to hear that. I'll prepare OTA back to stock later. I guess that lights are blinking due to the device passing too much current, or do you see the same blinking inside the Z2M dashboard? Also, did you install the capacitor that comes with the device?

klu16, 2025-01-20:
> I'm sorry to hear that. I'll prepare OTA back to stock later. I guess that lights are blinking due to the device passing too much current, or do you see the same blinking inside the Z2M dashboard? Also, did you install the capacitor that comes with the device?

No problem :) 

I'm not sure if it's relevant, but I have a lamp with 2 bulbs and randomly 2 of them blink one by one very fast.

So, it's 1 bulb blinking and after 1-4 blinks, 2 of them blink fast. I can record video if that would help.

I did not install the capacitor, because it worked fine without it on the stock firmware.

klu16, 2025-01-20:
I have also just noticed that the state is not reported. So when I press the switch on the wall while on, it does not change to off in the app.

klu16, 2025-01-20:
I re-flashed it with EndDevice fw and blinking is still the issue, but it does not blink with 2 bulbs now. I assume this is due to the fact that Router consumes more energy or this could indicate a moment while passing a Z2M message to some other end device in the network (as this happens randomly).

PS. On the EndDevice fw, state reporting also does not work.

rumil, 2025-01-20:
Interestingly I have 2 TS0011 flashed (as routers, OXT brand) and they work
just fine.

klu16, 2025-01-20:
Hi again!

I tested the theory. Thankfully I had a little more powerful bulb (by not much, just 1,5W stronger LED bulb), but it seems it made a difference. So, when you put a new, stronger one (8W) in a pair with this "old" (6,5W) bulb, only the old one blinks, but it happens less often.
If I leave only the new (8W) bulb, the light is constant, but is very weak. Too bad I do not have a 2nd one 8W... It seems that old and new bulb behave differently, maybe because they are different brands and maybe LEDs used are different (constant light vs. blinking).

But I wonder, why it worked on the stock firmware with 2x 6,5W bulbs without any issue (did not blink)...?

klu16, 2025-01-20:
> Well, if you mean an old firmware like Enddevice, it obviously consumes less power. I had flashing behavior (random or cyclic) with the Router firmware as well. After the change I don't see any problems. Although the overall speed of operation decreased, but the problem with loss of one state when two buttons are pressed at a time - gone.

I have the same behaviour either on EndDevice (but less frequently blinks as consumes less power as you said) or the Router.

I will try to use the capacitor later.

klu16, 2025-01-20:
> I'm sorry to hear that. I'll prepare OTA back to stock later.

Hey @romasku. Is there any chance to provide stock fw for TS0011 in the near future? :) I'd like to go back to original for now, until I figure out how to eliminate the blinking issue. Thx!

romasku, 2025-01-21:
Hi, sorry for the delay, I'll have some time this evening. I'll try to prepare the stock firmware as OTA, but I'm unsure if that is possible. If it is not possible, I'll post a guide on how to flash using the UART.

klu16, 2025-01-21:
> Hi, sorry for the delay, I'll have some time this evening. I'll try to prepare the stock firmware as OTA, but I'm unsure if that is possible. If it is not possible, I'll post a guide on how to flash using the UART.

Thank you!

romasku, 2025-01-21:
Hello again,  

I’ve spent some time creating a flashable OTA file from the factory firmware dump, but, unfortunately, it doesn’t work for me—the device goes into a boot loop. As a result, I can only suggest flashing it via wire.  

I’ve updated [the guide](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/flashing_via_wire.md), which now includes a photo for the TS0011. However, soldering is a bit more challenging since the SWS pin is blocked by capacitors.  
To flash the stock TS0011 firmware, use [this file](https://github.com/romasku/tuya-zigbee-switch/blob/main/bin/ts0011_factory.bin).  

I hope this helps.  

I’ve also done some minimal research into power usage. From what I’ve observed:  
- The Router firmware uses ~12mA.  
- The End Device firmware uses 6–7mA.  
- The stock firmware operates in a constant loop of sleeping for 350ms (with almost zero power usage) and then short bursts of ~12mA for a couple of milliseconds.  

This explains both the low power usage of the stock firmware (as the device spends most of its time sleeping) and the delay (since the device cannot receive anything during sleep, it will process commands during the next wake-up cycle).  

I’ll probably attempt to replicate the sleeping behavior, but it will result in the same reaction delay as the stock firmware.  

If you need a shorter reaction time, I recommend looking for LED bulbs compatible with the Router firmware. If that’s not an option, sticking with the stock firmware might be the better choice, as my firmware won’t be significantly faster.

klu16, 2025-01-21:
Thanks for for the instruction, I will give it a try. 

> If you need a shorter reaction time, I recommend looking for LED bulbs compatible with the Router firmware. If that’s not an option, sticking with the stock firmware might be the better choice, as my firmware won’t be significantly faster.

Could you elaborate a bit more? What bulbs could be compatible with Router fw? Do you mean high wattage bulbs? Or not LED bulbs?

romasku, 2025-01-21:
> Could you elaborate a bit more? What bulbs could be compatible with Router fw? Do you mean high wattage LED bulbs? Or not LED bulbs at all?

I can’t guarantee whether a particular bulb will work, but yes, higher-wattage bulbs should generally perform better, and incandescent (non-LED) bulbs are almost certain to work. Adding the resistor that comes with the device can also help. Having more lamps is better, so it’s a good idea to keep all sockets occupied. Additionally, compatibility may depend on the manufacturer. However, it’s possible that nothing will resolve the issue, unfortunately.

I did a small test (I use router firmware) and in my case single 8W bulb is not enough (it blinks every 2-3s seconds), but two bulbs 8W and 6W already work OK. But I use 5x 6-8W bulbs there so maybe two bulbs also have some issues, but they are not noticeable during a 1-minute test.

From what I understand, you have a TS0012 that works well with the router firmware. Since they can suffer from the same problem, the bulb configuration you’re using with them is "good enough".

klu16, 2025-01-22:
Thanks for the stock fw. I bought higher wattage bulbs just to check if that resolves the issue. We'll see tomorrow or on Friday. But just in case those new ones blink as well, I'd like to go back to the old fw. 👍

TS0012 work great for me, probably because I have stronger bulbs in other lamps (I'm almost certain that's the cause, because every lamp in my home has only 2 bulbs, so it has to be related to higher wattage - and it's the same for the TS0011, so 2 bulbs, but lower wattage). 

> I’ll probably attempt to replicate the sleeping behavior, but it will result in the same reaction delay as the stock firmware.

I'd say it's not worth all your great work to replicate shitty "stock firmware" behavior. Your firmware reacts instantly (at least on Router) and that's a HUGE benefit of it :) Otherwise I wouldn't personally to bother with all the fuss (buying new bulbs, changing FWs back and forth etc.), if there's no gain from it ;)

Thank you very much again for all the work! 💪

romasku, 2025-01-24:
Nice to hear! I'll close this issue now but will track progress on reducing power usage at https://github.com/romasku/tuya-zigbee-switch/issues/5.

closed 2025-01-24 by romasku