romasku/tuya-zigbee-switch #6: ts0001_switch_module

itmcompl, 2024-12-29:
Hi
Do you plan to create a similar firmware to ts0001_switch_module and ts0002? 
The main problem I can observe is the delayed response time to the activate.

romasku, 2024-12-29:
Hi,  
Unfortunately, I don’t have these devices at the moment. I also found online that the TS0002 (specifically the Lonsonho X702) uses the [TYZS3](https://developer.tuya.com/en/docs/iot/tuya-tyzs3-zigbee-module-datasheet?id=K96efadaercw6) module, which contains a completely different chip. As a result, this firmware will not work on it at all. However, it looks like the `ts0001_switch_module` should be compatible, so I might order it at some point to adapt the firmware.  

Additionally, both devices use a neutral connection, so I’d assume they are already Router-type devices. The main reason this firmware reduces response time is that it converts a device from an End Device to a Router. Therefore, I don’t expect a significant improvement for these devices.

mmalolepszy, 2024-12-30:
I have a bunch of TS0001's and TS0002's (and even 3's and 4's 😉), branded by OXT. 

[image]

They indeed have neutral connection and already act as routers. For me switching delay is acceptable, but I'd love them to have detached relay mode and long press support. I grabbed one TS0002 that was hanging around unused and opened it. 

[image]

It looks like it is [ZT2S](https://developer.tuya.com/en/docs/iot/zt2s-module-datasheet?id=Kas9gdtath9p0) module (the same series TS0012 has). I'm not sure if they have a chip compatible with your firmware, but if they do, I'd be happy to help with developing support for them.

romasku, 2024-12-30:
The ZT2S is compatible. I have the TS0001 at home, so it’s the next device I plan to work on. Unfortunately, I don’t have the TS0002.  

To port the firmware, I need the pinout details—specifically, which ZT2S pin is connected to what. For a 2-gang module, there are six pins:  
- On-board LED  
- On-board button  
- S1 contact for the button  
- S2 contact for the button  
- Power control for S1  
- Power control for S2  

If you can trace the connections using a multimeter, I’ll prepare a separate firmware version for you to try.

romasku, 2024-12-30:
Sorry for the confusion about me having the TS0001. I thought I didn’t have it because it’s branded differently.

mmalolepszy, 2024-12-30:
I've done some tracing with multimeter and it looks like the pinout is as follows:

- C2 - LED
- B4 - Switch S2
- B5 - Switch S1
- C4 - Relay L1
- C3 - Relay L2
- D2 - Button

Pin names are as in the ZT2S Tuya documentation.

romasku, 2024-12-30:
Great, thank you, that is what I need! I'll be busy the next few days, and after that, I'll prepare a firmware build for you to test.

romasku, 2025-01-02:
Also, what "Zigbee Manufacturer" do you see in z2m? Is it `_TZ3000_skueekg3`?

romasku, 2025-01-02:
Hi again,  

I’ve added support for the [WHD02](https://www.zigbee2mqtt.io/devices/WHD02.html) which, for some reason, is also named TS0001 in z2m library code. Additionally, I’ve implemented support for the [TS0002](https://www.zigbee2mqtt.io/devices/TS0002_basic.html) based on the provided pinout. Please test if it works. Be prepared to re-flash via wire, as it’s very risky—I haven’t tested it.  

I also don’t have the [TS0001 switch module](https://www.zigbee2mqtt.io/devices/TS0001_switch_module.html). If you can provide the pinout, I’ll adapt the firmware for it.

mmalolepszy, 2025-01-06:
Hi, 
Sorry for the late reply. 

I've just flashed my TS0002 via OTA. It had some issues while joining the network but a few resets later it joined and got interviewed. 
[image]
At a first glance it seems to be working fine. I can hear relays clicking when I turn them on/off via Z2M. The on-board LED of the module is constantly on (not sure if that was intended). I will do some more testing later this week.

As for TS0001, all my modules are reported as `_TZ3000_tqlv4ug4`. I don't have currently one available to open, because all are installed and in use. I will try to replace one with the freshly flashed TS0002 over the weekend and will come back with the pinout.

romasku, 2025-01-06:
That's nice to hear. If possible, please test your TS0002 a bit after you install it. If all is good, I'll add it to the list of supported devices.

> The on-board LED of the module is constantly on

This is intended, the idea is to help to find underpower issues on no-neutral devices.

grzegorzlabudda, 2025-01-10:
after updating ts0002 only one relay works, it does not react either to z2m or to the physical switch

romasku, 2025-01-10:
Hi, I'm sorry to hear. As I do not have TS0002 at hand, I can't help you much, unfortunately. It sounds like the wrong pin is being used for the relay.  
Hmm, does the state in z2m update after physical press?

grzegorzlabudda, 2025-01-10:
yes, after switching the physical switch, the state in zie changes in z2m, only the first relay does not react, when I change the relay settings to the button, I can only control relay 2 with the second button

mmalolepszy, 2025-01-10:
Hi @grzegorzlabudda, that's weird because my TS0002 works fine with new firmware. Both relays can be switched with the corresponding physical buttons and via Z2M. What exact model is it? What was the "Zigbee Manufacturer" reported in Z2M before updating? Do you have both "Switch relay index" set properly?

grzegorzlabudda, 2025-01-10:
unfortunately I didn't write down the exact switch model. these are oxt switches. yes the indexes are correct

mmalolepszy, 2025-01-10:
Mine is OXT as well (you can see pictures earlier in this thread). Can you open yours and check if the pinout is the same as I posted here?

grzegorzlabudda, 2025-01-10:
unfortunately I can't remove the relay today. I left it as is for now. I won't have time to take it apart until Sunday or Monday at the earliest.

grzegorzlabudda, 2025-01-12:
my switch looks like this, there is a zt3l system there, for now I'm trying to figure out which pin is for the relays
[image]
[image]
[image]
[image]

grzegorzlabudda, 2025-01-12:
> ZT2S jest kompatybilny. Mam TS0001 w domu, więc to jest następne urządzenie, nad którym planuję pracować. Niestety, nie mam TS0002.
> 
> Aby przenieść oprogramowanie sprzętowe, potrzebuję szczegółów pinoutu — konkretnie, który pin ZT2S jest podłączony do czego. W przypadku modułu 2-gangowego jest sześć pinów:
> 
> * Dioda LED na pokładzie
> * Przycisk pokładowy
> * Styk S1 dla przycisku
> * Styk S2 dla przycisku
> * Sterowanie mocą dla
> Jeśli potrafisz prześledzić połączenia za pomocą multimetru, przygotuję dla Ciebie osobną wersję oprogramowania sprzętowego do wypróbowania.

> The ZT2S is compatible. I have the TS0001 at home, so it’s the next device I plan to work on. Unfortunately, I don’t have the TS0002.
>
> To port the firmware, I need the pinout details—specifically, [...]

could I ask for firmware to handle zr3l. I know that S1 and S2 are correct as well as relay2. I can't find the pin for relay1

mmalolepszy, 2025-01-12:
Hi, I finally had some time today to open one of my TS0001's. Surprisingly it looks pretty much the same as TS0002 from @grzegorzlabudda, it just have some elements missing. Take a look at the photos:
[image]
[image]
If I'm not mistaken, pinout for my module is:

- C0 - LED
- B4 - Switch
- C2 - Relay
- D2 - Button

And if I check spots for missing elements that seems to be responsible for the other pair of switch and relay, then they are connected to the module as follows:

- B5 - Switch 2
- C3 - Relay 2

romasku, 2025-01-12:
@mmalolepszy Thank you very much! I'll adapt the firmware in the next couple of days. 

@grzegorzlabudda By looking at your photos and what you said I agree with @mmalolepszy theory about the pin-out of your device, so I hope the new firmware will also work for you.

grzegorzlabudda, 2025-03-10:
after the update the module died completely. uploading via cables did not help. the diode went out and it was not possible to control it either via ha or physical switches. I am waiting for a new module the other one went to the trash.

romasku, 2025-03-10:
Ouch, I'm sorry to hear that. I think firmware for the wrong device somehow got OTA flashed on it, and it bricked, but unfortunately, I do not see a bug inside the OTA file that may have caused it.

grzegorzlabudda, 2025-03-10:
> Ouch, I'm sorry to hear that. I think firmware for the wrong device somehow got OTA flashed on it, and it bricked, but unfortunately, I do not see a bug inside the OTA file that may have caused it.

nothing happened. there were problems with this module from the beginning. a new one is coming to me, probably the same one. we'll see when I unpack it and take a picture. of course I'll also upload your firmware 😉

closed 2025-11-16 by romasku