romasku/tuya-zigbee-switch #7: Not working with zigbee2mqtt 2.0

grzegorzlabudda, 2025-01-05:
after updating to 2.0 the OTA custom component update does not work

romasku, 2025-01-05:
Hi,
I've updated [doc](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/ota_flash.md) for changes in zigbee2mqtt 2.0. If you already have an old setup, please create `external_components` subfolder, and upload updated files [converter for the original device](https://github.com/romasku/tuya-zigbee-switch/raw/refs/heads/main/zigbee2mqtt/converters/tuya_with_ota.js)  and [converter for custom firmware](https://github.com/romasku/tuya-zigbee-switch/raw/refs/heads/main/zigbee2mqtt/converters/switch_custom.js) into it.

Hope this helps. Thank you for using the firmware!

grzegorzlabudda, 2025-01-05:
[image]

after adding these files to the new external_components directory the OTA update still does not work. the device works normally.

rumil, 2025-01-05:
Moreover, I'm still on Z2M 1.4, used instructions for 1.x and I get this error when checking for OTA:

```
Info 2025-01-05 17:06:20z2m: Checking if update available for '0xa4c138956cd617e0'
Info 2025-01-05 17:06:20z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"0xa4c138956cd617e0"},"error":"Failed to check if update available for '0xa4c138956cd617e0' (device.definition.ota.isUpdateAvailable is not a function)","status":"error","transaction":"axz96-1"}'
Error 2025-01-05 17:06:20z2m: Failed to check if update available for '0xa4c138956cd617e0' (device.definition.ota.isUpdateAvailable is not a function)
```

romasku, 2025-01-05:
Sorry, I missed updating some files. 
For z2m v1, please use files from https://github.com/romasku/tuya-zigbee-switch/tree/main/zigbee2mqtt/converters_v1 (I've updated links inside doc). 
For z2m v2, I missed updating one of the converters files, sorry.

Please re-download and check one more time, I'm sorry for that.

rumil, 2025-01-05:
Perfect, flashing on 1.4 worked now. Thanks for fixing quickly!

romasku, 2025-01-06:
Hi, 

> the states in ha are not updated 

Try pressing re-configure button in z2m: 
[image]

Then verify that binding and reporting look like this:

[image]
[image]
 

> the behavior after a power outage does not work

Hmm, my only guess is that maybe you are using an older version of the firmware, as in this case z2m will show the setting, but the device will ignore it. What do you see as "Firmware version"? Also, are there any errors in logs when you try to set it? Here is the latest version:

[image]

**And about OTA issue:**

Also, to confirm I understand your situation fully:
- You are using z2m 2.0
- You've re-downloaded [the latest files from here](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/ota_flash.md) after I fixed them one more time.
- You've restarted Z2M

But OTA is still not displayed in the OTA tab?

grzegorzlabudda, 2025-01-07:
yes after changing files from laptop (via phone it didn't work.) devices are visible in ota tab with unavailable update. I configured device to new but still the same. software version highest available as in screenshot. reporting and binding set the same as in your screenshot
[image]

rumil, 2025-01-07:
@grzegorzlabudda I had that happen to me once (state not being updated). What  _I think_ helped was:
- In z2m go to device settings -> Reporting
- Change “Min rep change” from 1 to 0
- Press Apply
- Change it back to 1 and Apply
From 3 switches I use currently only one showed that issue. I think it may also be related to signal strength. Even though it shows LQI of around 150 I do get occasional timeouts

closed 2025-01-12 by grzegorzlabudda