romasku/tuya-zigbee-switch #8: Error in HA log

rumil, 2025-01-07:
After installing firmware I have the following error showing in my HA logs: 

```
Logger: homeassistant.components.mqtt.number
Source: components/mqtt/number.py:184
integration: MQTT ([documentation](https://www.home-assistant.io/integrations/mqtt), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+mqtt%22))
First occurred: January 6, 2025 at 14:27:02 (10860 occurrences)
Last logged: 10:08:15

Invalid value for number.kitchen_light_switch_1_long_press_duration_1: 2000 (range 0.0 - 100.0)
Invalid value for number.living_room_light_switch_1_long_press_duration_1: 2000 (range 0.0 - 100.0)
Invalid value for number.living_room_light_switch_2_long_press_duration_2: 2000 (range 0.0 - 100.0)
Invalid value for number.bedroom_light_switch_2_long_press_duration_2: 2000 (range 0.0 - 100.0)
Invalid value for number.bedroom_light_switch_1_long_press_duration_1: 2000 (range 0.0 - 100.0)

```

Does not seem to cause any issues, all working fine for me.

mmalolepszy, 2025-01-07:
I guess that's a zigbee2mqtt converter issue. Haven't tested it, but changing `longPressDuration` to the following value and restarting z2m, should get rid of those logs from HA.
```
longPressDuration: (name, endpointName) =>
        numeric({
            name,
            endpointNames: [endpointName],
            cluster: "genOnOffSwitchCfg",
            attribute: { ID: 0xff03, type: 0x21 }, // uint16
            description: "What duration is considerd to be long press",
            valueMin: 0,
            valueMax: 10000,
        }),
```

romasku, 2025-01-07:
Yes, I also think this is the reason for these errors. I'll update the converters later.

romasku, 2025-01-07:
I pushed the fixed version of the converter, please re-download it.

rumil, 2025-01-07:
Tried updated z2m 1.4 converter and error log is not present anymore. Thanks!

romasku, 2025-01-08:
Good to hear! I'm closing the issue.

closed 2025-01-08 by romasku