romasku/tuya-zigbee-switch #9: TS0001 / TS0002 / TS0004 support (ZTU microchip)

paddenstoeltje, 2025-01-13:
Hi, I currently have some AVATTO rebranded Tuya switches at home:
- [Avatto 1 gang relay](https://www.zigbee2mqtt.io/devices/ZWSM16-1-Zigbee.html): rebranded TS0001  - reported manufacturer     _TZ3000_4rbqgcuv - ZTU microchip
- [Avatto 2 gang relay](https://www.zigbee2mqtt.io/devices/ZWSM16-2-Zigbee.html): rebranded TS0002  - reported manufacturer _TZ3000_mtnpt6ws - ZTU microchip
- [Avatto 4 gang relay](https://www.zigbee2mqtt.io/devices/ZWSM16-4-Zigbee.html): rebranded TS0004 - reported manufacturer  _TZ3000_5ajpkyq6 - ZTU microchip

Example of the 2 gang PCB:
[image]

Would I be able to use custom firmware as I'm experiencing some issues with the current firmware?

PS: I found the following [repo](https://github.com/Marplex/tuyaFree), which seems complementary to this repo and might be interesting?

romasku, 2025-01-17:
Hi, sorry for the delayed reply.
Yes, the firmware can be adapted for the ZTU module. I don’t currently have this device, and porting requires tracing the board to determine which GPIOs the firmware should use.

I’m currently working on a new version of the firmware that will allow pin configuration changes via the z2m interface, without the need for rebuilding and reflashing. This should greatly simplify the porting process. I’ll make sure to comment here when it’s ready.

The repository you posted contains a pinout for the 4CH version. I’ll add support for it here. Thank you for sharing it!

shatpav, 2025-01-29:
Hi, I have Avatto 1way relay LZWSM16-W1, no neutral. 

https://www.aliexpress.com/item/1005006809233813.html

[image]

ZHA was saying it is TS0011(MAC a4:c1:38:fa:02:84:ff:1a), before I flashed it using TS0011_END_DEVICE 1141-d3a3-ffffffff-tlc_switch-from_tuya.zigbee. After flashing device become a brick. I was able to take it back to "life" by flashing with wire using WHD02_factory.bin(no any other binary I have found worked) from this repository. But, it is not working as a relay, no physical button nor remote does not switch the relay, it is just exists in the network and generate events if I press physical button(on the module).

@paddenstoeltje If you be so kind to share factory binary from your device I would very appreciate it. :)

@romasku , I hope you will add support of ZTU chip devices soon. I'm flashed your TS0012 firmware on three other devices(not Avatto) and it really solved all the issues with on/off delays and simultaneous toggle two buttons. Thank you!

romasku, 2025-01-29:
Привіт:)

My guess is that it bricked because this model uses internal pull-up resistors, but the custom firmware for the TS0011 doesn't enable them. As a result, the device "sees" a lot of fake clicks and reboots. WHD02 uses internal pull-up resistors, which is probably why it worked.

There is no special support required for the ZTU; the only difference between modules (ZTU, ZT3L etc) is the pinout. Since I don't have this device, I can't provide much help. However, if you send detailed photos of the device, I can try to deduce the pinout, though it's quite difficult.  

Also, If you have time, try flashing the WHD02 custom firmware. If it joins the network, you'll be able to determine the pinout by adjusting the device configuration described [here](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/porting_to_new_device.md#setting-up-the-device-after-update). As a starting point, you can use:  

`Tuya-WHD02-custom;WHD02-custom;BB5u;LD3;`  

- Do not change the model/manufacturer for now, so that Z2M can display the correct UI.  
- If the on-board button generates switch events, it means that the `B5` pin is an internal button, so `BB5u` is the correct setting for it.  
- `LD3` is used to define an LED. The firmware may crash if this is not set.  
- Try adding different `R<pin>u>` and `S<pin>u`, using [pins present on ZTU module](https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp#title-5-Pin%20definition). Use `u` to enable pull-up.

shatpav, 2025-01-29:
Вітаю друже :)

Huge thanks! Solved. Unfortunately I don't use Z2M, ZHA only, so I was need to ping something(Switch and Button) with multimeter and for LED and RELAY build & try several times. This is the right pinout:
BB4u;LB5;SC0u;RC2;

Just in case if you consider to add this mod to the master branch, I don't know how to call this device. One more replica of Tuya TS0011. :) 

Дякую!

romasku, 2025-02-08:
Hello,

First, thank you very much for providing the pinouts! I've built additional versions of the firmware based on them.

Regarding 4-gang support—it's actually implemented in the code (I use an array for relays/Zigbee clusters, currently supporting up to four entries), but there was no converter for Z2M. However, since I’ve added your device, it should be there now. Please check.

As for inverting the LED, yes, I’ll add support to control it from the configuration. However, I won’t have much time this weekend, so it will have to wait. Sorry about that.

paddenstoeltje, 2025-02-08:
Hi,

thx for the quick support! I tried updating the 4 gang. The update is successful however when trying to reconfigure I get the following error:

```
info:z2m: Interviewing '0xa4c13877766f154e'

info 2025-02-08 18:17:24zh:controller:device: Device '0xa4c13877766f154e' is only compliant to revision '21' of the ZigBee specification (current revision: 23).

info 2025-02-08 18:17:25z2m: Successfully interviewed '0xa4c13877766f154e'

info 2025-02-08 18:17:25z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/interview', payload '{"data":{"id":"0xa4c13877766f154e"},"status":"ok","transaction":"qq4kz-32"}'

info 2025-02-08 18:17:40z2m: Configuring '0xa4c13877766f154e'

error 2025-02-08 18:17:41z2m: Failed to configure '0xa4c13877766f154e', attempt 9 (Error: ZCL command 0xa4c13877766f154e/5 genOnOff.configReport([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":60,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:349:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:954:26) at Endpoint.configureReporting (/app/node_modules/.pnpm/zigbee-herdsman@3.2.5/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:686:9) at Object.onOff (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/lib/reporting.ts:71:5) at configure (/app/data/extension/switch_custom.js:528:13) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.31.0/node_modules/src/index.ts:272:21) at Configure.configure (/app/lib/extension/configure.ts:124:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:49:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:204:17))

info 2025-02-08 18:17:41z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{},"error":"Failed to configure (ZCL command 0xa4c13877766f154e/5 genOnOff.configReport([{\"attribute\":\"onOff\",\"minimumReportInterval\":0,\"maximumReportInterval\":60,\"reportableChange\":1}], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"qq4kz-33"}'
```
I'm unable to control the 4 gang switch :(. Tried resetting already with no success.

romasku, 2025-02-08:
Can you please send screen "Clusters" tab of the device in Z2M?

paddenstoeltje, 2025-02-08:
Here you go.

[image]
[image]

paddenstoeltje, 2025-02-17:
Great, as the Avatto's are now completely supported I'll close the issue.

closed 2025-02-17 by paddenstoeltje

andrei-lazarov, 2025-06-20:
@paddenstoeltje does OTA flashing work on these AVATTO relays? Or did you flash by wire?

clumsy-stefan, 2025-07-15:
I received mine now, but have a number of issues I currently can't solve, even after rebinding, reconfiguring and interviewing:

- Only input 4 seems to work/react
- When sending onOff command to the relays, they react either not at all or only after I clicked on a second one. Then sometimes it works through all previous commands. Especially Relay 1 has this issue..
- by default input 4 switches relay 3
- It seems that relay output 1 is wrongly assigned.
- Physical S2 seems to be Switch 0 in the config

and various others...

As I don't want to change the config string, as long as it's not clear if it gets bricked I probably try the TS004_switch_module (istead of the _2)

andrei-lazarov, 2025-07-15:
@clumsy-stefan 

If you want to make some changes to the firmware and update OTA, here are the steps:
1. Fork the repo and create a new branch.
2. Edit just `device_db.yaml` as you wish, commit and push.
3. Run GitHub action build.yml on that branch. This generates the firmware, converters, OTA index etc.
4. Apply the generated index:
`https://raw.githubusercontent.com/USER/tuya-zigbee-switch/refs/heads/BRANCH/zigbee2mqtt/ota/index_router-FORCE.json`
and the generated converters:
`switch_custom.js` and `tuya_with_ota.js`.
(FORCE means it does not care about the version number)
5. Restart Z2M and update your device.

Also, very strange issues. 
Please take apart your device and use a multimeter to check the pinout.
`TS0004_switch_module_2` is correct as far as I can tell: [devices/tuya](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts)

clumsy-stefan, 2025-07-15:
Thanks, I'll give it a try, I have 6 units, hopefully I won't brick all of them 😂 ...

Will report any findings here..

andrei-lazarov, 2025-07-15:
@clumsy-stefan 
Please copy-paste the exact index entry you used and we can recover the device.

To receive an OTA update, the index must contain an entry with identical image_type and manufacturer_name to the device.
If it has stock firmware, it checks tuya_image_type and tuya_manufacturer_names.
If it has custom firmware, it checks firmware_image_type and the first word in the config_str.

clumsy-stefan, 2025-07-15:
Ah, ok, so I can probably fix/fake it in the index....

I added `"_TZ3000_5ajpkyq6"` to my local index file (which I copied from the official URL), so I guess it took one of these two entries:

```
  {
    "fileName": "1141-aa14-01033011-tlc_switch.zigbee",
    "fileVersion": 16986129,
    "fileSize": 164754,
    "url": "https://github.com/romasku/tuya-zigbee-switch/raw/2a5d7744325396167866ea5a86e36cc71edbb99e/bin/TS0004_AVATTO_NEUTRAL/1141-aa14-01033011-tlc_switch.zigbee",
    "imageType": 43540,
    "manufacturerCode": 4417,
    "sha512": "74a3558886a907fd060736c540b6b89e17283906bfb384b1c509dc7f7baf43be30e8eb535cb894fcf4d233358a7e62e9a0ecda97cc6c53eed530660be84e8b55",
    "otaHeaderString": "Telink Zigbee OTA\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "manufacturerName": [
      "_TZ3000_5ajpkyq6",
      "_TZ3000_ltt60asa",
      "_TZ3000_mmkbptmx",
      "TS0004-Avv"
    ]
  },
  {
    "fileName": "1141-d3a3-ffffffff-tlc_switch-from_tuya.zigbee",
    "fileVersion": 4294967295,
    "fileSize": 164754,
    "url": "https://github.com/romasku/tuya-zigbee-switch/raw/2a5d7744325396167866ea5a86e36cc71edbb99e/bin/TS0004_AVATTO_NEUTRAL/1141-d3a3-ffffffff-tlc_switch-from_tuya.zigbee",
    "imageType": 54179,
    "manufacturerCode": 4417,
    "sha512": "366b6485ee7d182baa24123027a0c789f27cfc3cecd374860573fb132f217c1c615feec32de61563405ace96e7298e278d6a31070b4c8bc611727b6bfbb2f299",
    "otaHeaderString": "Telink Zigbee OTA\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "manufacturerName": [
      "_TZ3000_5ajpkyq6",
      "_TZ3000_ltt60asa",
      "_TZ3000_mmkbptmx",
      "TS0004-Avv"
    ]
  },
```

andrei-lazarov, 2025-07-15:
@clumsy-stefan can you add me on Discord or something so we stop the spam here and avoid bricking more devices? 😅 
I'm andrei.lazarov on Discord.

clumsy-stefan, 2025-07-15:
Right... actually I don't use discord... would need to create an account I guess.. google chat, whatsapp, mail (clumsy.stefan@gmail.com)?

andrei-lazarov, 2025-07-16:
@romasku do you know what could block the device from updating OTA?

Stefan was able to update his _TZ3000_5ajpkyq6 OTA from stock fw to custom fw.
But the pinout is different and further updates do not work anymore.

We tried the force index and the normal index with bumped version number.
It finds the update, but does not start it. (Device didn't respond to OTA request)

romasku, 2025-07-20:
I might have a guess at what is happening here:

The device was flashed with the firmware for `TS0004_AVATTO`, which uses `firmware_image_type: 43529`.  
Now you're trying to force an update with `TS0004_AVATTO_NEUTRAL`, which uses `firmware_image_type: 43540`.  
When a firmware update is initiated, the device receives the OTA header and verifies that the `imageType` matches the one it currently has. However, since the code currently flashed to the device has the value `43529`, and the OTA header contains `43540`, the device rejects the update.

I suggest sticking with `TS0004_AVATTO` during testing to avoid complications.  
Also, the device config string supports the `i` option to override the image type in the configuration. For example:  
`TS0004-AV-CUS;TS0004-AV-CUS;BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;SA0u;RC4;i43540;`  
will force a different `imageType` value.

andrei-lazarov, 2025-07-20:
@romasku sorry for the confusion, Stefan has multiple units of this device:

1 of them was flashed with incorrect firmware.
It did not respond to OTAs so we tried to update the config string and that bricked it.

2 of them flashed with correct firmware.
But the pinout is wrong for some reason (even though it's the same manuf id).
They also do not respond to OTAs even though they have the correct image type.
The logs are from one of these 2.

andrei-lazarov, 2025-08-04:
Some updates:

The OTA issue was already fixed #90 by Stefan.

I also got the 4-gang Avatto from [AliExpress](https://vi.aliexpress.com/item/1005007247647375.html) and I can confirm it does have the new pinout:  
`BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;SA0u;RC4;`
I created a PR #99 to make this the default pinout.

Here are some pictures with my unit. It might help someone identify their device/pinout:
`MK-Z4-001-A4_V1.0 20221226`

[image]

[image]

[image]

[image]
<br></br>
Also, I find this a little funny: there is a unique code on the box to check authenticity on Avatto's <a href="http://www.miuctech.com/AntiFake.aspx?ClassID=32">anti-fake website</a> 🤔

andrei-lazarov, 2025-12-06:
@Ariel14ss14 index router.json
But make sure you connect a big load: non-smart bulb, at least 8W

Ariel14ss14, 2025-12-06:
if i have a 2 gang one, with both output together have 8W it's still ok?

andrei-lazarov, 2025-12-06:
I'm not sure. I think it needs 8W on the first gang.