romasku/tuya-zigbee-switch #10: No OTA despite properly configured files.

Yokosa, 2025-01-21:
First of all I would like to thank you for your work! 
On zigbee2mqtt version V1 I was able to reflash one relay TS00012, after upgrading zigbee2mqtt to v2 I have a problem with new devices as well as the already reflashed one. 

I have configured the external converters correctly as well as the OTA files but it doesn't find me OTA updates for the reflashed relay as well as for the new devices.

- [[2025-01-21 08:52:06] debug: 	zh:controller: Received payload: clusterID=25, address=25691, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=220, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":230,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4417,"imageType":43521,"fileVersion":16986117},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
[2025-01-21 08:52:06] debug: 	z2m: Received Zigbee message from '0xa4c13847e4ca34e2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16986117,"imageType":43521,"manufacturerCode":4417}' from endpoint 1 with groupID 0
[2025-01-21 08:52:06] debug: 	zhc:ota: [0xa4c13847e4ca34e2 | TS0042-CUSTOM] Got request '{"fieldControl":0,"manufacturerCode":4417,"imageType":43521,"fileVersion":16986117}'
[2025-01-21 08:52:06] debug: 	zhc:ota: [0xa4c13847e4ca34e2 | TS0042-CUSTOM] Checking latest image availability, current: {"fieldControl":0,"manufacturerCode":4417,"imageType":43521,"fileVersion":16986117}
[2025-01-21 08:52:06] debug: 	zhc:ota: [0xa4c13847e4ca34e2 | TS0042-CUSTOM] Getting image metadata...
[2025-01-21 08:52:06] debug: 	zhc:ota: Downloaded main index
[2025-01-21 08:52:06] debug: 	zhc:ota: Loading override index '/config/zigbee2mqtt/index_end_device-FORCE.json'
[2025-01-21 08:52:06] debug: 	zhc:ota: [0xa4c13847e4ca34e2 | TS0042-CUSTOM] No latest image currently available, current: {"fieldControl":0,"manufacturerCode":4417,"imageType":43521,"fileVersion":16986117}'
[2025-01-21 08:52:06] debug: 	zhc:ota: [0xa4c13847e4ca34e2 | TS0042-CUSTOM] Upgrade available: NO
[2025-01-21 08:52:06] info: 	z2m: No update available for '0xa4c13847e4ca34e2'] -

grzegorzlabudda, 2025-01-21:
I have the same device it is seen as ts0042 and you can't update the firmware to the latest one. it doesn't bother me but still...

romasku, 2025-01-21:
Hi,
I found an issue with OTA for the TS0012 on older firmware. Initially, I mistakenly named it TS0042, and I corrected this in a later release. However, the OTA index uses the Zigbee manufacturer information from the device to determine compatibility, and because of this change, it was unable to find the update. Please re-download the index file.

I'm still investigating why the stock TS0011 and TS0012 OTA do not work. Could you please share some details:
- Is the TS0011 visible on the OTA tab?
- Could you share a screenshot of the device info tab (similar to what you did for the TS0012 on custom firmware)?

Yokosa, 2025-01-22:
I was able to force an update of the old software to the new via a different index.json 
`{

"url": "ota/1141-aa01-01033006-tlc_switch.zigbee".

"force": true

}`
You need to download the .zigbee file and create the ota folder.

As for ts0011, I zflashed it manually yesterday via ft232rl programmer.

I still have a problem with the TS0012 which is still on Tuya software. Unfortunately, even though I have the OTA file uploaded correctly it does not search for updates for the custom software.

romasku, 2025-01-22:
Is it visible in the OTA tab? For tuya firmware -> custom firmware, [1141-d3a3-ffffffff-tlc_switch-from_tuya.zigbee](https://github.com/romasku/tuya-zigbee-switch/blob/main/bin/TS0012_END_DEVICE/1141-d3a3-ffffffff-tlc_switch-from_tuya.zigbee) should be used, as tuya firmware has different image type id. This file is only added to the non-force index file.
Here is the config index that should work:
https://github.com/romasku/tuya-zigbee-switch/blob/74ffa50f964b67faece4dba86a17ba3e6cbabd0f/zigbee2mqtt/ota/index_end_device.json#L53-L71

Also, I sometimes re-upload ota files and change `sha512` due to a change of "build date", and it may cause issues. I'll update index files to use permalinks to avoid this problem.

romasku, 2025-01-31:
Hi,
 
Did you manage to install firmware? I've changed index files to use permalinks so there should be fewer issues with Z2M unable to install the firmware.

shatpav, 2025-02-02:
Hi! Just installed z2m. Moved customised TS0012 from ZHA to z2m. Same situation. Device is configured correctly(previously OTA flashed in ZHA). External converters and index_end_device.json installed. z2m restarted, check all activated. No results, device does not appear in the OTA tab. 

[image]

[image]

Device model: TS0012-custom-end-device
Exists in the switch_custom.js

index_end_device.json:
{
    "fileName": "1141-aa01-01033007-tlc_switch.zigbee",
    "fileVersion": 16986119,
    "fileSize": 125730,
    "url": "https://github.com/romasku/tuya-zigbee-switch/raw/6333b8ee3ad4ba4e7c531ea5c080d7eec7d2ea40/bin/TS0012_END_DEVICE/1141-aa01-01033007-tlc_switch.zigbee",
    "imageType": 43521,
    "manufacturerCode": 4417,
    "sha512": "2b6cc3cad37d453100334e298e16eb5c35731f60bd5fa7bb29532e9f6a6754563867252773a23584d3c3effa4a2dfa6ff5650c8d2a5d1be6aee00e2a9c55eb15",
    "otaHeaderString": "Telink Zigbee OTA\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "manufacturerName": [
      "_TZ3000_jl7qyupf",
      "_TZ3000_nPGIPl5D",
      "_TZ3000_kpatq5pq",
      "_TZ3000_ljhbw1c9",
      "_TZ3000_4zf0crgo",
      "Tuya-CUSTOM",
      "Tuya-TS0012-custom",
      "TS0012-custom-end-device"
    ]
  },

Thanks!

romasku, 2025-02-02:
Hi,
Sorry, I really want to pinpoint this issue, but I don't see any problems on my end. My only idea is that it might be related to Z2M 2.0. What version of Z2M are you using? In 2.0, they changed the format for specifying OTA support, so I have different converters for the 2.0 and 1.x versions. Also, the OTA tab should display the device there even if the index file is missing, so the problem is likely somewhere in the converter.

shatpav, 2025-02-02:
Hi! Nothing to sorry about. It is a process and you did a lot already. :) This issue will be solved somehow as well. Thanks!

[image]

shatpav, 2025-02-02:
Just in case, I see in both external converter files this(three 'e' before OTA):
ota: ota.zigbeeeOTA,

Looks like it should be ota: ota.zigbeeOTA?

romasku, 2025-02-02:
It should be `ota: true` for Zigbee 2.0

romasku, 2025-02-02:
The correct file for Z2M : https://raw.githubusercontent.com/romasku/tuya-zigbee-switch/refs/heads/main/zigbee2mqtt/converters/switch_custom.js

romasku, 2025-02-02:
> Just in case, I see in both external converter files this(three 'e' before OTA): ota: ota.zigbeeeOTA,
> 
> Looks like it should be ota: ota.zigbeeOTA?

Yes, this is definitely a typo that was causing issues with OTA on Z2M 1.x, thank you!

shatpav, 2025-02-02:
> The correct file for Z2M 2.0 : https://raw.githubusercontent.com/romasku/tuya-zigbee-switch/refs/heads/main/zigbee2mqtt/converters/switch_custom.js

Works. Thanks!

[image]

maxwell-01, 2025-03-06:
Hello,
I am happy to open a new thread but I am having what I think might be the same error. I have the following device (TS0012):

[image]

I followed the steps here: https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/ota_flash.md

However when I get to step 4 the device does not show up. Please note that my device is a 2 gang light switch, not the small box that goes in the wall. Is this supported? 

Thanks!

romasku, 2025-03-06:
Hi, do you use the latest Z2M (version 2.1.3)?  

There was an update in the converter format that unfortunately broke the `tuya_with_ota.js` converter. I've [fixed it](https://github.com/romasku/tuya-zigbee-switch/commit/393a1adf1dd845e419ba2eb3261b9bb98a6d8790). Please re-download the converter.  

Regarding support for your device: I've recently added support for it, thanks to [this issue](https://github.com/romasku/tuya-zigbee-switch/issues/12). However, it has not been fully tested yet, so you can try it, but please be aware that it might not work correctly. I also recently received the same physical device myself, so I'll likely add more stable support soon.

maxwell-01, 2025-03-06:
Yes I’m on the latest Z2M version.

I’m a little nervous about being the first to try the new version as these are used at home by my family. I don’t have the facilities to flash it using wires.

However I will try it as soon as you’re more comfortable with it. I really appreciate your work. I have five of these sat on my desk which I was about to return because of this 0.5s delay driving me nuts! 

One thing the WiFi version of the same switch allows you to do is have a very dim light on when the switch is off. This allows you to see where they are in the dark. Is this possible with a custom firmware as the Zigbee versions don’t do it?

romasku, 2025-03-06:
Hi,

Flashing lights indicate that the device is attempting to rejoin, which means the firmware is at least partially working—so that’s a good sign. However, it tries to rejoin immediately after rebooting and then only occasionally. Try pressing the button multiple times (more than 5). There should be a brief pause in the flashing, which indicates that the device is rebooting. Make sure to open the Zigbee network beforehand.

If pressing multiple times doesn’t help, try the following steps:

- Disconnect the device from power (either by using the circuit breaker or manually).
- Open the Zigbee network.
- Reconnect the device to power.

Also, is the flashing interval stable or somewhat erratic? Do you use a neutral connection? If there is no neutral connection, the device might not be receiving enough power. Try connecting an incandescent bulb, if you have one, to ensure it has sufficient power. You can switch to the end device version later if needed.

maxwell-01, 2025-03-06:
> Also, do physical controls work as expected? I mean, are you able to turn
> on/off the light?

Hey, thank you for coming back to me so quickly! I’m pretty sure the device
is restarting when I press the keys many times as for a short while it
stops responding then starts flashing again. The flashes are perfectly
timed not sporadic.

I tried killing the power from the mains and after starting up it’s no
longer flashing but still isn’t pairing. The switch still works as expected
eg it’s turning the light on. I have it wired in with a neutral wire.

I’ll continue trying to get it to work in the morning! Night and thanks
again :)

Maxwell Newton

maxwell-01, 2025-03-06:
Couldn’t help but play a bit more.

I’ve found that I can pair it with ZHA, but not Z2M… it also pairs very
slowly eg it takes a few minutes to be discovered and then another minute
or two to finish pairing.

Maxwell Newton

maxwell-01, 2025-03-07:
@romasku I’m happy to have a bit of a play with trying to get it connected to Z2M but was wondering if you might be able to point me in the right direction of where to get started?

romasku, 2025-03-07:
Hi again,

Since it can join ZTA, it should also be able to join z2m. My current guess is that the Zigbee signal near the switch might not be very strong. Since my firmware is a bit sensitive to signal quality during the joining process, it may be failing to connect.

If you have a Zigbee socket plug or any other Zigbee router, you could try moving it closer to the switch and then attempting to rejoin.  Or you can even move your coordinator closer if it is possible. Additionally, since ZTA joins can take some time, you might try opening the network, resetting the device, and waiting for 5 minutes a couple of times. 

Unfortunately, to OTA update the firmware, you'll need to connect to z2m, so I can’t assist much with the firmware side at this point.

maxwell-01, 2025-03-08:
Hey @romasku,

Thank you for your suggestions.

Unfortunately this is the closest switch to my zigbee router and is less than 2m away. I have devices on other floors which connect without issue.

I tried three times yesterday without any luck joining Z2M. Is there anything else I can try?

Is there any way to increase the Z2M timeout so it stays open to join for longer? I may also try and investigate if there are any logs showing that a connection is at least established.

romasku, 2025-03-08:
Hmm,

Routers often have a limited number of devices that can connect to them as child devices. This is purely speculative, but perhaps the router near the switch has reached its limit of supported devices. In this case, your switch might detect the router when attempting to join and try to connect through it, fail, and then repeat the process. To verify this, you can allow joining only through specific devices—try using other nearby routers; this might help.

[image]

Regarding z2m timeout, as I see, 4:14 (254 seconds) is [the maximum duration](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-request-permit-join).

maxwell-01, 2025-03-08:
Thank you for the suggestion. However currently all my devices are on ZHA and I only have two devices on Z2M. I was going to start replacing all my WiFi switches with these moes ones but paused because the 0.5s delay that you’ve fixed.

I’ve tried several times leaving it until the timer runs out but no connection on Z2M. I’ll see if I can find some docs and then some verbose logs.

maxwell-01, 2025-03-12:
I tried this process on a second Moes 2 gang switch and I had the same result. It wont connect to Z2M. Here are some errors from the log after the update:

```
info 2025-03-12 13:56:03z2m: Interviewing 'Kitchen 4 gang left'
error 2025-03-12 13:56:11z2m: Failed to configure 'Kitchen 4 gang left', attempt 2 (Error: ZCL command 0xa4c138e5f670a541/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:545:23) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:642:24) at /app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:443:20 at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20) at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:441:16) at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:82:20) at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@3.2.7/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:324:20))

error 2025-03-12 13:56:54z2m: Publish 'set' 'state' to 'Kitchen 4 gang left' failed: 'Error: ZCL command 0xa4c138e5f670a541/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'
```

It seems like it fails to connect after the update. If I remove the device i cant add it back into Z2M but I can join it to ZHA.

Before I tried this, I tried deleting the device with the stock firmware and adding back into Z2M. It added instantly, eg within a few milliseconds of me clicking permit join. 

Any advice welcome!

maxwell-01, 2025-03-12:
Hey Romasku,

Having them running temporarily in ZHA is fine and I can see both switches (which is all I need for now), but it does limit the upgrade path for future OTA updates.

For Z2M I use a [SONOFF Zigbee 3.0 USB Dongle Plus](https://www.amazon.co.uk/dp/B09KXTCMSC)
For ZHA I use a [SONOFF Zigbee Hub](https://vi.aliexpress.com/item/4001143062435.html) with Tasmota on it

maxwell-01, 2025-03-12:
Some odd behaviour with the switch I updated today. I can turn the light on and off through HA (in ZHA) but if I manually press the switch on or off then HA does not update. It is wired in with a neutral wire so its not an energy issue.

romasku, 2025-03-12:
Hi once again,  

I've rechecked and found that the firmware retries joining approximately once a minute. I've updated the code to make it more frequent, but to enable this, you'll need to update the firmware :(. Additionally, if the device hasn't joined within ~15 seconds after resetting, it's better to reset it and try again. This way, you'll be attempting to join four times per minute instead of just once.  

I was able to reproduce your issue somewhat, but I had to disable two out of three routers near the test device. One router was still nearby, but it was overwhelmed by EndDevices and couldn't accept the device. However, as soon as I re-enabled another router, the device joined almost instantly.  

You can try enabling OTA in ZHA as described [here](https://github.com/pvvx/ZigbeeTLc/issues/7), but using the OTA file from my project. I haven't tested this myself, as I don't have ZHA at home or a spare Zigbee coordinator to try it with, so I can't provide much guidance.

maxwell-01, 2025-03-12:
Hey Roman,

Thanks again for your message. I’ve only got two devices on Z2M so far so I don’t think it’s overwhelmed but it was an interesting idea.

I’ve been troubleshooting with ChatGPT and it suggested the following:

If the new firmware has a hardcoded Zigbee network key, it may not match the key that Zigbee2MQTT expects.
	•	Zigbee devices joining a network use a TC Link Key—if ZHA uses a different default key than Z2M, this might explain why it joins ZHA but not Z2M

Does this mean anything to you? I wondered if possibly there was something hard coded into the firmware specific to your network.

I didn’t know about ZHA OTA so thank you for bringing that to my attention, I’ll check it out and see if I can install your new version.

All the best :)

romasku, 2025-03-14:
Hello again,  

I see that CRCinAU was able to update the device over-the-air (OTA) in ZHA. Check out this comment: https://github.com/romasku/tuya-zigbee-switch/issues/14#issuecomment-2723169672

I don't think I have anything specific to my network hardcoded into the firmware, especially security keys.

romasku, 2025-03-15:
Hi @maxwell-01,

I added (a bit unpolished) a custom quirk for ZHA to enable full support of the device. See [guide here](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/zha_ota.md). Please let me now if it works for you.

maxwell-01, 2025-03-25:
Thanks! I've not had time to give this a go just yet but I will.

I have an unrelated question: the wifi / tuya version of this switch has a setting in the tuya app called 'backlight switch'. When this is enabled it keeps the LED on very dimly when the switch is off. This means at night you can see where the switches are. The LED still emits at full brightness when the switch is turned on.

Is it possible to replicate this behaviour in your firmware? 

[image]

kubenalu, 2025-05-12:
Hi @romasku  , I have a similar issue with TS0012 - manufacturer _TZ3000_jl7qyupf - Z2M version 2.3.0. The device is not shown in OTA. When I put external converters 2.0 in the folder and restart z2m, the file tuya_with_ota.js is automatically renamed to tuya_with_ota.js.invalid which probably causes it. Can you help me with this, please?

romasku, 2025-05-12:
Interesting, I'll take a look tomorrow.

kubenalu, 2025-05-14:
I forgot to mention I already have one older TS0012 - _TZ3000_jl7qyupf with your custom firmware which works perfectly and I can see it in OTA.

gresuk, 2025-05-19:
Hello. Is it possible to flash via OTA on Z2M 2.3.0? I don't see the converters being loaded in the log. The paths seem to be correct (homeassistant/zigbee2mqtt/external_converters/tuya_with_ota.js, /homeassistant/zigbee2mqtt/external_converters/switch_custom.js). The device appears in the OTA section, but it says that no update is available. I'm trying to flash _TZ3000_skueekg3. Thank you.

gresuk, 2025-05-19:
manufacturerCode":4417,"imageType":518,"fileVersion":268449287}

kubenalu, 2025-05-26:
@romasku  I just tested that and the issue is gone. Thank you

maxwell-01, 2025-08-22:
> Thanks! I've not had time to give this a go just yet but I will.
>
> I have an unrelated question: the wifi / tuya version of this switch has a setting in the tuya app called 'backlight switch'. When [...]

Hello, I was wondering if any of the contributors have seen this and whether it is something that can be achieved with the custom firmware? It's very useful at night to see a very dim LED to help locate the switches. This becomes a bright LED with the switch is turned on (the same as the custom firmware currently does). The only change here is the dim LED when the switch is off.

maxwell-01, 2025-08-22:
@andrei-lazarov thank you for the speedy response. 
That is the correct device and yes I am suggesting dimming instead of turning off.

Thank you - is there anything I can do to help, eg should I raise a separate issue tagged as a feature request?

andrei-lazarov, 2025-08-22:
@maxwell-01 
> is there anything I can do to help

Not much other than getting a hardware flasher and coding it yourself 😂 
You may open a new issue, sure.

andrei-lazarov, 2025-08-29:
@MakAndJo indeed that looks like a version incompatibility.. don't think I can help.  

There isn't anything hidden in the attributes. Everything useful should be exposed in the quirks / converters.  
You can search the source code for SETUP_ATTR or ask on Discord what exactly you want to know.

MakAndJo, 2025-08-30:
Thanks, ive read out some sources and gotcha how it works and how it can be configured.
Also i thought i have an issue with momentary switches. Logic is inversed because i have NC switches. 
Now i wanna build my own version for `jl7qyupf` with removed bindings, level moves (bcs i dont need that) and only for momentary NC swithes.

andrei-lazarov, 2025-08-30:
@MakAndJo If you manage to do it, please add it as an option (toggle, momentary_no, momentary_nc) and create a PR so we can merge it to the firmware and you can stay on the main branch.  
Not sure why you would remove the bindings and level moves.. you can just not use them.

bieskholodov, 2025-11-30:
[image]

andrei-lazarov, 2025-11-30:
@bieskholodov Yes. They probably have **wrong imageType** from the initial install.
(You can normally change it in the config string with `i00000;` - but the option doesn't work on v1.1.0-1.1.1)

Please do the following, to **find the imageType of the switches that are not updating:**
1. Enable debug logs in Z2M user interface: Z2M > Logs > Log level config
2. Click 'Check for updates'
3. View the logs on the host machine: HA > Add-ons > Z2M > Log, or `docker container logs`
4. Look for imageType and manufacturerCode, or paste the logs here.
5. Disable debug logs (change back to info).

Also maybe **create a new issue** - so we don't spam the people in this thread.

bieskholodov, 2025-11-30:
[01.12.2025, 00:39:53] z2m: Failed to check if update available for 'Свет детская' (Device didn't respond to OTA request)
[01.12.2025, 00:39:53] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Failed to check if update available for 'Свет прихожая' (Device didn't respond to OTA request)","status":"error","transaction":"zr3hq-3"}'
[01.12.2025, 00:39:53] z2m: Failed to check if update available for 'Свет прихожая' (Device didn't respond to OTA request)
[01.12.2025, 00:39:53] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Failed to check if update available for 'Свет ванна' (Device didn't respond to OTA request)","status":"error","transaction":"zr3hq-5"}'
[01.12.2025, 00:39:53] z2m: Failed to check if update available for 'Свет ванна' (Device didn't respond to OTA request)
[01.12.2025, 00:39:53] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Failed to check if update available for 'Свет зал' (Device didn't respond to OTA request)","status":"error","transaction":"zr3hq-6"}'
[01.12.2025, 00:39:53] z2m: Failed to check if update available for 'Свет зал' (Device didn't respond to OTA request)
[01.12.2025, 00:39:53] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{},"error":"Failed to check if update available for 'Температура влажность' (Device didn't respond to OTA request)","status":"error","transaction":"zr3hq-13"}'
[01.12.2025, 00:39:53] z2m: Failed to check if update available for 'Температура влажность' (Device didn't respond to OTA request)``

romasku, 2025-12-09:
Closing as the initial problem was fixed long ago. If you think we need to continue tracking of some ideas from this topic - please open a new issue.

closed 2025-12-09 by romasku