romasku/tuya-zigbee-switch #12: Relay based light switch

DeadSomeOne, 2025-02-16:
Hi,

I own a [Moes 2 Gang light switch, no need for neutral wire](https://www.aliexpress.com/item/1005005178430448.html). HA, identifies it as `TS0012` by `_TZ3000_18ejxno0`. It has a ZTU module.
Adding OTA, shows it's up to date. File version and manufacturer code match, but image type on my switch seems to be unset, reports as 65535.
Would your existing image(s) for TS0012 work? Or do I need to follow the porting guide, to add for e.g. button presses etc.?

Adding cluster information from zha_toolkit [TS0012.txt](https://github.com/user-attachments/files/18815372/TS0012.txt)

Thanks for your work and thanks in advance.

romasku, 2025-02-18:
Hi,

I don't think the existing image for TS0012 will work, unfortunately, as ZTU modules expose more pins than ZT3L, and even the common pins are placed differently. This makes it unlikely that the pinout is the same. So yes, you'll probably need to deduce the pinout from scratch.

If you're going to follow the porting guide, I suggest using the TS0002_AVATTO image as a base instead of TS0012, since TS0012 uses floating pins, which may cause issues. Additionally, TS0002_AVATTO is a ZTU-based device, increasing the chance that the pinout will be the same or at least similar.

Hope this helps.

romasku, 2025-02-23:
Hi,  
No, there's no need to remove `switch_custom.js`. This script enables Z2M to work with custom firmware—without it, you'll get an unsupported device error. You can remove `tuya_with_ota.js` after upgrading via OTA, but it's not necessary.  

Also, I'm not sure what you mean by "second upgrade." After the first OTA update, if everything is OK, the device should reconnect to Z2M, and you'll see this in the *Exposes* tab:  

[image]  

Then, you need to change the "Device config" value. After each change, the device reboots (which takes less than a second), and the new configuration is applied.  

Yes, you set the relay pin using `R`. For example, if the pin is `B1`, you need to add `RB1` to the config. Note that you should use the pin names from the Tuya documentation: [Tuya Module Datasheet](https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp). Instead of "pin 5," you need to use "C4," so the correct config entry would be `SC4`.  

If you're willing to wait a bit, I can create a new firmware binary with the config based on your pinout. This way, there will be less risk of bricking the device due to an incorrect config. Alternatively, you can follow the guide yourself, but be prepared to use UART to restore the device if something goes wrong.

DeadSomeOne, 2025-02-24:
Hi,
I will take up your offer and wait for a your firmware, planning to make it a router, if that matters.
>second upgrade

in [Setting Up the Device After Update](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/porting_to_new_device.md#setting-up-the-device-after-update), it states `the same way as for the converter for stock firmware` so I thought I need to update the device again. I was unsure if I needed to add both `tuya_with_ota.js` and `switch_custom.js` . I guess I should just create them and leave them be.

romasku, 2025-02-24:
Ok, I'll try to build the firmware for you this evening.

And thank you for your comment regarding the guide. I'll try to improve it to make it clearer. I appreciate it, as my current goal is to make porting to new devices as easy as possible.

romasku, 2025-02-26:
Sorry for the delay, 
I'm working on your pinout. As this is the first device with leds per switch for this firmware, I need to write some code to support them. So please wait, I hope to finish soon.

romasku, 2025-02-26:
I've built the firmware for you. Just download the latest `index.json`, `tuya_with_ota.js`, and `switch_custom.js`, then follow the [OTA instructions](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/ota_flash.md). It should now work for your device.  

The device configuration is not final: the 2x4 connection is used for controlling relays, but I'm not sure which relay is controlled by which pin. So, I added all three relays—when you test, please let me know which one is working so I can fix the configuration. Additionally, the status LED order may be shuffled, as I'm also unsure about the relay order. Finally, the LEDs may be inverted—if that happens, I'll update the configuration accordingly.

And thank you for providing the pinout, it really helps with porting the firmware.

DeadSomeOne, 2025-02-27:
Hi.
So I just paired in zigbee2mqtt, just to be safe of any mistakes. OTA wasn't found. Seems z2m reports different info from HA...
```zh:controller: Received payload: clusterID=25, address=36134, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=228, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":247,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":80},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}```

So I just changed [this value](https://github.com/romasku/tuya-zigbee-switch/blob/main/zigbee2mqtt/ota/index_router.json#L59) to `54179`, seems to work, but now header is different in the OTA. Sorry for the mistake.

Waiting for your reply.

romasku, 2025-02-27:
I've rebuilt the image, please try again.

DeadSomeOne, 2025-02-27:
I've updated successfully,

[image]

Now, current state. Button 1 doesn't do anything. button 3 does not save the state once pressed. Power works only when holding it. LEDs seem to work fine. Button 1 and Button 3 should be swapped places, configuration wise, so it'd be the same as stock. `Relay indicator` option does not do anything.

Are you planning to make this firmware universal other 1, 2 and 3 gang switches to have all 3 relays available? Would be nice to have button/relay numbers on bottom section

[image]

romasku, 2025-02-27:
> Power works only when holding it.

Please change the mode to momentary.

[image]

romasku, 2025-02-27:
> Button 1 doesn't do anything. 
So only one relay can be controlled via physical switches, correct? Please try turning on all entries for relays:

[image]

I think the relay is currently attached to the missing switch, but it should be possible to turn it on via Zigbee.

DeadSomeOne, 2025-02-27:
Currently tuning it. Bit overwhelming with the options :D.
It's a bit of a mess with LEDs/relays. LEDs seems to be defined fine. but 2 gang switch does not have left relay/relay_1. On 2 Gang switch, with **default fw**, with Button 1 you control relay_3 and left led. Button 2 does not exist. Button 3 controls relay_2 and right led. I just missing the option to change the controlled LED light to fix the above issue. rest seems fine, apart it still being an end device (no neutral is connected)

romasku, 2025-02-27:
> being an end device

Try re-interviewing it:

[image]

Z2M doesn't expect the device to change its type so it displays the old one.

I'll update the device config later to drop unexisting buttons (there will be a separate build for 3-gang devices), and I'll probably hide settings that are not applicable for a switch (like "momentary/toggle" setting). Your input is actually what I need to do it, so thank you!

DeadSomeOne, 2025-02-27:
As additional information,
You cannot easily configure on your firmware when using HA, compared to z2m. So stuck using default values which kinda don't work nicely physically.

[image]

DeadSomeOne, 2025-02-28:
Additionally I noticed, relays turn off if slim to none load is applied to them. E.g. I turn off my zigbee light bulbs and their status does not reflect in HA

romasku, 2025-02-28:
Hello,

I believe this is because you are using Router firmware. The router requires more power since it stays on constantly, whereas the End Device sleeps most of the time. My experiments show that the device needs approximately 10–15W of load to function properly.

I suggest you:

- Try adding a bypass resistor, which sometimes comes with no-neutral devices.
- Use dumb lights, if possible.
- Reflash the End Device firmware if nothing else helps.

On a separate note, I'll try to work on a better version of the firmware for your device tomorrow when I have time.

DeadSomeOne, 2025-02-28:
Wow, 10-15W is quite a bit,... I've noticed it's fine when one relay is active. If I activate both relays and turn off my lights, fairly quickly they shut off. 

- How would I do that? Heard of only capacitors for bulbs.
- :(
- I'll try end device fw tomorrow, to check if it's the same or not

I'll be back later than usual tomorrow, so take your time. I appreciate your efforts, the delay on the buttons were driving me nuts.

romasku, 2025-03-02:
Hi again,

I've updated the config to remove non-existent buttons/relays and also tried to synchronize the LEDs, switches, and relays. Please try updating the firmware.

Regarding the 10-15W load — this is not a constant load but rather an "attached load." Let me explain. In tests with a router device, I observed that if there was only a single dumb LED bulb of 7W, the device did not function stably: it sometimes rebooted both when the light was on and off. My understanding is that the device requires a relatively low resistance load to pass a small amount of current to power itself. As a result, it effectively needs an attached load of about 10-15W to allow a tiny current to pass (approximately 0.5W, though I haven't measured it).

I'm not sure how this will work with smart bulbs that are turned off. I assume that when they are off, they function as approximately 0.5W bulbs, which might not provide enough "attached load" for the device to operate in router mode.

As for capacitors and resistors, they should theoretically help. However, since I haven't experienced the issue myself (I have more than one bulb attached to a single gang), I haven't installed them and can't confirm whether they are effective or not.

DeadSomeOne, 2025-03-02:
Hi,
can you bump the version? It says no update available
```mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1383c39acc13d', payload '{"device_config_1":"Moes-2-gang;Moes-2-gang;SB6u;SC1u;SC4u;RD2;RB4;RB5;ID3;ID7;IC0","linkquality":244,"power_on_behavior_left":"off","power_on_behavior_middle":"off","power_on_behavior_right":"off","relay_left_indicator_mode_left":"same","relay_middle_indicator_mode_middle":"same","relay_right_indicator_mode_right":"same","state_left":"OFF","state_middle":"OFF","state_right":"OFF","switch_1_action_mode_1":"toggle","switch_1_mode_1":"toggle","switch_1_press_action_1":"released","switch_1_relay_index_1":"relay_1","switch_1_relay_mode_1":"raise","switch_2_action_mode_2":"toggle","switch_2_mode_2":"toggle","switch_2_press_action_2":"released","switch_2_relay_index_2":"relay_2","switch_2_relay_mode_2":"raise","switch_3_action_mode_3":"toggle","switch_3_mode_3":"toggle","switch_3_press_action_3":"released","switch_3_relay_index_3":"relay_3","switch_3_relay_mode_3":"raise","update":{"installed_version":16986119,"latest_version":16986119,"state":"idle"}}'```

romasku, 2025-03-02:
I actually already did, please re-download `index.json` and `switch_custom.js`.

DeadSomeOne, 2025-03-02:
Still no update... Even downloaded source and copy pasted the files

DeadSomeOne, 2025-03-02:
Was the head version bumper (if it checks by that)

romasku, 2025-03-02:
Hmm, strange, if even FORCE doesn't work, then there are probably wrong selector somewhere. Let me check.

romasku, 2025-03-02:
Ouch, I see the reason. I forgot that `I` config entry I added for **I**ndictator was already there and used for **I**mageType. So currently, the device thinks that it's image_type is 0.

Please set config to `Moes-2-gang;Moes-2-gang;SB6u;SC1u;SC4u;RD2;RB4;RB5;` and then try updating.

DeadSomeOne, 2025-03-02:
I get the following error, when trying to set it:
`error:    z2m: Publish 'get' 'device_config' to '0xa4c1383c39acc13d' failed: 'Error: ZCL command 0xa4c1383c39acc13d/1 genBasic.read([65280], {"timeout":30000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '5235'.)'`
`debug:    z2m: Error: ZCL command 0xa4c1383c39acc13d/1 genBasic.read([65280], {"timeout":30000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '5235'.)`

DeadSomeOne, 2025-03-02:
No leds are lighting up when a button is pressed physically. I'm guessing it has gotten bricked

romasku, 2025-03-02:
Is this after an update or after applying the config? Config removed LED support, as those config entries were breaking the OTA process.

DeadSomeOne, 2025-03-02:
Nor leds are working, nor I hear any relays switching when any button is pressed. pressing 5 times doesn't seems to do anything this time as well

romasku, 2025-03-02:
Ouch, that sounds like it is completely dead. Sorry to hear that. Do you have a UART programmer to try to revive it?

DeadSomeOne, 2025-03-02:
Currently no. I'll either buy [this one](https://www.waveshare.com/wiki/FT232_USB_UART_Board_(mini)) or borrow one from work

DeadSomeOne, 2025-03-03:
Hi, I soldered the wires, before I start flashing. which fw file do I need? the [tlc_switch.bin](https://github.com/romasku/tuya-zigbee-switch/blob/refs/heads/main/bin/MOES_2_GANG_SWITCH/tlc_switch.bin)? or [1141-a9f6-01033008-tlc_switch.zigbee](https://github.com/romasku/tuya-zigbee-switch/blob/refs/heads/main/bin/MOES_2_GANG_SWITCH/1141-a9f6-01033008-tlc_switch.zigbee)

romasku, 2025-03-03:
tlc_switch.bin

DeadSomeOne, 2025-03-03:
Seems it's still dead...

romasku, 2025-03-03:
Sad to hear.

Did you manage to flash it, but it doesn't start after flashing? Or is flashing via UART not working at all?

DeadSomeOne, 2025-03-03:
got OK from the flasher, but it doesn't start.

romasku, 2025-03-03:
Can you also try this version: https://github.com/romasku/tuya-zigbee-switch/raw/4cb497c8847a37d61758dbe649b296e4ea55a641/bin/MOES_2_GANG_SWITCH/tlc_switch.bin

This is an old build that initially partially worked for you.

DeadSomeOne, 2025-03-03:
Nothing :/
Also with my big *** iron, managed to rip VCC and GND pads, however measuring with a multimeter still shows voltage at other pins

DeadSomeOne, 2025-03-03:
[image]

[image]

romasku, 2025-03-03:
Flashing via UART can be a bit unreliable, as the flasher has no feedback from the device, but I see that you've connected RTS->Reset, which often solves most of the issues. All other connections also look correct. Can you try using a bigger Baud? Also, try connecting 3v3 pins during "Activate 5s". And also try flashing without RTS->Reset, just to be sure.

Also, could you please power up the device from UART's 3.3V (do not connect reset and SWS) and check the following:

- Is there any voltage on the LED pins?
- Is there any voltage on the Relay (10, 11, 12) pins?
- Please measure the current consumed by the module using a multimeter. It should be around 10mA.

Hunter32R, 2025-03-03:
USB-UART bridges on ftdi chips cannot flash tlsr8258, since they cannot emulate the swire protocol, another bridge is needed.

DeadSomeOne, 2025-03-03:
> Flashing via UART can be a bit unreliable, as the flasher has no feedback from the device, but I see that you've connected RTS->Reset, which often solves most of the issues. All other connections [...]

Nothing.
Draws around 4.1mA

DeadSomeOne, 2025-03-03:
> USB-UART bridges on ftdi chips cannot flash tlsr8258, since they cannot emulate the swire protocol, another bridge is needed. In my experience, bridges on pl2303hx chips are the best, as they support high bitrates up to 12 Mbps.

I can only get something like this. if it works.

[image]

Hunter32R, 2025-03-03:
> > USB-UART bridges on ftdi chips cannot flash tlsr8258, since they cannot emulate the swire protocol, another bridge is needed. In my experience, bridges on pl2303hx chips are the best, as they support high bitrates up to 12 Mbps.
> 
> I can only get something like this. if it works.
> 
> [image]

This should work.

Hunter32R, 2025-03-03:
5 volts can damage the chip, as an option you can take power from another adapter.

romasku, 2025-03-03:
I personally use FT232RL (a red one), and it allows me to flash tlsr8258. It is also an ftdi chip.

DeadSomeOne, 2025-03-03:
Okay, fair enough. Green is tx and white is rx. So I'd assume green goes to SWS and white to Reset

DeadSomeOne, 2025-03-04:
Hi again, I just tried with the USB cable one UART. Still no signs of life :/

DeadSomeOne, 2025-03-05:
Since flashing via wire is bearing no fruit, I've ordered a new switch. Should get it late next week or the week after.

DeadSomeOne, 2025-03-14:
Hi, today i will be receiving a new switch. Is the current OTA okay? If yes, I'll test out tomorrow and let you know how it works

romasku, 2025-03-14:
It seems that the device [from this issue](https://github.com/romasku/tuya-zigbee-switch/issues/14) is indeed almost the same one, as the pinout I reversed matches the one you provided earlier. It looks like the latest firmware version works properly, so I think you can give it a try. However, I cannot guarantee anything, so please proceed with caution.

DeadSomeOne, 2025-03-15:
Hi. I have flashed your firmware again. Flashed end device. The fw seems fine on HA via ZHA, but missing LED control, like in stock FW.
Tried flashing from end device to router. Although lengthy, at 30 minutes (15 when from stock), it was successful. Works fine.
As for my before mentioned issue about relays turning off. I guess it's a thing with the LED bulbs. I can take out 4 bulbs, and it barely works, but it does. Although if I have 5 bulbs installed (3 bulbs on relay_3, 2 bulbs on relay_2), it works just fine now.

 While I don't mind, maybe device manufacturer and model could be changed up, kinda confusing right now.

[image]

Once relay LED support is added to ZHA, I'd consider this issue closed, as this custom fw has complete stock fw functionality, as well as extras.

A bit off topic. just a small notice about the new device table. Maybe you should update the model name for the switch to `ZS-EUB_2gang` as you did for the single gang one? \
[image]\
P.S. WHD02 has `router / router` type in the table.

romasku, 2025-03-15:
Hello,

I added (a bit unpolished) a custom quirk for ZHA to enable full support of the device. See [guide here](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/zha_ota.md). Adding a custom quirk should be completely safe, so please give it a try.

DeadSomeOne, 2025-03-16:
Hi.
Oh wow that added quite a bit of options. Can control LEDs now. 
also this doesn't seems to update/work

[image]

and thank you. I consider this as issue closed. You can close it if you don't have anything more to add

romasku, 2025-03-16:
> also this doesn't seems to update/work

Try pressing the "reconfigure" button on the ZHA device page.

DeadSomeOne, 2025-03-17:
Welp.
Didn't work until I reconfigured and restarted HA. Now it does. Thank you

romasku, 2025-03-17:
Good. If you come across any other ZHA support issues, please open a new issue. Thank you for your testing—it’s truly helpful!

closed 2025-03-17 by romasku