romasku/tuya-zigbee-switch #14: New device support request - Moes zigbee light switches

MnM001, 2025-02-19:
Hi @romasku ,

I have a few of Moes zigbee light switches:

_TZ3000_hhiodade - 1 gang - identified in z2m as Moes ZS-EUB_1gang
_TZ3000_18ejxno0 - 2 gang - identified in z2m as Moes ZS-EUB_2gang
_TZ3000_qewo8dlz - 3 gang - identified in z2m as TuYa TS0013
_TZ3000_mrduubod - 4 gang - identified in z2m as TuYa TS0014

I will love to have custom zigbee firmware for them if possible.  What I am looking for is:

detached mode
bindings
maybe make them work as routers (currently they only work as end devices)
button led support (have the led on or off)
power-on behavior (previous, off, on)

It seems that at this point in time they are not supported. Please let me know if I can help with anything - sadly I am not good with coding or writing software. I can help with testing, information required, etc

romasku, 2025-02-19:
Hi,  

Does [this article](https://smarthomescene.com/reviews/moes-zigbee-3-gang-wall-switch-zs-eub-review/) contain photos of the correct device? If so, then it is possible to adapt firmware for these light switches. However, this requires the correct pinout (i.e., which pins of the microprocessor are connected to which buttons/relays). It can be reverse-engineered using a multimeter and some trial and error. Unfortunately, I can't do this myself since I don't have the device on hand.  

If you can't do reverse-engineering of pinout yourself, you can send me high-resolution photos of the device's PCB (similar to those in the article above, but in higher resolution), and I'll try to deduce the pinout. However, I can't guarantee that it will be possible to determine everything from photos alone.

MnM001, 2025-02-20:
Hi,

No they are not the right devices. More information about the devices is here:

https://github.com/doctor64/tuyaZigbee/issues/32

However, I am willing to go a bit further than just the above link or the pictures in that link. I can buy one of each (1/2/3/4 gangs)  and have them shipped to you or, you can buy them from the above link and I send $$ to you to cover the expense. I am willing to do this as I am not good with tracing the pinouts and whatever else is required. This way if you actually have the devices you will know exactly what to look for and how to go about it.

What do you think? Is the proposed approach better?

romasku, 2025-02-20:
Hi,

Thank you for your generosity—I truly appreciate it! I’d prefer to order the devices myself rather than share my address publicly. You can send the funds to my PayPal at [romasku135@gmail.com](mailto:romasku135@gmail.com) (aliexpress shows $50 as the total price), and I'll start working on them as soon as they arrive.  

Regarding the features you mentioned, most are already included in the firmware. However, there is currently no support for LED behavior, as I primarily focus on modules rather than switches. That said, adding support for it should be relatively straightforward.  

Let me know if this approach works for you.

MnM001, 2025-02-20:
Hi @romasku,

Thank you for agreeing to help me with the devices. $$ was send via PayPal so please go ahead and order them from Ali Express.
really looking forward to have custom firmware for them.

Cheers.

MnM001, 2025-02-21:
one thing that I forgot to mention (not sure if your project does this or not) - will be good if there is a way to backup the original firmware before flashing the customized one. just in case its needed to revert to at some point later on.

romasku, 2025-02-21:
Thank you,

Ordered, I'll notify here when the switches arrive.

Regarding backup, it is possible by soldering UART wires and running a backup script, but I do not have instructions posted at the moment. I also have an idea for adding backup functionality to the OTA process, which I’ll be working on next.

DeadSomeOne, 2025-02-22:
Hi, looks like this ticket is somewhat related to #12 I created, but I have a switch with physical buttons. The tracing is fairly simple and you could figure it out easily, but still used a multimeter. On my physical switch, buttons are traced like this:
- button 1 (S1) - pin 5
- button 2 (S2) - pin 19
- button 3 (S3) - pin 17

LEDs:
- Led 1 (L1) - pin 1
- Led 2 (L2) - pin 2
- Led 3 (L3) - pin 3

All leds share the same 3v3 (pin 14) line, if believed [this doc](https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp)
Note: While my switch has 2 buttons, traces for the 3rd button exists.

Pinned out the 2x4 connection as well. Attaching the whole photo of what I found. 

[image]

MnM001, 2025-02-22:
Hi @DeadSomeOne ,

Thank you for supplying the picture and information. It might end up that the devices are the same internally but I really wanted to be sure of success with my devices, hence why I don't mind if @romasku can order them and work on them. 
It will be much easier for him to figure out what's required and in my view) a better chance of him supplying a custom firmware that have all what I need in it.

DeadSomeOne, 2025-02-22:
Speaking from experience, flashing something always has risks in breaking something, let it be tuya fw or other. 
Regardless, I hope for a custom fw as well, the button delay is really annoying on the physical switch. I do understand programming somewhat, but not to the extend to build fw and etc., just here to chime in with the findings and help with testing, if needed :).
P.S. I appreciate people like you, who sponsor developers with to get the physical device. <3

MnM001, 2025-03-05:
Hi @romasku - have you got nay news on the devices? is it long before delivery?

Cheers.

romasku, 2025-03-05:
Hi,
I just checked, and it arrived at the customs office this morning. Hopefully, I'll receive it soon.

romasku, 2025-03-05:
The devices arrived this evening (the local post worked magic and delivered them just 12 hours after customs!).

I've unboxed them and noticed that the 1, 2, and 3-gang models use almost the same board and Tuya's ZTU module as the main controller. Porting the firmware should be straightforward, and I'll work on it in the next few days.

Unfortunately, the 4-gang model uses Tuya's ZS3L, which features a completely different microcontroller — the Silicon Labs EFR32MG21 instead of Telink's TLS8258. While creating firmware for it isn’t entirely impossible, it would require starting from scratch due to the completely different architecture. I might give it a try (especially since I also have some Silicon Labs devices at home that I'd like to re-flash), but it will definitely take weeks to develop.

MnM001, 2025-03-05:
Hi,

Thank you for the update.
Great that the devices arrived! Will be good if you can could create custom firmware for the 4 button device too. I understand that it might take longer as it is new and different architecture. Hopefully you can get there :)

Thanks again for your help with this!!

maxwell-01, 2025-03-07:
Your table [here](https://github.com/romasku/tuya-zigbee-switch?tab=readme-ov-file) links to this page next to TS0012.

Looking back at post 2 in this thread. You linked [this article](https://smarthomescene.com/reviews/moes-zigbee-3-gang-wall-switch-zs-eub-review/). This does accurately represent the Moes TS0012 that I have at home. 

Here is a [link to the devices on AliExpress](https://vi.aliexpress.com/item/1005005178430448.html).

Is it possible the [table](https://github.com/romasku/tuya-zigbee-switch?tab=readme-ov-file) is linking to the wrong issue?

romasku, 2025-03-07:
No, as far as I understand, this is the same device but with US and EU versions. I'm not completely sure, as I haven't had time to reverse-engineer the pinout for the US version I received but, since they share the same manufacturer name, they are probably identical in terms of the controller and will likely use the same firmware.

romasku, 2025-03-08:
@MnM001
Hello once again,

I've ported the firmware to the 1, 2, 3-gang versions of the switch. I tested the OTA update, and it worked fine for me. You can try installing it on your devices, but I suggest proceeding with caution by flashing the least important switch first, just to be sure.

I also dumped the stock firmware and uploaded it to the repo, but flashing the stock firmware will require a UART converter.

MnM001, 2025-03-09:
Hi @romasku - that was very fast!! Thank you for this!
 Can you please let me know which options were you able to achieve with this firmware?

detached mode
bindings
maybe make them work as routers (currently they only work as end devices)
button led support (have the led on or off)
power-on behavior (previous, off, on)

Also I have another question - on a switch is it possible to have one button work with detached mode and one button to work normal? Some of my 2 button switches have one button controlling a light and the other button is controlling a fan. While the light can be toggle on/off the fan need normal operation from the its button.

Is there any chance that the stock firmware will ever be able to be updated via OTA?

Thanks again.

MnM001, 2025-03-09:
Hi @romasku - I have flashed a couple of 1 button switches. Such a straightforward and painless process, especially with the direct link to github for the update file!!

I have one issue that I have not managed myself to make it work:

- in detached mode, I use HA to toggle the light ON/OFF. This works as it should the light stays on the network as is only the light that toggles. However, the LED on the light switch does not change at all when the light is toggled form HA. I cannot see any options in z2m that will allow me to correct this behavior. Maybe is that the LED needs its own entry in z2m so that when I have any automation in HA to toggle the light the same automation can also turn on/off the led? Or maybe is something else and I am missing the point all together).  In the normal configuration like below the LED light behaves as it should when the switch button is pressed for ON/OFF (but that also cuts the power to the downlight:

[image]

Another thing that I am trying to see if it can be fixed is more cosmetic than anything else.
Can the red Tuya icon that appears after the customs firmware was flashed to a device revert back what z2m used to display before?

From this

[image]

To this

[image]

Thanks again.

MnM001, 2025-03-10:
Also one more thing - I have tried to change from end device to router just by changing the OTA line under z2m's configuration.yaml:

```
ota:
  zigbee_ota_override_index_location: >-
        https://github.com/romasku/tuya-zigbee-switch/raw/refs/heads/main/zigbee2mqtt/ota/index_router-FORCE.json
```

But the device is not seeing an update. Should this change from end device to router also work with ITA and github file (rather than a local file)?

Cheers.

romasku, 2025-03-10:
Hi,

Regarding the LED state for detached mode, please select the LED mode "Manual." Then, the "Relay relay indicator" should allow you to control the LED manually, for example, through automation. This approach is necessary because the device does not know what it controls or whether it is on or off at any given moment.

As for the LED state after a power cut: I will likely need to add additional logic to the firmware, so I’ll ping you here when I do. And thanks for reporting the issue.

romasku, 2025-03-10:
Update: I identified the issue with the LED state after a power cut. The LED mode ("opposite") was not being saved properly and was reverting to the default ("same") state. This has been fixed in version 1.0.10.

MnM001, 2025-03-10:
I have discovered another "issue" this time for the router mode. After a power cut if the device is in end user mode the device comes back to the network straight away after power has been restored.

In router mode it seems that the device is not advertising that it is back, so it comes back to  the network after z2m polls it (in my case this polling is set at 15 minutes). Will be good if in router mode it can be back on the network as soon as the power is back like in end device mode.

romasku, 2025-03-13:
Thank you for a detailed report. Unfortunately, I do not have much time currently, but I'll definitely go through your list later.

CRCinAU, 2025-03-14:
Just arrived here from the above mentioned github project.

I configured this up in HA using ZHA integration as follows:

```yaml
zha:
  zigpy_config:
    ota:
      extra_providers:
        - type: z2m
          url: https://raw.githubusercontent.com/romasku/tuya-zigbee-switch/refs/heads/main/zigbee2mqtt/ota/index_router.json
```

Restarted HA and eventually, HA prompted me to upgrade firmware on a couple of devices. I'm currently doing a MOES 1 Gang switch:

[image]

I'm curious that given the version number of `0xffffffff` in the new firmware, how would / will it know if there is an update later on?

Either way, once its up and running, I'll do some testing on this single gang switch before going to others.

CRCinAU, 2025-03-14:
Ok, I did get it going within ZHA - which is really nice.

The Tuya firmware has an entity for the backlight LEDs which I switch on and off using the following automation...

At 8pm, turn the backlight to 'opposite' - ie LED on when the light is OFF
```
alias: 20:00 Jobs - Light backlights on
description: ""
triggers:
  - trigger: time
    at: "20:00:00"
conditions: []
actions:
  - alias: Bedroom Light Backlight On
    action: zha.set_zigbee_cluster_attribute
    metadata: {}
    data:
      cluster_type: in
      ieee: a4:c1:38:f2:25:e5:32:77
      endpoint_id: 1
      attribute: 32769
      value: SwitchBackLight.Mode_2
      cluster_id: 6
    enabled: true
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 500
  - alias: Wardrobe Light Backlight On
    action: zha.set_zigbee_cluster_attribute
    metadata: {}
    data:
      cluster_type: in
      ieee: a4:c1:38:b3:05:7d:9e:a8
      endpoint_id: 1
      attribute: 32769
      value: SwitchBackLight.Mode_2
      cluster_id: 6
    enabled: true
mode: single
```

Then at 11pm, turn the backlight leds back into 'normal' mode so that it only lights up when the light is on:
```
alias: 23:00 Jobs - Light backlight off
description: ""
triggers:
  - trigger: time
    at: "23:00:00"
conditions: []
actions:
  - alias: Bedroom Light Backlight Off
    action: zha.set_zigbee_cluster_attribute
    metadata: {}
    data:
      cluster_type: in
      ieee: a4:c1:38:f2:25:e5:32:77
      endpoint_id: 1
      attribute: 32769
      value: SwitchBackLight.Mode_1
      cluster_id: 6
    enabled: true
  - delay:
      hours: 0
      minutes: 0
      seconds: 0
      milliseconds: 500
  - alias: Wardrobe Light Backlight Off
    action: zha.set_zigbee_cluster_attribute
    metadata: {}
    data:
      cluster_type: in
      ieee: a4:c1:38:b3:05:7d:9e:a8
      endpoint_id: 1
      attribute: 32769
      value: SwitchBackLight.Mode_1
      cluster_id: 6
    enabled: true
mode: single
```

How would we be able to set this via ZHA? Would it be done via a new quirk?

ZHA currently shows the device as:
[image]

The original firmware shows as:
[image]

The firmware from here shows the following signature:
```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 74,
    "maximum_incoming_transfer_size": 404,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 404,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0000",
        "0x0007",
        "0x0012"
      ],
      "output_clusters": [
        "0x0006",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "Moes-1-gang",
  "model": "Moes-1-gang",
  "class": "zigpy.device.Device"
}
```

What would we need to add to bring up similar backlight control to the stock firmware?

What is needed to add group support so we can RF-link different light switches instead of via HA automation?

What would be needed to change the type of control from a switch to a light? This would simplify things quite a bit.

benbender, 2025-03-14:
@CRCinAU 

I did not try the new fw yet, but for your question regarding:

> What would be needed to change the type of control from a switch to a light? This would simplify things quite a bit.

see https://www.home-assistant.io/integrations/switch_as_x/. This is a normal feature of home-assistant.

MnM001, 2025-03-14:
@CRCinAU these are my entices in HA (1 Button switch)

[image]

A better view of he same entities in zigbee2mqtt:

[image]

The I control the lights in detached mode is (looking at the ZigBee view) is to make the "Relay relay indicator mode" to "manual". After this the "Realy relay indicator" can be manually controlled from zigbee2mqtt or HA.

[image]

There seems to be a lot more entities in zigbee2mqtt than ZHA.

romasku, 2025-03-14:
@MnM001 

Thank you for the report once again; it will definitely help me to improve the firmware!

> A. End Device Mode
> 1. "Relay relay indicator" - it does not remember the state/position of the button after a power cut.

Sure, I will update the firmware to store it in flash memory so that the LED state is not lost after a power cut.

> 1 button switch is hooked up to 2 downlights (9w each, 18w in total) in detached mode. If I turn lights off in HA, then the switch goes offline and nothing works on the switch itself. However, the lights are still on Wi-Fi and can be controlled from HA. Maybe this is something (the switch going offline when lights are turned off in HA) that can be fixed in firmware? I am not sure.

It seems like the device isn't getting enough power. Does it work with the stock firmware in such a configuration? Currently, the power management logic for End Devices is basic. I plan to improve it to decrease the power usage of the End Device firmware. But it might never work with turned-off smart bulbs, as they allow only a tiny current to pass through them, and the switch has to power itself from this tiny current. The best indicator of whether it is possible will be the stock firmware, as, despite its drawbacks, it has very good power efficiency.

> B. Router Mode - assume the router mode will have same issues as the End Device Mode + its own issues

> 1. LED state after a power cut - it is not remembered.
> 2. "Relay relay indicator" - it does not remember the state/position of the button after a power cut.

What is the difference here between 1 and 2?

> After a power cut the device is not immediately shown on the ZigBee network until the ZigBee network polls it. In end device mode the device "advertises" itself that it is back and is shown on the ZigBee network immediately. Will be good to make it work as it works in end device mode.

Can you please elaborate a bit here? What exactly do you mean by "shown on the ZigBee network"? Is it about the "Availability" column in z2m? In any case, I think I can add one extra state report after power on that should probably fix this issue.

> C. General cosmetic suggestions

A zigbee2mqtt is a bit limiting in what can be done, but I'll definitely need to make names consistent and reorder entries to group them properly. It’s just a low-priority task that I always leave for later, sorry :)

> I will like to link some switch buttons so multiple buttons on different switches can control the same light. I have tried to do this with bindings but I could not find a way to make it work. I have also tried with groups and same result. I am not sure if I am doing something wrong when doing the bindings or is something in the firmware to stop me form doing this. If you could have a look and advise that will be great.

It is possible via binding, but I need to add better documentation on ZigBee clusters/endpoints usage. For now, I can say that for an N-gang switch, endpoints 1-N are buttons, and N+1 - 2N are relays. So, for example, if you want switch 2 of one device to control relay 3 of another 3-gang device, you need to bind `OnOff` on endpoint 2 on the first device to `OnOff` on endpoint 6 (3 + 3) on the second device. As I said, I'll provide better documentation at some point with clearer explanations.

> In detached mode - is it possible to add an arbitrary ON/OFF state for the button and the led? Will make it so much easier in HA automations.

Not sure I'm getting your idea here. The LED state is already exposed, and I'm not sure what the ON/OFF state for the button in detached mode should be. In detached mode, some other smart device is controlled, and its state should be used as the source of truth. The switch cannot track changes to that device that were triggered from HA, so it will become out of sync. What you can do is create a "Boolean Input" helper in HA, set up automation that updates this helper based on device events, and then use this helper as if it were the device's internal state. But this helper can also become out of sync with the real device being controlled.

As I said, maybe I'm missing your goal here, so please elaborate if you still think there is something that can be added on the firmware side.

> Is there a way to display the original ZigBee pictures in GUI rather than the Tuya red icon, without using external converters for each device?

To do so, I have to use the same device name as the original one. I'll see if it breaks anything, and if not, I'll update it to re-enable pictures.

> In the change log for the latest firmware on the main GitHub page will be good to also list if we need to re-download the convertor files. Now with the OTA update happening direct from GitHub it is easy to forget that these files might need to be updated every now and then.

I'll try to add such a note to the changelog going forward.

MnM001, 2025-03-15:
Hi @romasku thankyou for your reply!! I will try and clarify some of the questions below:

>

Unfortunately I don't have the answer as the stock firmware does not allow for detached mode. With the original firmware, the device works fine but that is because when the light is turned off it actually cuts the power to the light. With your firmware in detached mode it cut the emitting light but leaves the device powered and on the wifi. I suspect that it is like you said - not enough power. If your firmware could improve power management and help this issue that will be great.

> What is the difference here between 1 and 2?

1 = Relay relay indicator mode
2 = Relay relay indicator
1 has been fixed in End Device Mode and 2 is something that you will fix ( your reply to A.1 above)
Above, these are the same but in Router Mode. Mentioning them as I have no idea how the code works and if you have to fix it in both End Device and Router Mode or the code is shared.

> Can you please elaborate a bit here? What exactly do you mean by "shown on the ZigBee network"? Is it about the "Availability" column in z2m? In any case, I think I can add one extra state report [...]

Ok so in z2m one device goes offline. It is shown in z2m as offline. As it is offline it is not available in HA either. See picture below as an example:

[image]

A device comes "alive" again (lets assume power was cut and the device went offline and then power is restored). If this device has End Device firmware on it it is shown as Online in z2m immediately (and therefore is available in HA immediately too). So I think it advertises itself on the network that it is back.
However, if the device has Router firmware on it, when it comes back after a power cut it is still shown as offline in z2m (and in HA by default) until z2m polls it  (in my configuration is set at 15m). 

[image]

So the device with the Router firmware, while it might be back on the z2m network and actually works, it is shown as offline until it is polled, while an end device is shown as online immediately after a power cut. If the router firmware can be made to work (this particular section) work as the end device then devices will be shown as online in z2m immediately  and be usable immediately.

> Not sure I'm getting your idea here. The LED state is already exposed, and I'm not sure what the ON/OFF state for the button in detached mode should be. In detached mode, some other smart device is [...]

> Not sure I'm getting your idea here. The LED state is already exposed, and I'm not sure what the ON/OFF state for the button in detached mode should be. In detached mode, some other smart device is [...]

I think I understand the out of sync issues better now than when I wrote the above. Best not to change anything and leave as it. So is a non issue.

Hope the above answers help. If not let me know and I will try to explain them differently.

Thanks again for your great work!!!!

CRCinAU, 2025-03-15:
What commands are sent to change the backlight operation mode etc?

I've been trying to figure out how to make a Zigbee Quirk for ZHA to expose these settings... The stock firmware has:

[image]

As this is a Tuya Quirk, what commands would be sent to your firmware to do the same thing?

Would it be useful to make it seem Tuya Compatible and implement the same command structure?

romasku, 2025-03-15:
@CRCinAU 

I added (a bit unpolished) custom quirk for ZHA to enable full support of the device. See [guide here](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/zha_ota.md). It adds LED control as a normal HA entity, so it should be easy to control it now. Hope it will work for you. 

@MnM001 

I've released 1.0.13 with some of the issues you described above. Please note that the new converter will change the names of some properties, as I standardized left/right vs 1/2, so it may break some automation you may have.

CRCinAU, 2025-03-16:
> [@CRCinAU](https://github.com/CRCinAU)
>
> I added (a bit unpolished) custom quirk for ZHA to enable full support of the device. See [guide [...]

Yep - I see a whole lot more options now. I'll play around with it and add some suggestions when I figure it out - mostly the names of what things are called to make them more intuitive.

I do have one suggestion though - to put the device in pairing mode current takes 5 x quick presses. The problem with that is that the load connected will turn on and off 2.5 times.

Would it be better to make this function be a 10 second hold of any button instead? That way, we're not turning on/off connected devices in rapid succession.

MnM001, 2025-03-16:
@romasku - what can I say? Thank you so much for incorporating some of the changes so quick! This firmware is just getting better and better and better.

One thing - you did say you have release v1.0.13 but I can only see 1.0.12 on the main page. Did you mean 1.0.12?

MnM001, 2025-03-16:
HI @romasku - one idea that I want to run past you. Your firmware has the capability to work as stock standard Tuya firmware - in the sense that it operates fine when you press the button on the switch and it handles the LED and the downlight on its own with the need for HA or automations in there. And this is very good.

However, as soon as I select Detached Mode I become 100% reliable on HA and MQTT. If ever these 2 are offline then I have no lights in my house until the issue with HA/MQTT is sorted. In order to minimize this impact I was thinking if your firmware can have some emergency mode in it. For example in Detached Mode if Press and hold the button for 30s (as an example) then the firmware configures itself go back into Raise mode and the Relay Indicator goes back into same or opposite mode.
I am not sure if this is possible to do but if possible to do but if it can be done will massively improved the detached mode and risks.

Thanks again.

CRCinAU, 2025-03-16:
As a side note as well, you can force HA's ZHA to check for a firmware update via going to:

Developer Tools -> Actions -> Go To YAML mode and paste:
```yaml
service: zha.issue_zigbee_cluster_command
data: 
  ieee: <ieee address>
  endpoint_id: 1
  cluster_type: out
  command_type: client
  cluster_id: 25
  command: 0
  args:
    - 0
    - 100
```

Replace the <ieee address> with the address of the device obtained from within the Device Info.

MnM001, 2025-03-16:
Hi @romasku  - Updated to v1.0.12 - all seems to be OK except this:

[image]

It doesn't show the device confg anymore.

CRCinAU, 2025-03-16:
> However, as soon as I select Detached Mode I become 100% reliant on HA and MQTT. If ever these 2 are offline then I have no lights in my house until the issues with HA/MQTT are sorted.

I kinda thing detached mode is just kinda bad. You can 'fake' binding groups with a HA automation as follows:
```yaml
alias: Living Room Light Group
description: ""
triggers:
  - entity_id:
      - light.living_room_light_switch_light_2
      - light.kitchen_light_switch_switch
    trigger: state
conditions:
  - condition: template
    value_template: "{{ continue and to_state in state_list and targets | count > 0 }}"
    enabled: true
actions:
  - target:
      entity_id: "{{ targets }}"
    action: light.turn_{{ to_state }}
variables:
  entities:
    - light.living_room_light_switch_light_2
    - light.kitchen_light_switch_switch
  state_list:
    - "on"
    - "off"
  continue: >
    {{ trigger | default(none) is not none and trigger.to_state is defined and
    trigger.from_state is defined }}
  to_state: |
    {{ trigger.to_state.state | default(none) if continue else none }}
  targets: >
    {{ expand(entities) | rejectattr('entity_id', 'eq', trigger.entity_id) |
    rejectattr('state', 'eq', to_state) | map(attribute='entity_id') | list if
    continue else [] }}
```

This will mirror commands in the group to allow multiple lights to control the same physical light. You must edit it in YAML mode every time - but I wonder if this is what you're after?

MnM001, 2025-03-16:
@CRCinAU - not really. In detached mode the switch doesn't kill the power to the light at all anymore. So the light is always on Wi-Fi.
There is no on/off concept anymore in detached mode.  Its all controlled by automations in HA. Click the button and nothing happens if there is no HA/MQTT/Automation.

For me this is the biggest value out of this custom firmware. Lights are always on Wi-Fi so when toggled on it can immediately have effects applied etc

However the danger is that in detached mode when HA/MQTT are gone no changes can be done to the device anymore. So no lights till everything is sorted. An emergency mode helps with this.

CRCinAU, 2025-03-16:
> In detached mode the switch doesn't kill the power to the light at all anymore. So the light is always on Wi-Fi.
> There is no on/off concept anymore in detached mode. Its all controlled by automations in HA. Click the button and nothing happens if there is no HA/MQTT/Automation.

This has got to be the worst way to do home automation. Maybe you're looking at doing these things the wrong way.

MnM001, 2025-03-16:
> This has got to be the worst way to do home automation. Maybe you're looking at doing these things the wrong way.
> 
> But you're controlling wifi lights with zigbee switches. Ew...

Well this is the way I am doing things. And if you things differently I am not going to criticize you or anything like that simply because that's not the way I am doing things. I understand the different people have different ways of using it and I will respect that.

Thanks to detached mode I can have scenes, colors etc applied immediately.

MnM001, 2025-03-16:
> But you're controlling wifi lights with zigbee switches. Ew... Maybe you'd be better off hard-wiring in the light socket, and then using the zigbee switch with no load connected. You'd still have no lights if _anything_ failed, which is still bad design, but seems more sane...

No neutral in my light switches. So I cant do that.

CRCinAU, 2025-03-16:
> No neutral in my light switches. So I cant do that. Unless hardwiring means to join the cables for an always on light?

Ouch. I tend to think of home automation as a safety thing when it comes to lights. Lucky though, I have a neutral at every switch - so the light switches are wired with both active / live and neutral.

I strongly suggest that you try and think of a way that you at least could have some lighting if needed in an emergency - as these things really are safety related to anyone inside your home. Think about scenarios like if your home automation has an issue, but emergency workers need to get into your house and be able to see.

MnM001, 2025-03-16:
> I strongly suggest that you try and think of a way that you at least could have some lighting if needed in an emergency - as these things really are safety related to anyone inside your home. Think [...]

Hence why I started the discussion earlier about an emergency fall back mode :)

CRCinAU, 2025-03-16:
> Hence why I started the discussion earlier about an emergency fall back mode :)

Yeah - but nobody else will understand how that works - so at least in my view, wouldn't satisfy safety requirements that a light work like a light unless power is gone.

MnM001, 2025-03-16:
I guess I can have some lights that work in non detached mode. But will still be good to have an emergency fallback in the firmware.

jzxandy, 2025-03-16:
I just signed up so I could post here and thank @romasku for all the great work on the custom firmware with these switches.

I flashed several of my moe's switches earlier this week and have been running a bunch of lights in detached mode. My dumb lights/fans/etc are running in raised mode and behave just like they do on the factory firmware. Having the flexibility to pick and choose modes per button/relay is super handy. Having the switches become super responsive (to the point where i can press multiple buttons at once without issue) is also awesome.

All of these features are a gamechanger for no-neutral installations - thank you again for everything so far.

Really hope this can be done with the 4-button switch in due course!

benbender, 2025-03-16:
Before I start playing around with this, can someone give a brief overview about what's currently missing in this firmware as it is still labeled as "in progress"?

And I would like to second the big thanks to @romasku for putting that work into improving the laking vendor-firmware. It really is a gamechanger as it should enable many scenarios that weren't possible with the stock-fw before! <3

romasku, 2025-03-16:
@benbender
I'd say all the features are there; I just wanted some confirmation from someone in this thread that the device works properly. I'll probably update the table entry today. And you're welcome—I'm really happy to help you all!

CRCinAU, 2025-03-16:
@romasku - I have noticed one issue with the ZHA integration. After updating firmware, I need to delete the old device, then add the device again. This seems to mostly work, but I get stuck at the following screen:

[image]

If I restart Home Assistant, then the device ends up being added ok, but if I leave it at the above screen, the device never finishes adding.

I'm wondering if there's some kind of acknowledgement missing that HA is waiting for, but is never sent?

MnM001, 2025-03-16:
> I just wanted some confirmation from someone in this thread that the device works properly.

@romasku - I can give you that conformation. Apart from me there are other 2 family members that have these switches installed in their own houses. All of us have been testing and using some switches in our houses with your firmware. One family member has been using around 10 of these switches without any issues.

@benbender  We (me and the other family members) can confirm that as far as we can see all the stock functionality is there plus a lot more functionality of course.
Whatever is left is just correcting little issues and adding more features if possible.

DeadSomeOne, 2025-03-16:
Hi,

> I think the only 2 things that need to be addressed (from a z2m and HA point of view but not ZHA) are:
>
> This
> 
> [image]

I don't think you should worry about device config. IMO it's more relevant for new testing devices, porting or debugging process. So as long as the device works, it fine if it shows empty. also clicking the refresh button should make it appear

> "Would it be better to make this function be a 10 second hold of any button instead? That way, we're not turning on/off connected devices in rapid succession." (this about a device leaving or joining a network)

I would like this as well, especially since I'm using zigbee bulbs, so repairing the switch makes my bulbs go into pairing mode as well :D

MnM001, 2025-03-16:
> also clicking the refresh button should make it appear

Unfortunately this does not work It was the first thing I have tried.

romasku, 2025-03-16:
>  2025-03-16 23:07:03z2m: No converter available for 'device_config_switch_1' ("")

Hi, there was a small bug in the last `switch_custom.js`, please re-download converter file to fix this issue.

MnM001, 2025-03-17:
Hi @romasku - just tried it and the issue is the same.

claudio-n85, 2025-03-28:
Hi @romasku, thank you very much for the work you have put in to the firmware. Have been testing on the single and 2 button and working great. Looking forward to the firmware for the 4 button switches. Any progress with it?

jzxandy, 2025-04-09:
@romasku Just wanted to chime in and say my light switches have been flawless since I first flashed them with your custom firmware. Thanks again!

How are things going with 4-gang switch development?

romasku, 2025-04-09:
Hi all, sorry for the late reply. I currently have too much work at my main workplace, so I don’t have much time to work on this project. Support for the 4-gang is still planned, but it will have to wait until I have a bit more time.

MnM001, 2025-04-21:
@romasku - Using a few light switches with the custom firmware you have done for us. They work great!!

I hope you will have the time to figure out the 4 button switch soon too. I have a few 4 button switches that I would love to have them work in detached mode.

romasku, 2025-04-21:
Hi,
Thank you for waiting. Support for the four-button device is currently a work in progress. I’ve acquired a dedicated programmer compatible with the SoC and successfully extracted the stock firmware. I’m now in the process of developing and flashing custom code. While there is still significant work to be done, I wanted to share this initial progress update.

MnM001, 2025-04-22:
Thanks for the update. Sounds like you are making good progress already! Please let me know if I can help with anything.

[13 more replies]