romasku/tuya-zigbee-switch #15: Physical switch press action and state not updated in z2m

shatpav, 2025-02-20:
Hi @romasku!

Maybe you faced with something like this and could give an advice?

Observation(same issue on 5 devices). No exact/special circumstances, just normal working environment. 

Right after paring(or reconfiguration) everything is updated as expected, but some time later(~ 24 hours, may be less) physical switch actions/state not updates status in z2m.  

For example:

normal behaviour
toggle ON physical switch 1, you see in z2m action press/long_press and state ON
toggle OFF physical switch 1, you see in z2m action released and state OFF

some time later 
toggle ON physical switch 1, you don't see updates in z2m action still "released" and state still "OFF" 

physical bulb light switched as expected, force update(press "Refresh" button on Exposes tab) switching action/state to the right value. 
[image]
[image]
[image]

Device: GIRIER _TZ3000_jl7qyupf 2 gang, no neutral, https://www.aliexpress.com/item/1005006084744526.html

Huge thanks in advance!

shatpav, 2025-02-21:
Short update. As of now no one of my 5 devices faced with the issue(all devices has been reconfigured yesterday). All devices are updating press action and state triggered by physical switch as expected. I'll investigate the issue further. It is some probability, that previous cases related to one of my zigbee router which I removed from the network a couple of days before because of malfunction. I hope it was the reason. Will look.

Thanks.

romasku, 2025-02-21:
Hello,

As far as I know, Zigbee report packets are not acknowledged: the device does not check for confirmation that the report was received. Therefore, it is possible that the devices were children of the malfunctioning router and were attempting to report by sending packets to it. However, they would rejoin the network after receiving any command, not only after the reconfigure reporting command.

So, did you experience the following events in this order?

1. Physical press, no report in Z2M
2. Manual refresh or turn on via Home Assistant
3. Physical press, no report in Z2M

If so, the devices had a chance to reconnect to another router after step 2, so this issue requires further investigation.

shatpav, 2025-02-21:
> So, did you experience the following events in this order?
>
> 1. Physical press, no report in Z2M
> 2. Manual refresh or turn on via Home Assistant
> 3. Physical press, no report in Z2M
>
> If so, the [...]

Hi! Exactly, yesterday all devices works without malfunction router(it has been completely removed from the network a day before). All devices works as expected via remote zigbee control, but no updates on physical switches actions. And this behaviour I observed during last week several times on different devices. Each time  when I discovered it, reconfiguration solved the issue, but a day-two later issue back again in some kind of random order. There was a lot of overall network reconfigurations, that is why I just put all the issues on the network instability and don't tried to look on this carefully and understand what is the trigger. But last couple of days the only change I made is removal of one bad router. Unfortunately, Issue repeats again. Maybe this exact router was the issue all this time. I'm not planning to do any network changes in nearest future and will investigate the issue next couple of days more precisely. Based on previous experience I'm sure it should happen at least on one device during couple of days. Will keep you updated. Thank you once again!

shatpav, 2025-02-26:
Everything looks good. Most probably it was a network issues. Thanks!

closed 2025-02-26 by shatpav