romasku/tuya-zigbee-switch #16: AVATTO _TZ3000_ljhbw1c9

IMitsan, 2025-02-25:
Hi,

Thank you for the great job!

This is more of a report than an issue: After flashing the 2-gang AVATTO _TZ3000_ljhbw1c9 with the ROUTER firmware version, it stopped working. The LEDs do not glow, and the Z2M is not connecting. I probably did something wrong. 
If you need the device for further development, I can transfer it to you (I live in Kyiv).

romasku, 2025-02-26:
Hi,

Thanks for reporting this. I mistakenly added this device to the OTA list, as it shouldn't be there since it isn't supported at the moment.

If you send it to me, I can likely restore it, add proper support, and then send it back to you.

Please reach out to me on Telegram: @romasku.

romasku, 2025-03-02:
I have sent your relay back to you. When you receive it, please check if it works, and if everything is okay, I'll close this issue.

IMitsan, 2025-03-04:
Thank you so much for such great support. I received the relay, and everything works perfectly!
Once again, thank you for the excellent work!

bylazarov, 2025-04-19:
Any tips on why I can't get the device to show up on the OTA tab? I have the following in my configuration.yaml for zigbee2mqtt:

external_converters:
  - switch_custom.js
  - tuya_with_ota.js
ota:
  zigbee_ota_override_index_location: >-
    https://raw.githubusercontent.com/romasku/tuya-zigbee-switch/refs/heads/main/zigbee2mqtt/ota/index_end_device.json

I tried with local index_end_device.json file without any change. The converters are both in the root dir as well as in a subdir in my zigbee2mqtt data folder.

I'm pretty sure I have the same device:

[image]

romasku, 2025-04-21:
Hi,

> can't get the device to show up on the OTA tab

This means the `tuya_with_ota.js` converter isn’t being applied. I see you’re using the configuration for Z2M version 1.x—are you sure you’re on that version? If not, please check the [guide for v2.x](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/ota_flash.md).

bylazarov, 2025-04-21:
@romasku 

Thanks! It seems I had the extensions part left in my config file from the older version - I deleted it and now its fixed! Updated and no longer have the delay when switching both channels! Thanks a lot!

romasku, 2025-04-21:
Glad to hear that. This also confirms that the initial OTA problem has been resolved, so I'm closing this issue.

closed 2025-04-21 by romasku