romasku/tuya-zigbee-switch #18: AUBESS  _TZ3000_46t1rvdu

Alex9premium, 2025-03-01:
First of all thank you for this project, I've got one connected to a doorbell so relay is not used and detached mode would be perfect here.

This is recognized as WHD02 in zigbee2mqtt

Which firmware is the most suitable to testing?

[image]
[image]
[image]

romasku, 2025-03-02:
Hi,  

Thank you for your interest in the project. Using your photos and some documentation on the WiFi version of this device I found online, I was able to come up with a possible pinout for your device. However, I can't be sure that it will work, as I can't test it myself. If you're willing to try, just follow the standard OTA instructions. If OTA doesn't start, look for error logs like this:  

```
[2025-01-18 16:37:38] debug: 	zh:controller: Received payload: clusterID=25, address=44312, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=160, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":79,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4417,"imageType":43521,"fileVersion":16986117},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
```  

If you find such logs, please post them here, and I'll update the firmware with the proper `imageType` to make it work. Also, if something goes wrong, it's almost always possible to recover the device by flashing the firmware via wire, but that will require some soldering and a UART flasher.

Alex9premium, 2025-03-02:
The new 2.1.3 Zigbee2mqtt update changed the structure of definitions to return object instead

I had to change the first line to make external converter work again

```
tuya_with_ota.js

const { definitions: tuyaDefinitions } = require("zigbee-herdsman-converters/devices/tuya");
```

Alex9premium, 2025-03-02:
Accidentally closed it

Here are the logs as it says no update available
```
[2025-03-02 19:10:07] debug: 	zh:controller: Received payload: clusterID=25, address=60106, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=83, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":218,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":70},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}
```

romasku, 2025-03-02:
Update image type. Please re-download index.json and try again.

Alex9premium, 2025-03-02:
I've replaced the index_router.json but it still says no update available. manufacturerCode is set to 4407 instead of 4417 and even after correcting it still doesn't work.

romasku, 2025-03-02:
Yes, sorry I missed that both manufacturerCode and imageType where wrong. I've rebuilt it one more time, so please redownload and try.

Alex9premium, 2025-03-02:
Update was successful. It connected after a few resets. I can hear it toggle on/off so that works.

Thank you very much for your help!!

I'll wait for end_device version so I can switch to it.

romasku, 2025-03-06:
Hi,

I've built an EndDevice version for your device. I didn't create it before, as I thought there was no need for it for the Neutral-connected module. Is it for power savings?

andrei-lazarov, 2025-09-26:
Sorry for reviving this thread, I am filling the database with more info about each device: #155   

@Alex9premium Can you tell me if this device needs Neutral or not? (without / optional / required)  
I assumed it's required, but reading you asked for end_device firmware confused me.

andrei-lazarov, 2025-09-28:
@Alex9premium Thanks for clarifying!  
Normally we don't build end_device firmware for devices with Neutral. But I was already considering this, even if it doubles the total build time.  
There are some reports of devices freezing due to routing errors, so end_device would also help there. This plus your usecase of sometimes offline device (low signal or even just powered off) should be enough to justify it.

romasku, 2025-11-16:
Device support added, discussion about end device version for all devices moved to https://github.com/romasku/tuya-zigbee-switch/issues/193. Thank you!

closed 2025-11-16 by romasku