romasku/tuya-zigbee-switch #19: Device support for _TZ3000_zmlunnhy

Ledarium, 2025-03-03:
I have two of these devices `_TZ3000_zmlunnhy`. The one with stock firmware is identified by z2m as follows:

[image]

For the second one I have flashed custom firmware for TS0012.
- I've tried to do OTA update using `index_router.json` with updated definitions for `_TZ3000_zmlunnhy`. It didn't work (device was undiscoverable after updating) - maybe I was doing something wrong or it requires different zigbee firmware? Not sure. If it's possible to revert to stock firmware somehow (maybe dump it from the original one and reflash it), I would be able to test the OTA update.
- I've flashed TS0012.bin using the guide here: https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/flashing_via_wire.md
- I had to connect A0 to 3v3 - in stock settings for this firmware this pin is configured as reset button, but as it is not connected on this device, I suspect that it was floating and triggering reset. After modifying the config string, the connection could be removed.
- The main issue right now is that the relays on the device are connected with two pins for each relay: B5 turns relay 1 on, and C4 turns relay 1 off. Same for relay 2 - C2 turns it on, D4 turns off. It kinda works with this config `Tuya-TS0012-custom;TS0012-custom;LD7;SC3U;SD2U;RC2;RD4`, so that switch 1 turns it on, and switch 2 turns it off.

I've tried to setup two additional relay definitions adding `RB5;RC4;` to config string, but after this config is pushed, device stops responding with the following symptom: 
```
z2m: Publish 'set' 'device_config' to '0xa4c1381c52c4c5e1' failed: 'Error: ZCL command 0xa4c1381c52c4c5e1/1 genBasic.write({"65280":{"value":"Tuya-TS0012-custom;TS0012-custom;LD7;SC3U;SD2U;RC2;RD4;RB5;RC4;","type":68}}, {"timeout":30000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '62307'.)'
```

If I understand the flow correctly, to make it work correctly there is a need for a custom converter. Could you help with that?

Device photos, if needed, are attached. I wasn't able to remove the relay part as there was no need for it, but if it would be needed for debugging, I can do that and help with testing.

[image]
[image]
[image]
[image]

Ledarium, 2025-03-03:
One additional minor thing - I wasn't able to figure the second led connection, I'm not sure if multiple leds are supported. If it is possible to show the state of the switch using leds, please let me know how to configure it. Right now the led (D7) is lit up always and turns off briefly after the switch is pressed.

romasku, 2025-03-03:
Hi,  

It's interesting that the relay is controlled using two pins. I'll need to add support for it directly in the firmware to avoid hacks in the converter. Could you provide more details on how it works? For example, what is the relay state for all four combinations (LOW + LOW, LOW + HIGH, HIGH + LOW, HIGH + HIGH)?  

The firmware supports indicator LEDs using the `I` prefix, for example: `ID7;ID4;` to set indicator LED pins `D7` and `D4`. I only added this feature over the weekend and haven't updated the documentation yet.

Ledarium, 2025-03-03:
With the following relay config `RC2;RD4;` the relay is turned ON when I set the switch 1 (C2) to enabled, no matter the state of the switch 2 (D4). Following changes to the switch 1 do not change the state of relay. 

When the relay is ON, switch 1 is disabled and switch 2 is set to enabled, the relay is turned OFF. If switch 1 is enabled, the relay stays ON. Again, following changes to switch 2 do not change the state until switch 1 is enabled.

To turn relay ON:
[image]
To turn relay OFF:
[image]

If both switches are enabled, after some time (couple of seconds) device stops responding to updates and I have to reset the power. Looks like this situation shouldn't happen, and as device is connected without neutral wire it doesn't get enough power and either wireless or the whole controller shuts down, or some other weird stuff happens.

Ledarium, 2025-03-03:
That's correct, just the pins here are C2 and D4, not C4.

romasku, 2025-03-10:
Sorry, I haven't had time to work on supporting this two-pin relay yet. I'll ping you when it's ready.

romasku, 2025-03-10:
Update: I was able to add support for this relay faster than expected; it was easy to implement. I've prepared builds for your device:  
https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/ZEMISMART_2_GANG

The pinout is: `SC3U;SD2U;RC2D4;RB5C2;ID7;`. However, it is missing the second LED for the second switch. If you can, please trace it so I can add it to the pinout and complete the porting of the firmware for this switch.

Ledarium, 2025-03-11:
Thank you, could you advice how to update the device currently flashed with TS0012 firmware? Latest `index-router.json` is downloaded and when trying to do OTA update I get this: `z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c13817d77391f5', payload '{"device_config_1":"Tuya-TS0012-custom;Tuya-TS0012-custom;ID7;SC3U;SD2U;RC2;RD4;","last_seen":"2025-03-11T22:27:52.478Z","linkquality":156,"power_on_behavior_left":"off","power_on_behavior_right":"off","state_left":"OFF","state_right":"OFF","switch_1_action_mode_1":"toggle","switch_1_long_press_duration_1":null,"switch_1_mode_1":"toggle","switch_1_press_action_1":"released","switch_1_relay_index_1":"relay_1","switch_1_relay_mode_1":"raise","switch_2_action_mode_2":"toggle","switch_2_long_press_duration_2":null,"switch_2_mode_2":"toggle","switch_2_press_action_2":"released","switch_2_relay_index_2":"relay_2","switch_2_relay_mode_2":"raise","update":{"installed_version":16986119,"latest_version":16986119,"state":"idle"}}'`

romasku, 2025-03-11:
Hmm, are you sure you are using the latest `index_router.json` file? The latest version should be `16986122`. Also, I suggest using a direct link:

```
ota:
  zigbee_ota_override_index_location: >-
    https://raw.githubusercontent.com/romasku/tuya-zigbee-switch/refs/heads/main/zigbee2mqtt/ota/index_router.json
```

To reflash the device flashed as TS0012, set config to `Zemi-2-gang;Zemi-2-gang;ID7;SC3U;SD2U;RC2;RD4;i43536;`. This will rename the device to `Zemi-2-gang` and change the image type to `43536` so that it will use the OTA file for your device.

Ledarium, 2025-03-11:
Okay, I wasn't able to do OTA even with the config you mentioned and index location set to url you provided. I have flashed the file https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/ZEMISMART_2_GANG/tlc_switch.bin using UART, but now after resetting the device and opening zigbee network I'm getting the following message: `z2m: Request 'zigbee2mqtt/bridge/request/device/interview' failed with error: 'interview of '0xa4c1384fd77257c5' (0xa4c1384fd77257c5) failed: Error: Interview failed because can not get active endpoints ('0xa4c1384fd77257c5')'`. The device appears in z2m UI but has warning that the interview was not completed: 

[image]

Ledarium, 2025-03-11:
Also, the second (left) indicator led is on pin `B7`

Ledarium, 2025-03-12:
After a couple of resets I was able to connect the device to the network. Looks like the config was incorrect and might have  - relays have the same pin shared `config_str: Zemi-2-gang;Zemi-2-gang-ED;SC3U;SD2U;RC2D4;RB5C2;ID7;`. So the correct config string should be `Zemi-2-gang;Zemi-2-gang;SC3U;SD2U;IB7;ID7;RB5C4;RC2D4;`. Using this string relays work as intended

romasku, 2025-03-12:
Ouch, sorry, looks like I made a typo when preparing the config for your device. I've rebuilt the firmware with proper configuration and re-generated `switch_custom.js` to enable control of both LEDs.

> defice frequently disconnects

What exactly happens?

Ledarium, 2025-03-17:
Sorry for that, but I can't continue testing because after the last try to flash firmware seems like I've killed the controller. The second switch that I have is important to have in working condition and I'm not sure that it will work as stable as the original firmware.

The issue that I was having with an incorrect config string is that as soon as it is pushed to a device, it disconnects from the network with the message "Unable to deliver to (device address)". I had heard faint clicks from the device - maybe it had been trying to trigger relays. The same thing happened previously when both relay pins were turned on.