romasku/tuya-zigbee-switch #20: Ota auto update

grzegorzlabudda, 2025-03-06:
Hello
is it possible to automatically update ota without re-uploading files to the ha server? the idea is for z2m to know that there is an update without going to github and downloading new files.

romasku, 2025-03-06:
Hi,

Good point. I've discovered that it is possible to use links inside z2m's `configuration.yaml` :

```yaml
ota:
  zigbee_ota_override_index_location: >-
    https://raw.githubusercontent.com/romasku/tuya-zigbee-switch/refs/heads/main/zigbee2mqtt/ota/index_router.json
```

Also, I've updated the docs to include this method. 

It still may require downloading a new converter if it was updated, but it should happen rarely.

grzegorzlabudda, 2025-03-11:
Is it possible to set both links to update both types of devices?

alan3664, 2025-10-19:
I don't think it is possible to update both types of devices and keep the router/end device operation mode with the current json files. 
As mentioned in the updating section, with the index_router.json  "Both L and L+N switches get Router FW" 
whereas with the index_end_device.json  "L-only switches get EndDevice FW &  L+N switches do not get anything".
We would need a  "index_mixed_switch.json" that points to router firmware for L+N Switches & enddevice for L switches.