romasku/tuya-zigbee-switch #22: Tuya TS0012 Firmware

fkb80, 2025-03-07:
Hi, 
thank you for your nice work.

I tried to update my Tuya switches at home but then realiced that I have a different type: _TZ3000_rbl8c85w
Could you support me to adapt the firmware for them? First I think I need to open them and check the chip type, correct?

Another question: I integrated the OTA Firmware update from you but when I click on "Check for update" I get this error:
z2m: Failed to check if update available for 'BR1-SW2-Bed-Right' (device.definition.ota.isUpdateAvailable is not a function)

I assume I did something wrong or missed something?

Thanks for you help!!

romasku, 2025-03-07:
Hi,

Thank you for your interest in the firmware. What version of z2m do you use? There are different versions of `tuya_with_ota.js` for z2m 1.x and z2m 2.x, and your error suggests that you might be using the wrong file.

Regarding your device, is [this](https://zigbee.blakadder.com/Zemismart_606EU-Z2.html) the correct picture of your device? Detailed photos of the internals would also be helpful. Please note that the porting process may result in a bricked device that can only be restored using a UART converter, so proceed with caution.

fkb80, 2025-03-07:
Hi and thanks for the fast response.

You are right, I used indeed the wrong version 😩 
Thanks for the hint!! 

[image]

This chip is used in the switches, they look like in the link you send me. 
I have 2 spare devices, I would first try with them to make sure the update is working

As I understood, your firmware is working with the Z2M modules, but my ones have the ZS3L used. So I don't even need to try with this version. Is my understanding correct?

Thank you!

romasku, 2025-03-07:
The ZS3L is not supported at the moment, as it is not a Telink SoC but rather a Silicon Labs one. I do have plans to try building firmware for it, but that will take time.

fkb80, 2025-03-07:
Sounds great! Let me know if I can support you with something for the new firmware. 
It's a great project! The Tuya firmware is not really good. 
I am struggling with the switch delay and from time to time after a power outage, some switches hang up and need a hardware restart (really not responding to anything anymore, not working at all anymore)

diogo-madeira, 2025-06-25:
> The ZS3L is not supported at the moment, as it is not a Telink SoC but rather a Silicon Labs one. I do have plans to try building firmware for it, but that will take time.

I have many (more than 10) ZS3L switch with the same problem... I planing to replace then if I can't find a good firmware for it.
How can I help you with this future plans? I don't have programing skill, but I have a lot of ZS3L devices to test....

andrei-lazarov, 2025-11-16:
@fkb80 @diogo-madeira 

ZS3L (Silabs) is now supported!
But unlike the Telink chips, it needs to be initially [flashed by wire](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/contribute/flashing_via_wire_silabs.md).
Once custom firmware is installed, OTA updates will work normally.

Let's keep this thread for the `_TZ3000_rbl8c85w`. Open new issues for other devices.
Please share some pictures with the whole PCB. I can see the traces are easily visible in the image above.

andrei-lazarov, 2025-11-22:
> I use the version without neutral

Thank you. This would be the listing then: [Zemismart KES-606US-Z](https://www.zemismart.com/products/kes-606us-z?VariantsId=14161)

Yes, we need the pinout from ZS3L to relays, buttons and LEDs. This should help:

[image]
[image]
[image]
<p>

However, after taking a closer look:
**⮡ We need to patch the firmware before we can continue.** 🙁 
Sorry for reviving the thread too soon.

I expect some issues with both of your devices @fkb80 and @diogo-madeira:
- bi-stable (latching) relays
  - Instead of constant power, they only need a pulse (on 2 pins: one on, one off)
  - Should be fixed soon: #196 
  Took very long because the main contributors didn't encounter these
- can't toggle multiple relays simultaneously
  - These Zemismart / Manhot switches seem to break when you press multiple buttons at once!
  - This is why the stock firmware is limited to toggling a single relay, even if you press multiple buttons. 
  Can you confirm this is the case? 
  - We must implement a small delay to prevent this (toggle them sequentially)
  - Both of these guys fried their switches: 
  #19
  #128

andrei-lazarov, 2025-12-12:
Romasku fixed the issues I mentioned earlier, so you guys should be able to flash once you obtain the pinout.

fkb80, 2025-12-12:
Hi together!

Very nice, already a big thanks. I got the flash adapter already, but currently on a business trip in China. From 20.12 I have holiday and would like to try the new firmware. 

Thank you!

andrei-lazarov, 2026-02-27:
Yes you define all buttons as switches S.
And you reset by 10x button press. 
If it needs a power-cycle, it sounds like a bug. I'll look into it once I flash a Silabs device.

Your change looks good. Thank you!
https://github.com/romasku/tuya-zigbee-switch/compare/main...FlorianKellnberger:tuya-zigbee-switch:main
If it's ready, just click 'create a pull request' on this link. 
The issue is you have 2 different accounts. I'll add the changes if you can't do it.