romasku/tuya-zigbee-switch #24: Help in porting Noname relay. Defined as TS000F (_TZ3000_skueekg3)

pablosmithru, 2025-03-08:
Module: ZT2S

Pin out:
Onboard led: C3 (GND)
Onboard button: D2 (GND On the board it is pulled up to VCC by resistor 203)
Relay: B4 (VCC On the board it is pulled up to GND by resistor 203)
Switch Pin: B5 (GND On the board it is pulled up to VCC by resistor 203)

Example:
Tuya-WHD02-custom;WHD02-custom;BD2u;LC3;SB5u;RB4;

I use firmware for WHD02 or TS0011.

https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/WHD02
https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/TS0011

When the pins are changed through the corresponding field, the relay begins to cyclically turn on and off. Help me set up the pins correctly or build a firmware with the pins already installed.

As I understand it, the ability to invert pins is needed, since the devices are assembled differently.

[image]
[image]
[image]

romasku, 2025-03-08:
Hi,

I'll dive deeper into your device later, but from what I see, it looks like you have the same "weird" ZT2S module as mine here: https://github.com/romasku/tuya-zigbee-switch/issues/6#issuecomment-2568045792. Its pinout is different from the pinout document on the Tuya site. Please try using the real pinout I provided in the comment.

pablosmithru, 2025-03-08:
Used an alternative pinout. With these settings, the relay started working. But after rebooting, the settings seem to reset.

Is it possible to build a firmware with these settings? I can't do it myself.

Tuya-WHD02-custom;WHD02-custom;BB1f;LB4;SD2f;RD3;

Pinout of fake ZT2S. From here: https://github.com/romasku/tuya-zigbee-switch/issues/6#issuecomment-2568045792

[image]

romasku, 2025-03-08:
I’ve built firmware for you: [https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/WHD02_2](https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/WHD02_2).  Please give it a try. 

Also, do you have the same device with the stock firmware? It's unfortunate that there are two devices with the same "Zigbee Manufacturer" but different pinouts. Perhaps the `manufacturer_id` and `image_type` values are different (these are the values used to match the OTA file). If you can provide me with these values, I can add proper OTA support for this device.

pablosmith, 2025-03-09:
> I’ve built firmware for you: https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/WHD02_2. Please give it a try.
>
> Also, do you have the same device with the stock firmware? It's unfortunate [...]

This is the data of the same relay

[image]

pablosmith, 2025-03-09:
how to put the relay into pairing mode? Do I need to hold or press 5 times? how fast do I press? press the reset button or the switch?

romasku, 2025-03-09:
Either press switch button 5+ times fast or hold on board button more 3+ seconds. On board led should start blinking.

pablosmith, 2025-03-09:
Tell me how to merge the original firmware? My test relay stopped connecting, although it is successfully flashed. I would like to merge the firmware from the same relay and flash the test one to check. I can also pour the firmware. I have two more types of cheap relays, but I do not want to touch them until I learn how to make a backup of the original firmware.

romasku, 2025-03-10:
Hello,  

I previously successfully dumped the firmware from the device using [this UART tool](https://github.com/Suxsem/Telink-Zigbee-serial-flash-tool/tree/main). However, it only worked for reading, not writing, for some reason. As a result, I had to use the web flasher to write to the device. Since then, I ordered the TB-03 and now use [this tool](https://github.com/pvvx/TLSRPGM), which works much more reliably.  

The connection issue is an interesting case. I previously thought it was mostly due to a poor Zigbee signal strength, but since this is the second report of this issue, I'll try to investigate further when I have time.

pablosmith, 2025-03-11:
> Hello,
>
> I previously successfully dumped the firmware from the device using [this UART tool](https://github.com/Suxsem/Telink-Zigbee-serial-flash-tool/tree/main). However, it only worked for [...]

I think it is connected with fake ZT2S modules. With different pinout. I will check the firmware you prepared for my relay a little later.

gresuk, 2025-05-22:
> > Привіт,
> > Раніше я успішно скинув прошивку з пристрою за допомогою [цього UART-інструменту](https://github.com/Suxsem/Telink-Zigbee-serial-flash-tool/tree/main) . Однак, з якоїсь причини він працював лише на читання, а не на запис. В результаті мені довелося використовувати веб-прошивальник для запису на пристрій. Відтоді я замовив TB-03 і тепер використовую [цей інструмент](https://github.com/pvvx/TLSRPGM) , який працює набагато надійніше.
> > Проблема з підключенням – цікавий випадок. Раніше я думав, що це здебільшого пов’язано зі слабким сигналом Zigbee, але оскільки це вже друге повідомлення про цю проблему, я спробую дослідити її детальніше, коли матиму час.
> 
> Я думаю, що це пов'язано з підробленими модулями ZT2S. З іншою розпіновкою. Я перевірю прошивку, яку ви підготували для мого реле, трохи пізніше.

Hi. Did you manage to read the firmware from your relay? I have a similar relay with the same ZT2S and pinout, but I think I accidentally erased the bootloader when I flashed whd02_factory.bin, and now nothing seems to help.

romasku, 2025-11-16:
Hi, not sure if this is still relevant, but if you flash custom firmware, there is no need for a separate bootloader for the Telink MCU. And in stock firmware, the bootloader is just at the start of the flash (0x0-0x8000), so it is part of the binary file, and you should not write the bootloader separately.
Device is supported as `TUYA_3_TS0001`, so I'll close this issue. If you experience any problems with this device, feel free to open a new issue.
Also, it seems like https://github.com/romasku/tuya-zigbee-switch/issues/80 is about the same device.