romasku/tuya-zigbee-switch #25: Commands mode

pablosmith, 2025-03-11:
Is it planned to add support for sending commands, not just events? This could be used for cheap buttons to make direct binding. I think your project has a big dance as a firmware designer for devices on cheap tuya modules.

romasku, 2025-03-11:
The firmware already supports both events (e.g., "release," "press," "long press") and commands, more precisely, the OnOff client cluster. You can use binding to connect directly to another Zigbee device. I personally have one switch in detached mode that is directly bound to a smart bulb.

I also have a cheap Zigbee wireless button. I'll check if it uses the same Zigbee module and if it's possible to port the firmware to it.

pablosmith, 2025-03-12:
I have several suitable devices on modules from tuya. Now waiting for the board for the programmer to make a backup copy of the standard firmware. After that I will be ready to test the firmware.

benbender, 2025-03-20:
@romasku I'm also trying to achieve a setup like yours described in: "I personally have one switch in detached mode that is directly bound to a smart bulb."

I've got a "Moes-2-gang"-switch where 1 switch is directly connected to the ceiling lamp and works perfectly. The other one is bound to a zigbee bulb and works perfectly when toggle the switch physically. That was not possible with the original firmware – so big improvement here! Thank you!

What doesn't work and I'm not sure why, is getting the same behavior when toggling the state of the switch via z2m. It just does nothing. I've tried to play with different settings for the switch and couldn't get it to work. Any advice about what I might miss here? I'm a bit confused tbh…

romasku, 2025-03-20:
Thanks for the thanks! :)

In detached mode, the "switch (button)" entity and the "relay" entity are independent. When you toggle the relay state in Z2M, you're controlling the relay itself, but the binding is between the "switch (button)" and the bulb.

The reason for this setup is that the bulb’s state can always be changed directly (via the device page, another binding, etc.), and the switch device won't be aware of these changes. As a result, the actual state of the bulb and the "switch" state in the UI may become out of sync. This relates to the "source of truth" for the bulb’s state: only the bulb knows its correct state, and there is no reliable way to sync this state with the switch.

What you want is to use the switch as a "proxy"—instead of directly controlling the bulb, you'll send commands via the switch. Also, I'm fairly certain the bulb supports at least brightness control, so you’d likely want to expose that in the Home Assistant (HA) UI.

My suggestion: in your HA dashboard, simply add the bulb itself. Physical button presses will control the bulb and send commands to it, and the bulb will report its state back to HA, ensuring the dashboard reflects the correct status. Then, pressing the bulb icon on the dashboard will send commands directly to the bulb, allowing it to function as expected. The "Relay 2" of the switch will always remain on, but you can just hide it from the dashboard.