romasku/tuya-zigbee-switch #26: Conflict with smart plug

Vanito1987, 2025-03-13:
Hi! Thanks for your great work!
Flashed first 3 of my _TZ3000_jl7qyupf, two buttons work at the same time now and much faster then it used to be.
But I have one issue with Mi smart plug lumi.plug.mmeu01, which I use as a router to connect devices in my bedroom with the coordinator. There was no such issue before flashing. As I understand, flashed relay being end device connects to the plug, and after that if I switch the state via HA, everything is ok. But if I use physical switch, flashed relay changes state, but the plug changes state too. Is it possible to fix it somehow?

P.S. and about what we all call bug, which you fixed with your firmware - I think it was done on protection purpose that we can’t switch both buttons at the same time - I burned 3 relays before understood that my load exceeds the 100W  limit when I turn on both buttons at the same time (around 30W one button and 130+W second button).

romasku, 2025-03-14:
Hello,  

I'm not sure what might be causing this behavior with your smart plug. Please try resetting the switch by pressing the button five or more times quickly and then rejoining it to the network. This may help if the issue is caused by a "binding" between your switch and your plug  -- when one device directly controls another by sending a direct packet. However, such binding requires manual setup. Here is an example of binding a switch to a smart bulb I have:  

[image]  

Regarding protection, I have to disagree. It is still possible to turn on both relays (or triacs, to be precise), albeit with a 0.5-second delay. So unless you have two fans connected, when both of them can create a power surge when they start spinning, this delay should not create a real difference. Additionally, while I am not sure about all devices, mine has a resettable fuse for safety. Still, I do not have the proper setup to test for overloads as my loads are well below the limit.

romasku, 2025-11-16:
Closing as stale. If you still experience issues, please open a new one.

closed 2025-11-16 by romasku