romasku/tuya-zigbee-switch #27: [Enhancement] Pairing and Factory reset via longer and even longer presses.

CRCinAU, 2025-03-17:
Splitting out of https://github.com/romasku/tuya-zigbee-switch/issues/14 for ease of gathering info.

There seems to be some appetite for a way to factory restore devices and to optimise workflow.

Currently, turning the switch on / off 5 x times in rapid succession causes the load connected to the switch to also be cycled 5 times - which may be undesirable, or in worst case, damage connected equipment.

I'm proposing the following idea to get feedback of what / how these things should be implemented.

My current idea is that any button can be held and then released for:
* 10 seconds to enter pairing mode; and
* 20 seconds to reset the switch to default behaviour.

I was originally thinking that holding multiple buttons for 10 seconds could be a factory reset, but this would rule out this functionality for single-button devices. Having consistent behaviour of multiple devices is probably preferred to having an 'optimised' workflow.

Open for suggestions on how this should / could work.

MnM001, 2025-03-17:
I think they are both good options and I will like to see them implemented too.

MnM001, 2025-03-21:
Hi @romasku - can you please let us know if the above is something that you can look at and make it part of your firmware?

romasku, 2025-03-21:
Hi,
Sorry for the late reply. I think I'll add it. It should be configurable since relay modules do not always have a concept of "press," as they can be attached to a toggle switch, but this should not be a problem.

However, it will have to wait a bit, as I won't have much time in the near future, and I also want to prioritize support for the 4-gang device.

MnM001, 2025-03-22:
> and I also want to prioritize support for the 4-gang device.

@romasku Yes I am really looking forward to that one too!!

clumsy-stefan, 2025-08-03:
I can look into this. However the longpress option will not work if you want to use levelCtr (see #93), which IMHO is more important to be able to dim end devices.

But we still have the config-button which is sepparate. Probably we could use that one.

I guess easiest would be to make the "number of presses" configurable (eg 0 to turn it off and 5-20 for reset).

chamuridis, 2025-10-31:
Hi @romasku, @CRCinAU, and the whole team thank you so much for the incredible effort you’ve put into this firmware. It’s truly a game changer for anyone using Zigbee switches

 I love the idea of moving away from rapid 10× presses for pairing/reset  especially in homes with kids or tenants, it’s so easy to trigger by accident, and re-binding everything becomes a real pain.
A simple, safer pattern could be:

10 short presses + 1 long hold (5s) → enter pairing mode
15 short + 1 long hold (8s) → factory reset

It’s intuitive, deliberate, and nearly impossible to trigger unintentionally.

andrei-lazarov, 2025-10-31:
Hello @chamuridis! Thanks for your suggestion and interest in the project!

The reset methods are by far the most debated topic.
As previously stated in this topic: 
Long press is incompatible with 'rocker switches' and we already assigned it to dimming the bound lights.

We will probably make the 10x press configurable in the next release - so you can set it to a very high number and avoid accidents.
Hopefully more customization will come after that.

Also, right now there is no factory reset.
In the latest version we agreed it's better to leave all options unchanged when you put the device in pairing mode (either by Z2M-remove or spam-buttons). - I don't remember the exact reason, but most Zigbee devices have this behavior.

Instead, I'm thinking to put a button in Z2M Exposes tab: 'Reset to default options'