romasku/tuya-zigbee-switch #28: [BUG] Pairing does not complete with HA's ZHA implementation.

CRCinAU, 2025-03-17:
Splitting from https://github.com/romasku/tuya-zigbee-switch/issues/14 for ease of tracking.

After updating the firmware from factory to this repo, the pairing does not complete successfully with ZHA.

I've noticed that the following workflow is a workaround, but obviously not ideal.

1) Update firmware and then remove the device from ZHA.

2) Use the Add Devices function in HA to start pairing

3) Switch appears, but gets stuck at the following screen:

[image]

4) Restart HA

5) Device is either half configured at this point, or set as 'unknown device / unknown manufacturer'

6) Delete the device again

7) Put the switch into pairing mode and re-pair from HA using ZHA.

The switch will now pair correctly.

benbender, 2025-03-19:
Can confirm the problem with a `Moes-1-Gang`. The device doesn't seem to send ACKs at the end of the interview:

````
2025-03-20 00:15:49.243 DEBUG (MainThread) [zha.zigbee.cluster_handlers] [0xA761:1:0x0019]: 'async_initialize' stage failed: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>
Traceback (most recent call last):
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zha/async_.py", line 64, in sem_task
    return await task
           ^^^^^^^^^^
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 472, in async_initialize
    await self._get_attributes(
        True, cached, from_cache=True, only_cache=from_cache
    )
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 595, in _get_attributes
    read, _ = await self.cluster.read_attributes(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
    )
    ^
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zigpy/zcl/__init__.py", line 576, in read_attributes
    result = await self.read_attributes_raw(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        to_read, manufacturer=manufacturer, **kwargs
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zigpy/zcl/__init__.py", line 378, in request
    return await self._endpoint.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<9 lines>...
    )
    ^
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zigpy/endpoint.py", line 270, in request
    return await self.device.request(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<11 lines>...
    )
    ^
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zigpy/device.py", line 378, in request
    await send_request()
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zigpy/application.py", line 835, in request
    await self.send_packet(
    ...<14 lines>...
    )
  File "/Users/bb/Work/home-assistant/venv/lib/python3.13/site-packages/zigpy_znp/zigbee/application.py", line 1100, in send_packet
    raise DeliveryError(
    ...<3 lines>...
    )
zigpy.exceptions.DeliveryError: Request failed after 5 attempts: <Status.MAC_NO_ACK: 233>
```

CRCinAU, 2025-06-03:
Managed to look at this today - I also couldn't reproduce anymore with the latest firmware.

closed 2025-06-03 by CRCinAU