romasku/tuya-zigbee-switch #29: _TZ3000_zmy4lslw not workink 1 relay

grzegorzlabudda, 2025-03-19:
I have a problem with another module, 1 relay does not work. Instead of a relay, the LED lights up

I have this diagram

Tuya-TS0002-custom;TS0002-custom;BD2u;LC0;SB4u;RC2;SB5u;RC3;

and there is a wall switch instead of a relay.

what should I change

romasku, 2025-03-19:
Hello,

Try swapping pins for relay and led:

With the first relay swapped:
`Tuya-TS0002-custom;TS0002-custom;BD2u;LC2;SB4u;RC0;SB5u;RC3;`

With the second relay swapped:
`Tuya-TS0002-custom;TS0002-custom;BD2u;LC3;SB4u;RC2;SB5u;RC0;`

> and there is a wall switch instead of a relay.

Sorry, I didn't get you here.

grzegorzlabudda, 2025-03-19:
thanks. tomorrow I will try coding again. and let you know. and about the icons displayed in the interface, it is 

[image]

instead of this

[image]

I don't know if it's a firmware error or if it's just supposed to be like that

grzegorzlabudda, 2025-03-20:
unfortunately now I have a small white brick :P I removed the device from z2m and it doesn't want to add it back.

The settings changes you suggested unfortunately didn't help. 

[image]
[image]
[image]

The device looks like this

romasku, 2025-03-21:
Hi,

I'm sorry to hear that. I see that the SWS contact is exposed, as well as the VCC and GND pins. If you have a UART adapter, it should be possible to recover it. Could you please provide an Aliexpress link for this device? I'll order one and properly reverse the pinout.

grzegorzlabudda, 2025-03-21:
this is the auction

Właśnie udało mi się znaleźć to na AliExpress: 
43,14zł | GIRIER Tuya ZigBee 3.0 Moduł przełącznika 10A Smart Home DIY Breaker 1 2 3 4 Gang Obsługuje 2-drożne sterowanie Współpracuje z Alexa Google Home
https://a.aliexpress.com/_EHHdXri

I have a uart adapter, I will try to upload new software, but when I have some time. Do you have a list of which pin is responsible for what?

romasku, 2025-03-21:
To flash by wire, follow [this guide](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/flashing_via_wire.md).

[image]

grzegorzlabudda, 2025-03-22:
hello. thanks for the pinout. as soon as I have a moment I will check it. I understand that I do not connect rst.

romasku, 2025-04-21:
Hello,

I finally received your device from AliExpress and found some time to work on it. I’ve dumped the firmware, corrected the pinout, and confirmed that it’s now working properly. And here is the correct pinout, in case you've managed to reconnect to Z2M:

```Tuya-TS0002-custom;TS0002-custom;BD2u;LC2;SB5u;RC4;SB4u;RC3;```