romasku/tuya-zigbee-switch #36: Revert to factory firmware

ifurnadjiev, 2025-04-21:
For testing purpose I want to revert one of my MOES 1-gang switches to its factory firmware. Is this possible? I saw "*factory.bin" firmwares in the "bin" folder. Are there an OTA method to revert back to factory firmware? If not, can you please provide a way to do it?

romasku, 2025-04-21:
Hello,

Can you confirm exactly which device you have? Is it the Moes relay module without a neutral?

You *can* revert to the stock firmware, but this is only possible by flashing via UART. You can find more details in this [guide](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/flashing_via_wire.md). Feel free to ask me here for additional help.

However, I believe I know what’s causing your issue: the router firmware consumes more power and doesn’t work properly on non-neutral devices unless there’s a sufficient load attached. It leaks a small amount of current to power itself, which can cause problems. In my testing, a single 8W LED bulb wasn’t enough, but 7W + 8W worked OK, so I estimate the lower limit as ~10W.

I recommend trying the EndDevice firmware — it should work better in your case and can be flashed via OTA.

ifurnadjiev, 2025-04-21:
Hi,

thank you for your answer. The device is exactly MOES without a neutral (ZS-EUB_1gang) and currently using EndDevice firmware. The current load is below 6W (5W + 0.5W). I'll try in the next days to attach a capacitor in order to improve it a bit. Do you now why your EndDevice firmware have a bigger lower limit than the factory one? Can this be optimized somehow? I believe that the factory one required 3W.

Regards

romasku, 2025-04-21:
My EndDevice firmware wakes up more often (every 50ms instead of ~350ms on a stock one), and I'm not sure that I'm using the same mode for CPU sleep. So I think it can be optimized further, but it will require some effort.

ifurnadjiev, 2025-04-21:
Thank you once again. Looking forward your optimizations. Does it make sense to parameterized this wake-up interval?

romasku, 2025-04-21:
My idea was to try to avoid the huge delay present in the stock firmware. All my devices at home work fine in router mode, so most of my understanding of low-power scenarios comes from users like you reporting issues:). As a result, I can't say I have a complete understanding of what works best.

Regarding the slider — I'll look into it. It should be easy to add, but I want to test whether large values might cause the device to miss Zigbee packets.

ifurnadjiev, 2025-04-21:
Low power is the future :)

ifurnadjiev, 2025-04-21:
Are there any difference in your firmware between wake-up interval in EndDevice and Router firmware? Probably for Router keep it smaller and bigger for the EndDevice.