romasku/tuya-zigbee-switch #39: OTA ability add whitelabel device ids

avbdr, 2025-05-03:
in z2m there is a possibility to define multiple device ids to the device definition in order to avoid copy pasting it.
I need to add _TZ3000_in5qxhtt and _TZ3000_ogpla3lh as an alias to _TZ3000_ljhbw1c9 firmware. 

Should I just copypaste _TZ3000_ljhbw1c9  entry twice in the ota device list config file?

romasku, 2025-05-03:
Hi,

In `device_db.yaml`, you can specify multiple `tuya_manufacturer_names`:

```
  tuya_manufacturer_names:
      - _TZ3000_ljhbw1c9
      - _TZ3000_in5qxhtt
      - _TZ3000_ogpla3lh
```

All other files, including the converters and `index_xxx.json` files, are generated based on this file.

Also, are you sure they have the exact same pinout? I've seen many devices that share the same Z2M converter but have different pinouts, which require different firmware.

avbdr, 2025-05-03:
thank you, let me try. I just found your firmware, will flash couple switches that I have and will confirm if the pinout is the same. 
I thought that this device id is assigned per some unique configuration. 
As well I want to test the avatto 3 gangs switch, hopefully with just adding extra pinout pair it will work.

avbdr, 2025-05-12:
@romasku :) you were right, i were able to flash the switch, but pinout is wrong :) I will work during a week to make new config.