romasku/tuya-zigbee-switch #40: "UNSUPPORTED_ATTRIBUTE" error in Zigbee2MQTT log

ifurnadjiev, 2025-05-07:
This error appears in Zigbee2MQTT log with my MOES switches (EndDevice) from time to time. Usually the error pops-up at the end of the firmware flash and than sometimes during the normal operation. I didn't find any functionality problem with the switch (everything is perfectly working), but only an entry in the log.

```
[2025-05-06 18:13:16] error: 	z2m: Failed to read state of 'Wall Switch Bedroom Ceiling Light' after reconnect (ZCL command 0xa4c138ff9d35ebb6/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))

[2025-05-06 18:53:34] error: 	z2m: Failed to read state of 'Wall Switch Bedroom Spot Light' after reconnect (ZCL command 0xa4c138daf19b9912/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
```

I don't know witch attribute is UNSUPPORTED.

romasku, 2025-05-25:
Hi,
Sorry for the late reply.
From the error message, I see that Z2M is trying to read the "genOnOff" state on endpoint 1. However, the first endpoint only has a "genOnOff" client. This means the device can only send commands like "ON", "OFF", "TOGGLE", etc., from this endpoint. The relay state is located on other endpoints.
I believe I know what’s causing this behavior - the "onOff" helper inside converted probably expects both server and client on the same endpoint. I’m not sure it can be easily fixed, as I do not want to rewrite part of Z2M. Anyway, thank you for reporting it, I'll try to look deeper into it later.

ifurnadjiev, 2025-12-04:
I don't see this error message any more with latest 1.1.2 firmware and Zigbee2MQTT 2.7.0.

MnM001, 2025-12-05:
I see it on 1.1.2 too:

```
[05/12/2025, 2:05:49 am] z2m: Failed to read state of 'HallwaySwitch-3' after reconnect (ZCL command 0xa4c138fe6b1f9887/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
```

BlakX-AT, 2026-10-13:
is it possible to support here? I am also facing this "issue"