romasku/tuya-zigbee-switch #41: Add support for avatto 1-2-3 touch switches

avbdr, 2025-05-16:
Im currently  slowly working  on tracing pins on those devices 

Avatto Zigbee 1Gang (ZTS02RD-US-W1)
Model TS0001 - Zigbee Manufacturer _TZ3000_w1tcofu8

Avatto Zigbee 2Gang (ZTS02RD-US-W2)
Model TS0002 - Zigbee Manufacturer _TZ3000_ogpla3lh

Avatto Zigbee 3 Gang (ZTS02RD-US-W3)
Model TS0003 - Zigbee Manufacturer _TZ3000_avky2mvc

Pictures can be seen at https://github.com/doctor64/tuyaZigbee/issues/44

If somebody will finish the pinout earlier then me, feel free to share :)

Pins to do:

Reset pin
Enable Backlight Switch pin
Network status Light

Switch 1
Relay 1
Switch Light 1

Switch 2
Relay 2
Switch Light 2

Switch 3
Relay 3
Switch Light 3

romasku, 2025-05-25:
Hi, 
Great to hear, thanks for interest in the project. Do you need any help from me?

avbdr, 2025-05-28:
@romasku i have extracted the pinout from old firmware  using extract_pinout_from_stock_fw.py . after reboot it seems that none of the switches was defined correctly. Ist there is any other way to reset the device rather then 5 on/off cycles on the switch? :)

romasku, 2025-05-31:
Hi,

Unfortunately, for a device with no physical reset button, the only way to reset it is by clicking five times quickly. This is because a long press can be programmed to a custom action.

> after reboot it seems that none of the switches was defined correctly

Do you mean the order is wrong, or does the device not work at all? Also, could you please post the original firmware? I'll take a look at the section where the Tuya firmware defines the pinout.

avbdr, 2025-05-31:
@romasku i have successfully flashed it over ota and after that nothing
 
[dump.bin.dmp](https://github.com/user-attachments/files/20535077/dump.bin.dmp)

romasku, 2025-06-01:
I've built firmware based on the extracted pinout: [AVATTO 3-Gang Touch switch firmware](https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/AVATTO_3_GANG_TOUCH), the `tlc_switch.bin` file. If you can flash via UART, please give it a try.

Tuya configuration directly from the firmware file:

```
rl1_lv:1,bt_type:1,bt1_pin:14,led_netn:1,rl3_pin:13,rl3_lv:1,net_trig:0,netled1_lv:1,jv:1.1.0,backlit_select:1,backlit_io_lv:0,led_nety:0,backlit_io_pin:12,total_stat:0,bt3_lv:0,bt1_lv:0,reset_t:5,netled1_pin:5,rl2_lv:1,total_led_st:2,module:ZTU,bt3_pin:15,dctrl_select:0,rl2_pin:11,zero_select:0,bt2_pin:16,ch_num:3,bt2_lv:0,rl1_pin:10,crc:17,
```

Extracted pinout from it:

```
LB5; SD3u; RC2; SD7u; RC3; SD4u; RD2;
```

Some observations based on the firmware file and photos:

* The extracted pins for switches 1 and 2 (SD7u, SD4u) match what I see in the photos—they're connected to the touch controller chip.
* I can't verify the other pins from the photos alone.
* It seems there's no way to control the LEDs independently of the relays.
* There is a single `netled` pin used to control the LEDs.
* There's also a `backlit_io` pin, likely used for a second set of LEDs.

avbdr, 2025-06-14:
good day Roman,

thank you very much for your help. Your version indeed worked right away after the flashing. Not sure why mine didnt worked out. 
I will install the switch and will play with it for a week to confirm if everything is operational. 

I will extract the firwmware from the other two during next week. Sorry, Im chronically run out of time on everything :)

avbdr, 2025-06-17:
@romasku i have tested OTA flashing and I were able successfully flash _TZ3000_avky2mvc by just changing id in the database.

```
--- a/device_db.yaml
+++ b/device_db.yaml
@@ -446,7 +446,7 @@ AVATTO_3_GANG_TOUCH:
   firmware_image_type: 43542
   stock_converter_model: TS0003_switch_3_gang
   tuya_manufacturer_names:
-      - _TZ3000_qewo8dlz
+      - _TZ3000_avky2mvc
```

andrei-lazarov, 2026-01-20:
All switches are now supported.
Added 4gang: `TS0004` `_TZ3000_eeswvvtm`

Pics from @avbdr 

[image]
[image]