romasku/tuya-zigbee-switch #42: TS0004_switch_module (_TZ3000_ltt60asa) support request + pins

nezg, 2025-05-17:
Hi, I started by trying to update the OTA firmware by substituting the firmware from _TZ3000_5ajpkyq6, but the pins did not match.

[image]
[image]

After replacing the pins, the Z2M first cursed at a long string, and when I reduced or increased their number, the device seemed to turn off:

(z2m: Publish 'set' 'state' to '0xa4c138ac7b687f7c' failed: 'Error: ZCL command 0xa4c138ac7b687f7c/2 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '18273'.)')

I managed to download the WHD02.bin firmware, and check each group of pins.
Here is a line that I'm sure will work, but due to the error above, I can't add it:

TS0004-CM;TS0004-CM;BB5u;LC1;SB4u;RD7;SD2u;RD4;SC3u;RC4;SC2u;RC0;

romasku, 2025-05-25:
Hi,

Sorry for the late reply — I should be more available now.

To confirm your situation: I have the device `_TZ3000_ltt60asa`. You managed to flash it with firmware for `_TZ3000_5ajpkyq6`, it connected to the Zigbee network, but the pinout was incorrect. Then you reversed the pinout and came up with:

```
TS0004-CM;TS0004-CM;BB5u;LC1;SB4u;RD7;SD2u;RD4;SC3u;RC4;SC2u;RC0;
```

However, when you tried to apply it, Z2M showed an error about the configuration being too long. You then removed part of it (which part?), applied the shortened version, and now the device is unresponsive. Is that correct?

What is the current state of the device? Does the LED light up? Does it respond to physical button presses?

The error from Z2M ("delivery failed") is a general one, indicating that the device is not responding. This is probably caused by an invalid configuration. I suggest trying to reset the device by pressing the physical button for more than 3 seconds or rapidly clicking any switch 5+ times. If successful, this should reset the config to the `_TZ3000_5ajpkyq6` version you originally flashed.

If that doesn’t help, you’ll likely need a wired UART programmer. If you have one, I can build a firmware with the pinout you provided.

nezg, 2025-05-26:
Yes, I have a UART. After trying various firmware versions, only WHD02.bin was able to connect to the network. (probably because of the matching pins). If I added the correct group of pins, I was able to check each group individually (switch 1 and relay 1, switch2 and relay 2, etc.) But I couldn't add the second group, the device seemed to reboot and no longer responded, even if the correct switch was just installed.

If you make a firmware for my configuration, it will be great. I think everything will work.

romasku, 2025-05-26:
Hi,
Here is the build of the firmware with your config: https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/TS0014_AVVATO_NEUTRAL
Use `tlc_switch.bin` to flash via UART.

nezg, 2025-05-26:
I flashed it a second ago, everything works fine. I'll fix the converter a little later and put it here.

Thank you so much for your work!!! I finally switched the whole apartment to modules. And finally, direct binding works. 

I would like to share my experience of using a 2gang switch, for the future: there is contact shaking when the switch is in the middle of the path.

nezg, 2025-06-01:
> 've rebuilt firmware with the config you provided, please try it: https://github.com/romasku/tuya-zigbee-switch/tree/main/bin/TS0004_AVATTO_NEUTRAL

It turned out funny. 
In the disassembled state, it turns out that I messed up the relay order.

I used the HEX editor, and corrected the bin file to the correct sequence (device config) and flashed it.

This line is correct:
TS0004-Avv;TS0004-Avv;BB5u;LC1;SB4u;RC0;SD2u;RC4;SC3u;RD4;SC2u;RD7;

I'm sorry for my carelessness. Now everything is working as it should.

romasku, 2025-06-01:
I've rebuilt firmware once more, with the updated config you provided.

p2baron, 2025-06-09:
So I flashed this firmware today and it is a working great!  Thanks you so much for making this firmware! 

I do notice a light whining sound when relay #2 is on. Not sure if that was there before I flashed it...

The only thing a bit confusing is the order the various elements under the "Exposes"  tab in Z2M.
Also there are a lot of duplicate endpoint in the "Bind"  tab. See
[image]

clumsy-stefan, 2025-07-06:
Hi
I also use one of these switches (4-Gang) which works excellent!! Thanks again to the author and contributors for this project, I was loking for this for months!

One thing I noticed, is that the status of relay 3 is not actively reported back! 0-3 work fine and I see the actual state immediately after the keypress in z2m. Relay 3 I have to get the status manually.

Any idea why?

thanks, regards

andrei-lazarov, 2025-07-13:
@clumsy-stefan did you try to interview, reconfigure and reset the device?

clumsy-stefan, 2025-07-14:
Yes, I think I tried that, but can do more tests if needed... As I only borrowed this unit until my own arrive I didn't want to risk to brick it, so after what happened to the other ones I got a bit cautious ;)

romasku, 2025-11-16:
Device support was added (thank you!), so I’ll close this issue. If there are any remaining problems, please open a new one.

closed 2025-11-16 by romasku