romasku/tuya-zigbee-switch #44: Please add WHD02 TS000F _TZ3000_skueekg3 zt2s

gresuk, 2025-05-20:
Hi. Is it possible to add this device? I've already tried flashing WHD02, WHD02_2, and WHD02_Aubess. Nothing works.

[image]
[image]
[image]
[image]

gresuk, 2025-05-20:
I don't understand anything, the pins don't match at all.

[image]

gresuk, 2025-05-20:
This config should work, I don't understand what I'm doing wrong.
WHD02_2:
  name: WHD02-custom
  config_str: Tuya-WHD02-custom;WHD02-custom;BB1u;LB4;SD2u;RD3;
  device_type: router
  tuya_manufacturer_id: 4417
  tuya_image_type: 518
  firmware_image_type: 43535
  stock_converter_model: WHD02
  tuya_manufacturer_names:
    - _TZ3000_skueekg3
  human_name: No-name 1 gang switch with no rf shiled ZT2S
  status: In progress, tuya_manufacturer_name is not confirmed
  github_issue: https://github.com/romasku/tuya-zigbee-switch/issues/24
[image]

gresuk, 2025-05-22:
I finally got it working. The problem was with the UART adapter—I replaced it, and everything started working. The firmware that worked for me was whd02_2. I bought this zigbee switch on AliExpress from the seller "sixwgh Official Store". OTA updates are working as well.

If you have the same data as me:
tuya_manufacturer_id: 4417, tuya_image_type: 518,
you can temporarily update using my link:
"https://raw.githubusercontent.com/gresuk/tuya-zigbee-switch/refs/heads/main/zigbee2mqtt/ota/index_router.json"
or build it yourself.

Thanks to romasku for the excellent project.

romasku, 2025-05-25:
Hi,
First, sorry for not replying earlier — I should be more available now.

Just to confirm — have you enabled "momentary mode" in z2m?
If so, your problem is probably caused by switch bouncing. I can increase the debounce delay to prevent such issues.

gresuk, 2025-05-25:
Hi. Yes, it is enabled. I also suspect it’s caused by switch bouncing. I would be very grateful if you could add this delay or the option to enable it in Z2M.

[image]

gresuk, 2025-05-25:
```
  // Add check debounce
  if (is_pressed != button->debounce_last_state)
  {
    button->debounce_last_change = now;
    button->debounce_last_state = is_pressed;
  }

  // Ignore change state
  if ((now - button->debounce_last_change) < 50) // 50ms debounce
  {
    return;
  }
```

It seems like it helped, currently testing.

romasku, 2025-05-31:
I've merged your PR, thank you once more. I'm closing this issue.

closed 2025-05-31 by romasku