romasku/tuya-zigbee-switch #45: Group support with ZHA?

CRCinAU, 2025-05-28:
I noticed with great joy that group support has been added to the firmware allowing raw zigbee bindings between endpoints.

YAY!

Is there any way to do this via the ZHA integration?

I was unable to figure out how to implement this.

romasku, 2025-05-31:
After some testing, it seems that the SDK doesn't support binding to a group. I can bind a device's output endpoint to another device's input endpoint, and I can add a device to a group and broadcast commands to the group, but binding a device's output endpoint to a group doesn't work. I'm investigating whether this can be fixed.

CRCinAU, 2025-06-04:
I'm trying this - but I note that none of my Moes light switches show up in the "Add Devices" area.

CRCinAU, 2025-06-27:
I seem to have gotten a bit further.... If I remove the light switch, then add it again, it then appears in the groups list.

I'm guessing this means that it needs to do a reconfigure / reinterview the device.

If I do this with a light switch I haven't re-added yet, I get:

[image]

If I then do one that I've already re-added, I see the groups area at the bottom:

[image]

That being said, when I add all endpoints of two devices to a single group (to test, in theory any button should turn everything on/off), nothing actually happens...

Still digging....