romasku/tuya-zigbee-switch #48: Would TS0013 _TZ3000_lrgccsxm be possible?

yannick-toolset, 2025-05-30:
Hi there, 

This project is amazing! Thank you for running it. 

I have a Zemismart TS0013 switch with the delay issue. Would it be possible to have this work with it? 

I set everything up for OTA updates, however I don't see any updates available even though it does show up in the OTA tab now. 

I assume that's because it's not compatible with the firmware upgrade and uses different internals? My IEEE address starts with 0x3425b4.

I'd love long or double click support too! 

If there is any information you need to be able to answer the above, please let me know.
 
Thanks,

Yannick

romasku, 2025-05-31:
Hi,

As I see it, [0x3425b4 is a SiliconLabs](https://maclookup.app/search/result?mac=3425b4) MAC address, so the firmware currently doesn't support it. However, I'm working on another SiliconLabs device, so support will be added later. Unfortunately, porting to a completely different MCU is difficult, and I don't have much time at the moment, so I can't say when it will be ready.

In any case, thank you for your interest in the firmware.

yannick-toolset, 2025-05-31:
Thanks @romasku !

That makes sense. When you get to it if I can provide any information let me know. And of course no rush at all, the fact that you're running this project is great in the first place. O look forward to it whenever it happens.

Have a great weekend!

Yannick

andrei-lazarov, 2025-11-16:
**Silabs devices now supported!** But they can't be converted OTA.

Your device needs [porting.md](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/contribute/porting.md).
Then [flashing_via_wire_silabs.md](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/contribute/flashing_via_wire_silabs.md)