romasku/tuya-zigbee-switch #49: Support for TS0002_basic (_TZ3000_bvrlqyj7)

andrazek, 2025-06-01:
Thanks for the great work — this really transforms these devices and finally makes them fully usable!

I’ve noticed that I have two types of Girier 2-gang switch modules. _TZ3000_zmy4lslw upgraded successfully using your instructions, while _TZ3000_bvrlqyj7, although listed in Zigbee2MQTT's OTA section, were not offered the new firmware. Their IEEE address starts with the prefix 0xa4c138 which suggests they are Telink-based devices. I have no idea what is the difference, but they are both described under the same product on [blakadder](https://zigbee.blakadder.com/Tuya_SML-02Z.html). Is it possible to make TZ3000_bvrlqyj7 compatible with your firmware in the future?

romasku, 2025-06-01:
Hi,

You're very welcome — I'm glad to hear the project has been useful!

Yes, it should be possible to add support for the `_TZ3000_bvrlqyj7` variant. Devices with the same product name but different Zigbee Manufacturer often have slightly different hardware inside — probably due to different board revisions.

To properly port the firmware to `_TZ3000_bvrlqyj7`, I would need a verified pinout for the device. I could prepare a firmware build that is OTA-flashable based on `_TZ3000_zmy4lslw`, but there's a risk of bricking the device. In that case, you'd need a UART to recover it.

If you have a UART flasher and are comfortable experimenting, you can follow the [porting guide](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/porting_to_new_device.md). If you have any questions regarding this guide, feel free to ask me.

andrazek, 2025-06-09:
Thanks! I'm not very comfortable experimenting, as I don't have much experience with soldering. I opened one of the modules, and the inside looks like this:
https://github.com/romasku/tuya-zigbee-switch/issues/6#issuecomment-2564877505

In case OTA flashing bricks the device, where exactly should I solder the wires? Do I need to remove any components to access the correct pins?

embrion44, 2025-07-29:
Guys, are you still interested in adding support to this? I'm happy to test mine with the probe

andrei-lazarov, 2025-07-29:
@embrion44 Please do! You can report the pinout here and I will build the firmware for your device

embrion44, 2025-07-29:
@andrei-lazarov , awesome!

I've tried but I may be doing something wrong. I've managed to probe for example S1 or S2 to specific 154 resistor but putting probe to every ZT3L pinout does not ring. I'm guessing I should be probing those leg-less round points on ZT3L, right?

embrion44, 2025-07-30:
Another update.  Two pinouts traced back to the relays daughterboard, but cannot tell which one is which. I've compared to 1CH model from the same comment/post above and assuming 2CH has CH1 wired the same (pretty logical), then I can confirm I have a match for both relays. Now just the switches need to be found/confirmed somewhere on the motherboard.

[image]

embrion44, 2025-07-31:
@andrei-lazarov, I can confirm the device id and I've even already used your build on my test switch. Working great, thank you a lot for adding it! Updating the remaining switches.

Since @romasku merged it already to main, can I switch back to the original OTA URL and external converters?

andrei-lazarov, 2025-07-31:
@embrion44 Yes, please do (so you can receive further updates).

andrei-lazarov, 2025-08-09:
Please close this issue. The device was already added.
@romasku can you give me permissions to close/re-open issues?

closed 2025-08-09 by romasku