romasku/tuya-zigbee-switch #56: Support for AVATTO ZWSM16-3-Zigbee

andrei-lazarov, 2025-06-18:
Hello and thanks for sharing this awesome project!

I bought this 3-gang relay with neutral:
Zigbee Model: **TS0003**
Zigbee Manufacturer: **_TZ3000_hbic3ka3**
Manufacturer: **AVATTO**
Model: **[ZWSM16-3-Zigbee](https://www.zigbee2mqtt.io/devices/ZWSM16-3-Zigbee.html#avatto-zwsm16-3-zigbee)**

I can not get it to show up in the OTA tab.
The exact model `ZWSM16-3-Zigbee` is not included in `zigbee-herdsman-converters/devices/tuya` or `zigbee-herdsman-converters/devices/avatto`.
I don't know what to do next. The porting guide does not describe this case.

Can this device be supported and flashed with OTA? If not, I would be happy with an link to buy a supported similar device.
Thanks!

andrei-lazarov, 2025-06-21:
I spent some time with a multimeter to identify the pins:
Button C2, Led D2, Switch1 D3, Switch2 D7, Switch3 B6.

For the relays I couldn't get a clear reading, but I got some values for C0, D4 and C1.

So this is my best guess: `BC2U;LD2;SD3u;RC0;SD7u;RD4;SB6u;RC1;`

[image]

andrei-lazarov, 2025-06-22:
I flashed it via OTA and it works well so I opened a pull request #58

andrei-lazarov, 2025-06-26:
Merged so we can close this.
TS0003 AVATTO ZWSM16-3-Zigbee is supported.

closed 2025-06-26 by andrei-lazarov

romasku, 2025-06-26:
Sorry for not being very helpful here, I didn't have time for this project.
It's great that you were able to port the firmware yourself; that is a huge help, thank you!