romasku/tuya-zigbee-switch #63: Avatto ZWSM16-2 ZHA quirk support

soosp, 2025-06-28:
I just updated an Avatto ZWSM16-2 switch and paired with ZHA. It introduce itself as TS0002-AV-CUS, but there is no such entry in `switch_quirk.py`.
The basic switching functionality is working, but there are neither enhanced config options nor power metering. Fortunately I do not need them, just report it.

andrei-lazarov, 2025-08-22:
We changed the device model name from AV-CUS to AVB.  
If you update your device to the latest version, interview and reconfigure it, it will receive the new name.  
Though Z2M converters keep old names for backward compatibility.. We should make ZHA quirks do the same.  
Thanks for letting us know!

soosp, 2025-08-22:
Thank you! The device was removed and re-added and recognised on the new name now.
BTW: Are there any concept on device namig?

andrei-lazarov, 2025-08-22:
The interview button updates manufacturer name and model name. The reconfigure button updates endpoints and reporting. In that case you're good, we don't change names often. And adding the old names to the quirk should be easy

soosp, 2025-10-18:
It works well for me. Thanks for your efforts!

closed 2025-10-18 by soosp