romasku/tuya-zigbee-switch #66: Support for _TZ3000_mmkbptmx (4Gang ZB Switch), changing PIN assignment bricks unit.

clumsy-stefan, 2025-07-05:
Hi

First: really great project! Exactly what I was looking for quite a long time!! THanks to all authors and contributors!!!

I sucessfully flashed the firmware on a generic tuya 4gang ZB switch from [Aliexpress](https://de.aliexpress.com/item/1005005748264739.html?spm=a2g0o.order_list.order_list_main.112.5aa75c5fKHGOa4&gatewayAdapt=glo2deu).

I used the Avatto TS0004-Avv firmware which seems to run fine, but the PIN assignment is wrong.

So I disassmebled the unit (Pictures further down) and traced/measured all push button and relays, resulting in the following assignments:
Relays:
 1: B7
 2: B5
 3: B4
 4: C4

Pushbutton:
 1: C1
 2: C2
 3: C3
 4: D2

Internal Button: B6
LED: B1

If I understood the config string correctly this would result in:
_TZ3000_mmk;_TZ3000_mmk;BB6u;LB1;SC1u;RB7;SC2u;RB5;SC3u;RB4;SD2u;RC4;

However, as soon as I enter and send this config, the unit goes dead and does not respond anymore. Unfortunately I did not succeed in flashing it via UART (not really sure what's the problem)..

This means the unit is currently bricked and I can't test further.

Probably someone has a similar unit and can support or point me to the right direction, how to preceed...

Thanks, reagards

STefan

[image]
[image]
[image]
[image]

andrei-lazarov, 2025-07-06:
> If I understood the config string correctly this would result in:
`_TZ3000_mmk;_TZ3000_mmk;BB6u;LB1;SC1u;RB7;SC2u;RB5;SC3u;RB4;SD2u;RC4;`

I don't you're supposed to change the first two fields **within the Z2M interface**.

`<device manufacturer name>;<device model name>;<pin setup 1>;<pin setup 2>;...;<pin setup n>;`

If I understood correctly:
The **manufacturer name** is used in the index (`index_router.js`) to receive the correct OTA updates.
The **model name** is used by Z2M to get the correct converter from `switch_custom.js` which is responsible for the device behaviour.

We assign new names for the device when we add it to the database. Then the build workflow will generate the correct index and converter.
But right now, you gave the device names that don't exist. The index does not matter, but the device will not work without a converter. 

To fix this you should edit your `switch_custom.js`:
Duplicate the entry for **TS0004-Avv** and replace the **zigbeeModel** with the one you used: **_TZ3000_mmk**.
Then restart Z2M, make sure your edits saved and try to reconnect your device. Also try the interview and reconfigure buttons from the interface.

About flashing by wire, what image did you try to flash?

clumsy-stefan, 2025-07-06:
Hi

Yep, that's understood, I changed switch-custom.js accordingly...

but I'm not even getting that far. I can't get the device to join anymore (if it doesn't find a converter it would show it as "unsuported")...

I have two of these units. on the first I only changed the BB6u. but as soon as I sent the string, the unit got offline and I never successfully could reach it anymore.

The second one I changed the whole string at once with the same result. The unit went offline and no matter what/how I tried to flash or reset it, I couldn't reach or pair it again...

I tried to flash it with TS0004_AVATTO/tlc_switch.bin and TS0014_AVVATO_NEUTRAL/tlc_switch.bin. However I think it's the flashing proces that is not working. I use a macbook with the mentioned web-flasher...

Unfortunately as long as the uart-flashing does not restore a pairable unit I can't test further...

BTW: with the avatto device everything works quite well (see issue #42)..

clumsy-stefan, 2025-07-06:
Actually not a big problem for me, I mainly wanted to share the PIN assignments that you probably want to include it in the build.. for me I ordered now the (hopefully right) supported avatto 4Gang switches (as in #42) which solve more or less all my issues (I have over 30 4gang push-button switches in my house with FS20 transmitter and until now I didn't find a solution to replace them with ZB, until I found this FW)..

So I'm happy to help and test if possible, just let me know what I can do... Probably I'll try to find out, why I can't flash my units over UART..

clumsy-stefan, 2025-07-06:
Hmm.. .I ordered [these](https://de.aliexpress.com/item/1005007247647375.html?spm=a2g0o.order_list.order_list_main.5.5ea05c5faQFLB3&gatewayAdapt=glo2deu).. I thnik these are the correct ones (with the blue sockets).... and right, the one I have is #42 with green sockets... just realised the differnece now...

I assume that's the only "visible" difference between #42 and #9? But both seem to be avatto?

and thanks again for your help!!

andrei-lazarov, 2025-07-10:
Surely it's possible. I have some IKEA remotes with this functionality. We need to implement the LevelCtrl cluster and change the longpress logic a bit.
Do you want to look into it or should I check it out?

clumsy-stefan, 2025-07-11:
I also have a miboxer FUT089Z remote which sends on/off dimm up/down commands to the fixed ZB groups 101-107.

It would be great if you could look into it, as my programming skills are very limited, especially in JS.... So possibly if you could make the base coding I could help to bugfix, if I understand how it works (for sure I try to ;)...

However I'm happy to help with testing, giving feedback etc. as much as my skills allow ;)

Thanks a lot for your help! really appreciated.

andrei-lazarov, 2025-07-13:
I added _TZ3000_mmkbptmx to the database: #78 so any new devices will be directly flashed with the correct configuration. Thanks for the pinout!
 
I also bricked my Avatto blue-socket by accidentally clicking the config field 😅.  
I opened a new issue #77 so we can move the discussion there and close this thread.

clumsy-stefan, 2025-07-14:
Great thanks!

Unfortunately I can't test anymore, as my units are bricked and I don't succeed flashing it by wire..

The ones I ordered and will arrive this week are different brand (avatto)...