romasku/tuya-zigbee-switch #77: Updating DEVICE CONFIG bricks unit

andrei-lazarov, 2025-07-13:
I updated my Avatto TS0003 with #76 to invert the network led.
I purposefully rebuilt the firmware and updated OTA instead of using the Device config field because @clumsy-stefan reported 2 bricked units after trying to change the pinout: #66 

After the update my device worked normally, but **I accidentally clicked on the text field**:

[image]

**I did not change anything**, but the moment I clicked away from the text field, Z2M sent the command to update the config.
(There is no Submit button, so **Z2M automatically submits after every interaction**.)

I got a generic error and now my device is fully unresponsive:

```
z2m: Publish 'set' 'device_config' to 'AVATTO Triple Switch' failed: 
'Error: ZCL command 0xa4c1387ac2684c1e/1 
genBasic.write({"65280":{"value":
"TS0003-AV-CUS;TS0003-AV-CUS;BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;",
"type":68}}, {"timeout":30000,"disableResponse":false,"disableRecovery":false,
"disableDefaultResponse":true,"direction":0,"reservedBits":0,
"writeUndiv":false}) failed (Delivery failed for '47753'.)'
```

This makes 3 bricked units. We should investigate and maybe remove the option until we fix it? @romasku

andrei-lazarov, 2025-07-14:
I do not have a UART flasher yet. But it would be useful to find the root cause of this issue so I might get one.

clumsy-stefan, 2025-07-15:
> This makes 3 bricked units. We should investigate and maybe remove the option until we fix it?

I guess just commenting out the `romasku.deviceConfig()` lines in `switch_custom.js` would be enough so that no one clicks accidentally in the field?

andrei-lazarov, 2025-07-20:
I got the UART flasher so I started debugging. I can probably figure this out

andrei-lazarov, 2025-07-20:
@romasku Thanks for the tip. I already found that so I was able to receive prints in the serial monitor. 
I immediately saw that updating the config from `oooooold` to `new` results in `newooold`.
So I have to delete the old config or properly terminate the string.
But even if I submit a config of the same length, the device still bricks so there might be more to investigate.

andrei-lazarov, 2025-07-22:
I'm back with some findings. Debugging took so long, but I was able to track the last line executed before it bricks. 
There is an issue with the third relay endpoint. But I don't know what to do with this information so any help would be appreciated.

Again I tested this on the Avatto TS0003 with its default config: 
`TS0003-AV-CUS;TS0003-AV-CUS;BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;`

Here are all the prints I added: https://github.com/romasku/tuya-zigbee-switch/commit/0e5f040f71a07ef0ef87396adc998374ffca3883

## Log 1
When I press the on-board reset button the device resets successfully and we get this log: [log_reset.txt](https://github.com/user-attachments/files/21355097/log_reset.txt).

## Log 2
When I update the device config with this string:
`TS0003-AV-CUS;TS0003-AV-CUS;BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;`
(identical)
the device breaks and we get this log: [log_update_config.txt](https://github.com/user-attachments/files/21355098/log_update_config.txt).

The last line executed in `config_parser.c` is this one, when index = 2 (third relay):
```c
for (int index = 0; index < relay_clusters_cnt; index++)
{
  printf("Adding relay cluster %d to endpoint %d\r\n", index, switch_clusters_cnt + index);

  ---> relay_cluster_add_to_endpoint(&relay_clusters[index], &endpoints[switch_clusters_cnt + index]);

  // Group cluster is stateless, safe to add to multiple endpoints
  printf("Adding group cluster to endpoint %d\r\n", switch_clusters_cnt + index);
  group_cluster_add_to_endpoint(&group_cluster, &endpoints[switch_clusters_cnt + index]);
}
```
And that function stops here: 

```c
void relay_cluster_add_to_endpoint(zigbee_relay_cluster *cluster, zigbee_endpoint *endpoint)
{
  printf("relay_cluster_add_to_endpoint START\r\n");

  relay_cluster_by_endpoint[endpoint->index] = cluster;
  cluster->endpoint = endpoint->index;
  relay_cluster_load_attrs_from_nv(cluster);

  printf("cluster: endpoint, startup_mode, indicator_led_mode\r\n");
  printf("cluster: %d, %d, %d\r\n", cluster->endpoint, cluster->startup_mode, cluster->indicator_led_mode);
  printf("relay: pin, off_pin, on_high, on\r\n");
  printf("relay: %d, %d, %d, %d\r\n", cluster->relay->pin, cluster->relay->off_pin, cluster->relay->on_high, cluster->relay->on);

  ---> cluster->relay->callback_param = cluster;

  (...)
}
```
because the values in the `cluster` of the third relay are wrong. From the log:
```
cluster: endpoint, startup_mode, indicator_led_mode                                                                       
cluster: 0, 83, 48                                                                                                        
relay: pin, off_pin, on_high, on                                                                                          
relay: 806453336, 39649539, 75, 78 
```

## Log 3

When I update the config string with this:
`TS0003-AV-CUS;TS0003-AV-CUS;BC2u;LD2i;RD4;RC1;RC0;SD3u;SD7u;SB6u;`
(random relay order, ends with switch)
the device breaks identically to Log 2.

## Log 4

When I update the config string with this:
`TS0003-AVvvvv-CUS;TS0003-AV-CUS;BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;`
(remove third relay and keep length to prevent other issues)
it works and we get this log: [log_update_config_2relays.txt](https://github.com/user-attachments/files/21355465/log_update_config_2relays.txt)

clumsy-stefan, 2025-07-23:
Interesting...

Just a wild guess: is it really the NV Ram and writiing to it? Or could it be th memory (management) when reading the NV Ram (multiple times) that the variable where the value is put in is either not freed or not correctly allocated? Could show a similar effect, when reading multiple times smaller values more memory is available until the problem occurs...

However I currently don't know anything about memory management of this devices...

Anyways, thanks for your efforts. I'm currently building a "flashing adapter" so I don't need to soldier... probably I can flash my units and support the tests then...

andrei-lazarov, 2025-07-24:
I found out the length issue. config.size is not used. It should be used when writing and reading to flash. Also the size itself should be written to flash.

andrei-lazarov, 2025-07-25:
It looks like I fixed the issues in #89.

But I will keep this open as I found another way to brick the device...
Give it a pin that is not on the board (E0 for example) and it becomes unresponsive.
Maybe we can program it to automatically factory reset if it encounters an error.

andrei-lazarov, 2025-08-12:
The initial issue was partly solved in version 18.

With the latest patch I can comfortably switch between 1,2,3 gang config strings and use the device normally.
Still, the 4-gang config string causes some problems..

Here are the strings I tested on a fresh install of the main branch:
```
TS0001-AVB;TS0001-AVB;BC2u;LD2i;SD3u;RC0;
TS0002-AVB;TS0002-AVB;BC2u;LD2i;SD3u;RC0;SD7u;RD4;
TS0003-AVB;TS0003-AVB;BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;
TS0004-AVB;TS0004-AVB;BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;SA0u;RC4;
```

Changing the string always works. 
Immediately after receiving the new string, the device appears to reboot.
At this moment, pinout is updated correctly and external switches work accordingly.
Also controlling the on-board LED from Z2M still works.

Next, I interview and reconfigure the device which always works for me.. but not for Stefan apparently.
Now the device has the correct name and converter and it functions properly.

However, in the case of a 4-gang, it completely freezes when I try to toggle relays 1-2 from Z2M.
It recovers after a power-cycle. Again everything works.. LED, external switches, Z2M control for relays 3-4, updating the config string, but it freezes on Z2M toggling relays 1-2.
Changing it back to 3 gang still leaves it in this unstable state where reconfiguration / some toggling freezes it.  
But changing to 1-2 gang looks stable.

I also tested this without the lastSeenVersion patch and with NV_MODULE_APP instead of _ZCL, but the results are the same regardless.
Unfortunately the exact behavior I'm seeing is probably not very useful in debugging this.. the code still looks correct to me and it feels like we are just walking around a hidden issue with the flash storage or long strings.  
Probably splitting the config string will help a bit..

clumsy-stefan, 2025-08-12:
Probably you're running in a memory / flash restriction, as you added 2 new storage items:
```
#define NV_ITEM_ZCL_DEVICE_CONFIG_SIZE      NV_ITEM_ZCL_RELAY_CONFIG(MAX_RELAY_CLUSTERS) + 3
#define NV_ITEM_ZCL_LAST_SEEN_VERSION       NV_ITEM_ZCL_RELAY_CONFIG(MAX_RELAY_CLUSTERS) + 4
```
Together with the enhanced cluster (dditional attributes) it probably could have strange efffects?

Just for info: the initial patch you did ran fpr quite a while "stable" in my environment...

romasku, 2025-08-24:
I think I found the problem, there is a bug in this code:

https://github.com/romasku/tuya-zigbee-switch/blob/f6c7ba36066e5ced3935260e2a0ad81378df8757/src/configs/nv_slots_cfg.h#L1-L15

If we have 4 gang device, then the third relay has endpoint 4 + 3 = 7, and entry is:
```
NV_ITEM_ZCL_RELAY_CONFIG(7) = NV_ITEM_ZCL_SWITCH_CONFIG(5) + 8 - 5 = NV_ITEM_APP_GP_TRANS_TABLE + 5 + 8 - 5 = V_ITEM_APP_GP_TRANS_TABLE + 7
```
But then NV_ITEM_ZCL_BASIC_CONFIG:
```
NV_ITEM_ZCL_BASIC_CONFIG = NV_ITEM_ZCL_RELAY_CONFIG(5)  + 2 =  ( NV_ITEM_ZCL_SWITCH_CONFIG(5) + 8 - 5 ) + 2 = V_ITEM_APP_GP_TRANS_TABLE + 7
```

So, they both use the same entry, which overrides each other, and this crashes the device.
Sorry for this kinda stupid bug, I'll verify that it fixes the issue tomorrow and push it to main.

clumsy-stefan, 2025-08-26:
Thanks for the fix!! 

I can try, but I didn't have the issue anymore for some time.. Also not in 1.0.19... 

Additionally I'm away for another 2 weeks, so not sure if I can test from remote...