romasku/tuya-zigbee-switch #80: WHD02 - no OTA cluster

andrei-lazarov, 2025-07-13:
I got a **GIRIER 1 gang switch** with Neutral and Z2M tells me there is no OTA cluster.

```
Failed to check if update available for 'GIRIER Single Switch'
([0xa4c138e672eaaff3 | TS0001]
Failed to find an endpoint which supports the OTA cluster)
```

Zigbee Model: **TS0001**
Zigbee Manufacturer: **_TZ3000_skueekg3**
IEEE Address: `0xa4c138e672eaaff3`
Manufacturer: **Tuya**
Model: **WHD02**

```
- Endpoint 1
  - Input clusters
    - genBasic
    - genIdentify
    - 6280
    - genGroups
    - genScenes
    - genOnOff
    - genTime
    - 57344
    - manuSpecificTuya_3
    - touchlink
```

[image]

I see **3 entries** for this device in the database, so am I doing something here? Or should we add it to the hall of shame?

romasku, 2025-07-20:
Hi,

If the device doesn't report an OTA cluster in Z2M, unfortunately, it can't be flashed OTA. It should be flashable via a wired connection, but otherwise, I agree — we should probably add it to the hall of shame.

Also, the device `_TZ3000_skueekg3` is a strange one — its versions differ by the `manufacturer_id/image_type` pair rather than the Zigbee manufacturer. I have one at home that did have an OTA cluster, so at least some versions of it can be flashed via OTA.

gpaesano, 2025-09-27:
Thanks a lot for the link.

Indeed the module is without shield, so I remapped the pins.

I can confirm it works with:
Tuya-WHD02-custom;WHD02-custom;BB1u;LB4i;SD2u;RD3;
(my understanding is that the network led is on when connected or switched on via the Z2M command).

Tested led, relay, button and switch with Z2M. 

Reliable flashing only at low baud rate.

I build a bin with :
WHD02_V03:
  config_str: skueekg3;WHD02-custom;BB1u;LB4i;SD2u;RD3;
  device_type: router
  tuya_manufacturer_id: 4107
  tuya_image_type: 524
  firmware_image_type: 43577
  stock_converter_model: WHD02
  tuya_manufacturer_name: _TZ3000_skueekg3
  old_manufacturer_names:
  - Tuya-WHD02-custom
  human_name: Mini smart switch
  zb_module: ZT2S
  status: to beflashed via wire only fake Z2TS with no shield pins remapped

but then I saw that WHD02_2 and WHD02_3 are very close.
The converter is using the WHD02_V03 key, I was expecting that it uses  WHD02-custom.
I will test it now for a long run, let's see.

andrei-lazarov, 2025-09-28:
@gpaesano I'm glad it works! Please flash the WHD02_2 image so you will receive OTA updates.  
If you don't want to flash again, add `i43535;` to the end of the config string in Z2M Exposes tab. (I didn't test this yet, but it should work)

gpaesano, 2025-09-28:
Flashed via wire as WHD02_2 - OTA seems to be ok and waiting for the next update :) , working as expecetd. May be led should be LB4i , if you want it on when network connection is established or coherent with Zigbee2MQTT.
Now I will keep it on for a while, before using.
So far so good.

andrei-lazarov, 2025-10-06:
Image renamed to: `TUYA_3_TS0001` in #155