romasku/tuya-zigbee-switch #87: Same device, different pinout

andrei-lazarov, 2025-07-15:
I opened this issue to keep track of devices that have the same manufacturer ID, but different pin-outs.
(as we don't have a method to deal with this yet)

<details>
<summary>_TZ3000_xk5udnd6</summary>

In #51 @solevi reports:  
`SB5u;SD4u;RB6C0;RD7A1;ID2;ID3;LC3;M;`

In #67 @aceindy reports:  
`SB5u;SD4u;RC0B6;RA1D7;ID2;ID3;LC3;M;`

</details>

<details>
<summary>_TZ3000_5ajpkyq6</summary>

In #9 @paddenstoeltje reports:  
`BC2u;LD2i;SD7u;RB6;SC0u;RD4;SA0u;RC1;SA1u;RC4;`

In #9 @clumsy-stefan reports:  
`BC2u;LD2i;SD3u;RC0;SD7u;RD4;SB6u;RC1;SA0u;RC4;`

</details>

solevi, 2025-07-24:
Hi, back from vacation.
I apologize, I tested the device on my desk and somehow didn't notice that the ON/OFF logic was inverted (compared to the blue status LEDs state).
Indeed, @aceindy 's configuration is correct:
`Bseed-2-gang;Bseed-2-gang-ED;SB5u;SD4u;RC0B6;RA1D7;ID2;ID3;LC3;M;`
However, I am powering the device from a lab power supply with an ammeter, and I cannot confirm increased current draw with my configuration.
With both configurations, the current draw from the 12V power supply was around 84.1-84.5 mA (depending on LED states), pulsing during the ON/OFF state change.
Not sure what can cause instability.

andrei-lazarov, 2025-07-24:
@solevi Thanks for clearing it up! That's a relief 😅
Do you need help getting the inverted configuration?
You should use the force index and perform an OTA.
Do not use the config string field. It might brick the device: #77

aceindy, 2025-07-27:
> With both configurations, the current draw from the 12V power supply was around 84.1-84.5 mA (depending on LED states), pulsing during the ON/OFF state change.
> Not sure what can cause instability.

With the current config, the bistable relays are always powered, so there shouldn't be much difference between on/off (or pulsed).

You could re-test with https://github.com/aceindy/tuya-zigbee-switch/tree/LatchRelay125msPulse
That one does pulse (125ms) but uses a blocking code, so not recommended for release.

That would allow you to measure if there would be a difference in amps though 🤔

aceindy, 2025-07-29:
@solevi , that is what I figured as well...hence I changed the logic to only give a 125ms pulse.
With that version it works fine and I can use the switch as a router too...
My only issue is that my version uses waitms(125), which basically holds the software for 125ms. 
This should be dealt with by a seperate timer in a seperate thread, that re-sets the pin after 125ms (but I didn't get it stable, it just freezes the whole unit after 3 or 4 times switching 🤔 )

paddenstoeltje, 2025-08-05:
I would propose to add 2 basic timers per relay:
* maximum time on: works like sonoff inching, this basically switches off the relay after max-time-on attribute and could be used for bi stable relays and would solve the issue above.
* minimum time on: some devices require a minimum pulse width; eg opening closing the garage door. I have the problem now that sometimes the push on the buttons is not registered as it switches on and off too fast. So whenever a off command comes via button or Zigbee, the device waits until the minimum time has elapsed before turning off the relay…

Would add a lot of functionality for many applications (eg bi stable relays, light push dim control, …)

andrei-lazarov, 2025-08-15:
I added a warning for devices known to have multiple pin-outs in #108.

Moving on, please use the respective's device issue page (or create one if needed).

Known devices will be listed in this doc: [multiple_pinouts.md](https://github.com/romasku/tuya-zigbee-switch/blob/main/docs/multiple_pinouts.md)

closed 2025-08-15 by andrei-lazarov