romasku/tuya-zigbee-switch #88: Feature request: Mixed Router + End device OTA JSON

soosp, 2025-07-16:
Please create OTA JSON file contains End device firmware links for no neutral devices and Router firmware links for with neutral devices to able to use one JSON for such mixed config because if you want to use End device firmware, and you have with neutral devices too the JSON files conflict if both are enabled.

aceindy, 2025-09-02:
I disagree; one of the reasons I re-flashed my no-neutral devices was because it did not route 🤔

andrei-lazarov, 2025-09-02:
@aceindy it doesn't mean we delete the all-routers index 😂 

I want to make it like this:
- router.json - all get router
- mixed.json - L+N get router, L get end_device
- end_device.json - all get end_device

But I'm waiting for some feedback on #122 to see if we actually need end_device for L+N devices. Because it would increase build time with a few minutes.

Also there are devices with optional Neutral... for which I don't know how to handle the mixed index.

romasku, 2025-11-16:
Hmm, I think it may conflict with https://github.com/romasku/tuya-zigbee-switch/issues/193. Ideally, it should work like this:
- `router.json` - all devices get `router`
- `end_device.json` - all devices get `end_device`
- `mixed.json` - devices that are already routers get `router`, devices that are end devices get `end_device`.