romasku/tuya-zigbee-switch #98: Proposal and discussion on conceptual design change

clumsy-stefan, 2025-08-01:
To all that are interested

I open the issue here mainly to collect some thoughts and views on something I came around while enhancing the current FW made by @romasku.

First I'd liek to say big thanks to @romasku, this project/FW is what I was looking for for quite a long time! It's fantastic, for me it was the missing peace to completely update my old FS20 infrastructure to a new ZB-based infrastructure. Only that I couldn't find a replacement for my 4-way push-button switches that had FS20 transmitters behind... So a "pure" transmitter with direct ZB bindings is the only solution for me. Notably I have around 40 such swichtes and probably around 200 end devices..

Also thanks to @andrei-lazarov who helped me to get this far!

Now digging deeper into the project I found that basically there is a sepparate FW produced for each and every variant of a device. But the main difference in the binaries is the config string, all the rest is the same (except image type nr. etc.).

IMHO this is unnecessarily overcomplicated and complex. The config string is uniqe to each device and shouldn't be hardcoced in a binary (my view).

So my suggestion would be to have one universal binary with basically an empty config string (except manufacturer and model) and have the device up and running, connected to the ZB network and then individually configure it (see also my PR #94).

Advantages:
- all devices use tha same binary (as the binary is only dependant on the chip and not the pinout config)
- a factory reset quarantees that the device can connect to the network, independant of any "bricking" device-config
- much faster build times, less configurations, etc. (ok, enduser doesn't care about this)
- automation is possible, as a homeautomation software can automatically push the correct config to a switch, once it's connected.
- less different z2m external-converters and zah quirks
- only need as many converters/configs/image-types as there are different endpoints (eg. 1-2-3-4Gang) and not as different pinouts (multiple identical configs for the same type of switch)

Disadvangates:
- the enduser needs to configure the device at least once
- probably other ;)

I did try this in my environment with 6 switches of 3 different brands and configs. For me this is an "easy" way to go. As in PR #94 prepared I made a binary with only the maufacturer and universal model in it and configure the variant while it's up and running.

For sure, there are probably a number of things that have to be taken care of when going that way, but currently I can't find any significant unsolvable ones.

So all of this is just IMHO, please feel free to comment, dispute or suggest on it!

Regards

STefan

andrei-lazarov, 2025-08-01:
Thanks for the shout-out. I support this direction!
The building process seems wildly inefficient to me.. build the same fw 40 times, tens of duplicate converters, troubles updating if you have the wrong combination.

Once we get the config string fully working, we can slowly move towards the universal binary.
Right now the risk of bricking a device is very high.. the config string needs much more validation: present pins on board, minimum and maximum lengths and probably more stuff..
I will try to split the string into multiple fields so we eliminate the too-long-string-issue and do more testing.
I'm also thinking about a recovery method: eg. 5 times power-cycle = reset

Finally, I think other projects already do this and it works for them!
Last time I checked out TASMOTA (similar project, but for WiFi devices.. we know better than that 😉) it was like you described.. universal binary and user configures the pinout (from a drop down selector or from a guide).

mikeS7, 2025-08-02:
I have specific usecase. There are 2 zigbee networks but one don’t use zigbee2mqtt or zha  and connected directly to the general zigbee hub. I need router the network and I flashed Moes 1 gang switch with custom firmware. Then I can connect it to any zigbee network without any issue. But with your proposal device will loose config and become unusable.

I understand my case is specific but it is working for me. I agree with @andrei-lazarov that special recovery method is required. I suggest to do not reset config string on deleting device from network. In such case it will be possible to configure device in z2m and use it with general zigbee hub.

mikeS7, 2025-08-03:
> I also thought about keeping the config string between resets. I will check if this is possible and add it as an option.

I think about 2 different methods reset and entering pairing mode. In my case I use Avatto relays with factory firmware with general zigbee hub but configured in z2m. Configuring in z2m allows me to set switch type (toggle, state or momentary) power-on  behavior. Entering pairing mode (manually or automatic) doesn’t clear the settings and allows to use preconfigured device in general zigbee network.

So entering pairing mode should preserve some settings but reset perform clean factory reset. @andrei-lazarov What do you think?

andrei-lazarov, 2025-08-03:
@marazmarci I don't want the power-cycle method to replace the spam-button method. 
We need to keep it, so resetting is possible without uninstalling the light-switch.
But I agree, the option to disable it is necessary. 😀 

@mikeS7 I also find it weird that reset = disconnect for zigbee devices..
Since Z2M and others don't have a disconnect-without-resetting button, we should keep it unified.  
And I think it's best that all recovery methods would have the same behavior.

Instead, we should add some options in Z2M to configure the reset:
- [ ] keep config string
- [ ] keep configuration (switch mode, action mode, power-on behavior etc) 
- [ ] disable button-spam method

mikeS7, 2025-08-03:
> Instead, we should add some options in Z2M to configure the reset:

I would prefer to separate factory (clean) and pairing reset. For troubleshooting you need factory reset but for general use pairing mode will be enough. Please keep in mind usage without z2m )

Just an idea how to separate:
- pairing mode - 3 quick presses and long (10 sec) one until entering pairing mode
- factory reset - 5 or more presses and maybe long one

Also we should think not only for momentary type but for toggle and state also. For toggle or state types my proposal is not suitable.
I'll think on that

andrei-lazarov, 2025-08-03:
@clumsy-stefan 

> For non ZHA/Z2M use: how will you configure the buttons if these are momentary or toggle? How are you conifiguring relais mode (synchronised etc.)?

You can't configure the device outside of ZHA/Z2M, so you are stuck with factory settings on other zigbee networks. That's why we need to customize the reset.

> But factory (tuya) firmware already does this, so why changing firmware then?

Mike installed the custom fw to add routing functionality to his switch. All Line-only devices are end_devices on stock fw, some L+N too..

clumsy-stefan, 2025-08-03:
ok, thanks for clarification..

I just browsed through the code and I think the easiest and fastest would be to change the 5x Click to "leave network and reboot" and keep the longpress config-button for factory reset. Like this we already have the possibility ro leave and rejoin networks without loosing the config...

Also this would prevent the kids/wifes/others "accidentially" playing with the switch and render it unusable...

clumsy-stefan, 2025-08-03:
@mikeS7: can also be done the other way round.. however I think it's "better" to have the factory reset not on the external buttons (due to the accidental resetting).

It could also be done both on the internal switch. Long press just reboot, 5x Press reset.

On the other hand, I have roughly 130 ZB devices in my network with over 30 different types, and nearly all of them work differently (for resetting, reconfiguring, etc.). So IMHO if you use a "new" or "different" firmware you shouldn't care too much what the standard FW does (that's why you use a new one) but how you could make best use of it... That's why you move away from original firmware, because the functioning is not as you would want it to be.... At some point you need to get rid of the backwards compatibility...

Again only IMHO, not meant to offend anyone!

romasku, 2025-08-03:
Hi everyone,

Thank you all for this discussion. Here are my thoughts:

First of all, having the same binary was the reason I introduced config strings in the first place, so I like this approach. However, having a device initially flashed with an empty config concerns me a bit, as it means the device will be unusable until configured. So, before we proceed, we should ensure there is a reliable way to configure it. Since the Telink chip supports simultaneous Zigbee and BLE connectivity, I was thinking of using BLE for the initial configuration. This would also address non-Z2M/non-ZHA use cases, as everyone has a phone with Bluetooth. 

Regarding config string reset: I believe the only time you'd want to reset it is if you've entered a bad config that breaks the device. But in such cases, the device is likely bricked and unresponsive anyway. So, I don't think we need a physical "reset to empty config" or "factory settings" button, only a re-join button. Instead, we should add safeguards against bad config strings. My idea is to implement a "confirm change with auto-rollback" mechanism, similar to how changing screen resolution works:

- The user updates the config.
- The device reboots and waits for config confirmation by setting a flag in Z2M for 10 minutes.
- If the device doesn't receive confirmation or detects a bootloop, it reverts to the previous config.

That's my vision, but I may be missing something. A couple of questions for you:

- Do you see any use case where reverting to an empty config via physical press would be useful?
- Is there a use case for resetting switch settings (e.g., Momentary/Toggle) via a physical press?

clumsy-stefan, 2025-08-03:
@romasku: thanks for your view! Very appreciated! And again thanks for initiating this project in the first place!!

I agree that there are not a lot of cases where you need to update the config string, but the other configs probably more often, eg:
- when you exchange the switch because it's broken or you need more buttons
- when the device braks and you exchange it for a new one
- when you use it in independant mode (eg for binding with other devices) and change the mode (eg. for a chime)
- when you rewire the leads physically
I came across most of them that's why I bring them up here.

Config-String changes are probably more when you have a new device so you can try&error to find pin assignments.

I like the idea of fallback after a config change... the problem is, that if a device really bricks, it probably won't even get that far and you are stuck (which is probably the same as reverting to an empty config).

I already started a private discussion with @andrei-lazarov about the config. My idea would be to use the already existing manufacturer and model config just expose them so you can change them and only a "config button, "led pin" and "number of buttons" and "number of relays" general config fields and then the individual pin config within each switch and relay. this would make all strings, etc. shorter and from my point of view more "logical" as one big config string..

I did some amendments that the device only reboots and rejoins when you 5x press the button (and keeps the config) instead of resetting it completely, only the internal config button will reset. However I can't find why the "delete from network" in z2m forces a clearing of the config. I'm still on that. If I succeed I'll add a PR so you can have a look at.

andrei-lazarov, 2025-08-04:
I just bought some switches, so they are still on stock (original) firmware.
I tested the AVATTO blue-socket 4-gang switch so we have a reference.

The stock firmware has 3 reset options:
- 5 times power-cycle (sometimes it needed 6)
- 5 times external switch press (any switch)
- long-press on-board button (spamming does not trigger a reset)

All of them work and have the same behavior:
The device keeps configuration, leaves the network and enters pairing mode.

(The configuration is tiny though 😂 only 3 settings. We've come a long way!)

andrei-lazarov, 2025-09-15:
@ThatMishakov these are some good suggestions.

**Networks without coordinators** are also interesting to me.  
In general, I don't understand why there isn't a way to set them up without Touchlink.  

I was thinking starting a temporary Z2M instance, binding the devices together and removing them.  
But the binds are lost on removal. And if you just stop the instance, I'm afraid the devices keep looking for the coordinator and drain battery.