romasku/tuya-zigbee-switch #106: Leave and rejoin ZB Network also clears NVRam config

clumsy-stefan, 2025-08-06:
When a device is removed from the network (either from the server side eg z2m leave or via config-button) all configuration stored in NVRam is cleared. Specifically the Button/Relay types, long press duration, etc. SO basically a factory reset is done.

I cam across this while trying to implement different reset options (one for multi-press switch, one for long press config button, one for multi-press config button). However no matter which reset method is used, as soon as the node leaves the ZB network all information is cleared. and a factory reset is done.

A similar method (different reset methods) is shown in [this](https://github.com/pvvx/ZigbeeTLc) project. However these various reset commands seem to behave differently within this firmware and always factory reset.

In the end the idea is to have a reboot, a reset network information and start binding and a factory reset option. Especially a "leave and rejoin" reset would be nice to be able to move the node between ZB networks without loosing switch/relay configs.

Any help appreciated!

Regards

STefan

andrei-lazarov, 2025-08-06:
We previously discussed this on #98 and everybody agrees it will be useful.  
I also confirmed this is how it works on original firmware.

Thanks for being the one to try it implementing it!
Can you please share what you already tried and didn't work?

We can see on the repo you linked: they were able to achieve 'leave without reset' with this function: [tl_bdbReset2FN](https://github.com/pvvx/ZigbeeTLc/blob/master/src/app_ui.c).
I understand you already tried that and it still clears NV memory. I'm curious why..

clumsy-stefan, 2025-08-06:
>Can you please share what you already tried and didn't work?

Tried a number of things, but didn't want to bore anyone here ;) (or probably didn't want to expose my cluelessness :joy:)

Differnent reset methods (one at a time obviously):
- `tl_bdbReset2FN();`
- `tl_bdbReset();`
- `zb_reset();`
- `zb_resetDevice();`
- `nv_resetAll();`
- `nv_resetModule(NV_MODULE_ZCL);` then reboot
- `zb_deviceFactoryNewSet(1);` then reset (or set to 0)
- `zb_assocJoinReq();`
- `SYSTEM_RESET();`
- `bdb_networkSteerStart();`

Or actively leaving the network:
```
     nlme_leave_req_t req;
     ZB_IEEE_ADDR_COPY(req.deviceAddr, g_zbMacPib.extAddress);
     req.rejoin = 0;
     req.removeChildren = 0;
     if(!g_bdbCtx.leaveDoing){
       if(RET_OK == zb_nlmeLeaveReq(&req)){
          g_bdbCtx.leaveDoing = 1;
       }
     }
```

Most of these work and reboot the node, but also factory reset...

Especially in the network leave callback it only does a `SYSTEM_RESET();` which from my POV should only reboot the mcu... but also then it factory resets...

>I understand you already tried that and it still clears NV memory. I'm curious why..

Yep, me too!! 🤔

romasku, 2025-08-24:
I've looked into it and can share some progress:

* When we delete a device in the Z2M interface (based on what I see from the code disassembly), the SDK resets the NV storage. They probably do this to drop the network configuration, but it's quite aggressive. There's no clean way to avoid this, except by patching `drv_nv.c` so it doesn’t reset `NV_MODULE_ZCL` (or `NV_MODULE_APP`, if we move it) in `nv_resetToFactoryNew`. It’s hacky, but it will work.
* When we trigger a factory reset, we call `zb_factoryReset`, which internally also resets the factory storage. We could try to avoid calling it directly and instead manually reset only some NV modules, but because of the previous point, we’ll still need a hack, and it will handle this case as well.

romasku, 2025-08-26:
I've tried to patch `drv_nv.c` and it works for me, configuration is preserved properly. You can check it in the next release branch: https://github.com/romasku/tuya-zigbee-switch/tree/v1.0.21

clumsy-stefan, 2025-08-26:
>the only hardware action we need is to force re-binding; all other (re)configuration should be done from Z2M/ZHA. 

somewhere there has been a discussion, that not everybody uses a BMS but use the switches standalone, so different options would be nice...

>And rebooting is just pointless, tbh :)

completely agree, was more meant as an example...

I would see:
- Leave ZB Network (keep APP Config)
- Delete APP Config (keep ZB Config)
- Reset everything

which IMHO is not too complicated, either by multi-press or by different press-and-hold times...

>Hmm, pvvx firmware applies exactly the same patch (I didn’t know they did, but it suggests that it is a good option):

Just by having a good option, doesn't mean it's the best or even worse the correct one ;) 

if you change the factoryReset instead of the leave, you'll have to remeber that (and be backwards compatible) for the rest of the firmware life or others will wonder, why the factoryReset dosn't behave like it should...