romasku/tuya-zigbee-switch #122: TS0001/TS0002 freeze until power-cycled

andrazek, 2025-08-26:
I’m seeing repeated freezes on Girier Zigbee switch modules flashed with custom FW v1.0.17 (2-gang TS0002_basic and 1-gang TS0001). When it happens, the device ignores physical toggles and the hardware reset button; only cutting power revives it. On the 2-gang units I’ve seen both states: lights turning on by themselves and staying latched on, and lights staying off and refusing to turn on. This has occurred many times across multiple modules, and I haven’t identified a trigger.

andrazek, 2025-08-26:
The 2-gang module is TS0002_basic (_TZ3000_bvrlqyj7) – screenshots from Z2M (about & exposes) are attached below. For the 1-gang module I can’t recall the Zigbee Manufacturer; it now just shows as Tuya-TS0001-custom.

The issue occurs roughly once every 2-4 weeks. All loads are dumb bulbs (6–10W). I’ll upgrade at least one device to the latest firmware and monitor further. It might also be a hardware issue—hard to tell. So far, I can confirm freezes on 4 modules, while 2 others (one heavily used, one rarely used) haven’t shown problems yet.

[image]

[image]

ThatMishakov, 2025-08-28:
One of my relays also just froze on 1.0.20 and dev zigbee2mqtt, it was off, and I've noticed that it went offline in z2m, and it stopped reacting to the switch. It was Girier one gang relay (I forgot which was the model) and I just flashed all my relays yesterday. It is connected to a couple of LED bulbs in the cabinets, and is behind a switch in its box, I don't recall ever seeing it freeze on a stock firmware (I had it for at least a year, I don't remember when I bought them)

[image]

[image]

[image]

andrazek, 2025-08-29:
The device has frozen again (with the latest custom FW). It appears this happens roughly every few days for one of the modules (_TZ3000_bvrlqyj7).

romasku, 2025-08-29:
Hmm, sorry -- I still can’t reproduce the issue, nor can I think of a clear bug in FW why it might be happening. Since this is a router, perhaps some devices connected through it are sending slightly malformed packets. Could you check the Map tab in Z2M to see if there are devices that are connected through it?
Also, do you notice any patterns when it occurs -- does it happen randomly, sometimes in the middle of the night, sometimes during the day? And do you use this switch often?

andrei-lazarov, 2025-08-29:
@andrazek @ThatMishakov Can you guys also specify some info about your Zigbee Coordinators (model, driver) ?  

Please also try end_device firmware (without routing) and see if the problem persists.  
I built it on my branch [force_ed_all](https://github.com/andrei-lazarov/tuya-zigbee-switch/tree/force_ed_all) as we don't have it on the main branch for devices with neutral.  
Use this index and perform an update:
```
https://raw.githubusercontent.com/andrei-lazarov/tuya-zigbee-switch/refs/heads/force_ed_all/zigbee2mqtt/ota/index_end_device-FORCE.json
```

ThatMishakov, 2025-08-29:
I haven't seen any devices crash since yesterday, but it happened in the morning, after I didn't use it during the night and when there shouldn't have been a lot of smart home activity. I don't use that particular switch very often, and out of two relays that were next to each other, only one has stopped working.

I guess I have a large network (more than 150 devices) with stuff from different vendors, and it was close to the coordinator (which was kinda in the corner of the house), so it was possible that something did route through it (on a separate note, I don't know if it's placebo, but this firmware did seem to make my network work much better compared to stock), though I've also just moved my coordinator to a different room, so I'll be on a lookout for it (I have about 15 modules scattered around the house, of several different models, where I've installed the custom firmware).

My coordinator is ZBDongle-E (with EFR32MG21) on 8.0.2 firmware, and I'm on dev branch of zigbee2mqtt

andrei-lazarov, 2025-08-29:
The coordinator is exactly what I'd recommend. 150 devices kinda feels like pushing the limits though, so it might be related.. @andrazek do you also have a lot of devices in your network?

andrazek, 2025-08-30:
I’m using a Zigbee Dongle-P with a total of 74 devices (25 end devices). I haven’t noticed any clear correlation with time or frequency of use. In my case, the issue occurs with the entrance light connected to a motion sensor (IKEA VALLHORN). I didn’t bind it directly, but the traffic may often be routed through this module. This one freezes more often than the others. It is also used more frequently.

andrei-lazarov, 2025-08-30:
Hmm, I remember reading about IKEA motion and contact sensors being problematic. Please research a bit, what other users are experiencing with the IKEA sensors. The remotes are okay I think.

ThatMishakov, 2025-08-30:
I remember that they had issues with requiring custom quirks with ZHA, but I don't know about other issues, and I have been using them for some time now with no major problems (and they're nice because of water resistance).

On a separate note, I've also tried to join VALLHORN sensor through the switch module, and it also refused to join through it (it joins the network and leaves it after a few seconds, exiting the pairing mode; whereas the TRADFRI sensors just didn't show up in Z2M in the first place), but they do join fine through CC2652P router and Aqara T2 light (and Ikea lights).

ThatMishakov, 2025-09-05:
I haven't tried the router firmware yet, but another one of these modules (bvrlqyj7) has just crashed on me, at about 3 AM. It was power cycled a few hours before that, and the out of ordinary thing that I did before it crashed was repairing a TuYa window contact sensor to a different device from my Ledvance light (which has started to turn on randomly and annoy me at night...) about 1 hour before Z2M stopped being able to ping it (I have my interval set to 30 minutes).

But I've just noticed that I have another TuYa device, which I think might have started to spam the network some time after this, and I'm getting interesting errors like this:
```
zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".
```

Also, I've updated my coordinator to 8.2.1 and the pairing issues went away, so it was probably a false alarm.

andrazek, 2025-09-21:
Just a quick update: flashing it as an end device didn’t improve things much. The module dropped out of the Zigbee network, while the other one (router FW) froze again. It seems that the bvrlqyj7 modules are the most problematic — could it be that their specifications are worse?

zavaruev, 2025-09-21:
Any chances that Stefan build it for "iHseno touch switches"? Cannot find it in his index(

andrei-lazarov, 2025-09-21:
@zavaruev I thought your problem was related to the touchpads, but yea.. maybe it is the routing. How many devices do you have?

So thanks for spotting this. It's a good idea to try the new sdk.  
I built it for the new devices on my [update_sdk_new_build](https://github.com/andrei-lazarov/tuya-zigbee-switch/tree/refs/heads/update_sdk_new_build) branch:
```
https://raw.githubusercontent.com/andrei-lazarov/tuya-zigbee-switch/refs/heads/update_sdk_new_build/zigbee2mqtt/ota/index_router.json
```
I will delete it once Stefan's branch is up to date.

zavaruev, 2025-09-21:
Flashed, testing. Thank you, Andy!

andrei-lazarov, 2025-09-24:
The update_sdk branch is on v1.0.21, which changes their behavior when removed from the network.  
Now they keep configuration instead of deleting it.  

Not sure how to help. Do your devices have Neutral? Did you install router fw?

andrazek, 2025-10-19:
Yes, with the new SDK I did not experience any freezes. Not sure if the official version is build with the new SDK though (I'm still on update_sdk branch).