romasku/tuya-zigbee-switch #128: Support for _TZ3000_tqwydnqn

rastakozza, 2025-08-28:
Hi,

I got a 3 gang wall switch zigbee model _TZ3000_tqwydnqn.

[image]

[image]

[image]

I cloned _TZ3000_qewo8dlz entry in index_router-FORCE.json:

```
  {
    "fileName": "1141-aa0d-ffffffff-tlc_switch-forced.zigbee",
    "fileVersion": 4294967295,
    "fileSize": 166418,
    "url": "https://github.com/romasku/tuya-zigbee-switch/raw/3fa33d1fb514f7f6a4ccf609c425198992882f0e/bin/MOES_3_GANG_SWITCH/1141-aa0d-ffffffff-tlc_switch-forced.zigbee",
    "imageType": 43533,
    "manufacturerCode": 4417,
    "sha512": "eb9199afe7cdc3bd6fde8f803c4ae7978680bba1875dd492ab5a8dde842851f3a404cc03f25abd079212d308e7addd528b249ce513674f1fe4028faa8a42af7e",
    "otaHeaderString": "Telink Zigbee OTA\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
    "manufacturerName": [
      "_TZ3000_tqwydnqn",
      "tqwydnqn"
    ]
  }
```

I see the device in OTA tab, but no update are available.
As specified in the porting guide, here is the log line:

```
[2025-08-28 15:45:29] debug: 	zh:controller: Received payload: clusterID=25, address=19774, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=255, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":250,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4417,"imageType":54179,"fileVersion":80},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35},{"name":"hardwareVersion","type":33,"conditions":[{"type":"bitMaskSet","param":"fieldControl","mask":1}]}],"name":"queryNextImageRequest"}}
```

Thank you!

rastakozza, 2025-08-28:
I'm sorry, I think you also need pull up/down config.
Buttons are pulled up (active low)
Leds are connected to 3.3V (active low)
Relays commands are pulled down (active high)

rastakozza, 2025-08-28:
I flashed device but it doesn't work fine.
Just left switch works. When i press center or right switch, led blink one time and then device stop respond for a few seconds.
Changing state of center and right switch from zigbee2mqtt I receive error messages:
```
error 2025-08-28 23:47:43z2m: Publish 'set' 'state' to '0xa4c1380de600a783' failed: 'Error: ZCL command 0xa4c1380de600a783/5 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '10817'.)'
error 2025-08-28 23:47:43z2m: Failed to read state of '0xa4c1380de600a783' after reconnect (ZCL command 0xa4c1380de600a783/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
```

I don't see device in OTA tab, probably because it changed its name:

[image]

andrei-lazarov, 2025-08-28:
Firstly, get the [newly generated converters](https://github.com/andrei-lazarov/tuya-zigbee-switch/tree/refs/heads/manhot/zigbee2mqtt/converters) and restart Z2M. Then Reconfigure and test again.

rastakozza, 2025-08-28:
Ok, it looks better.
Still having problems with middle and right switches:
```
error 2025-08-29 00:13:30z2m: Publish 'set' 'state' to '0xa4c1380de600a783' failed: 'Error: ZCL command 0xa4c1380de600a783/5 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '10817'.)'
error 2025-08-29 00:13:30z2m: Failed to read state of '0xa4c1380de600a783' after reconnect (ZCL command 0xa4c1380de600a783/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))
```

andrei-lazarov, 2025-08-28:
The log messages show a reboot happened.

First line is pretty generic. You get it when the device is frozen or offline.
Second line is an expected error that shows up every boot.

The first switch working is a good sign. Do the first relay and LED toggle when you press it?

I suggest you join the Discord server and we continue troubleshooting there. We should double check the pinout for the non working switches (with the switch disconnected from power!)