romasku/tuya-zigbee-switch #130: Support for _TZ3000_imaccztn

bigroma73, 2025-08-29:
Hi
I've got 4 channels Zigbee relay 
_TZ3000_imaccztn

Does it possible to support this relay? 

WBR
Oleksandr

[image]

[image]

andrei-lazarov, 2025-08-29:
I have not seen this form factor before, but it should be possible.  
I assume it works like a normal switch. And the ZT3L is supported.  

So you have to trace the pinout.  
Look up the [ZT3L diagram](https://developer.tuya.com/en/docs/iot/zt3l-module-datasheet?id=Ka438n1j8nuvu) and let me know what's the pin for: every button, the reset button, every relay, state indicator LEDs and network LED. How many LEDs does it have?  
Note that the diagram shows the back of the module. And you are viewing it from the front.

Also, I'm curious what you are using this for and what features from the custom firmware you would need.

bigroma73, 2025-08-29:
I've found buttons pins on module but now I still not found relays pins. They're connected not direct. 

I'm using switches like this for different applications: gate, watering, water tanks. This one and one like this from Avatto(same issues) I'm using for 24V AC watering in my garden.

bigroma73, 2025-08-29:
There's additional chip between ZT3L and transistors. Buttons K# connect directly K1:D7, K2:C0, K3:B5, K4:RX, but mode button, blue led and relays connect to 20 pin chip marked SMG(or SMC)-C45.
I can't googled this chip :( 
[image]

bigroma73, 2025-08-30:
Hi
Looks like
C3:RedLED
D2:relay3
C2:relay4
D4:relay1
A0:relay2

I can't find blue led and reset/mode button :(

andrei-lazarov, 2025-08-30:
I assume the red LED is the network indicator and there are also 4 blue LEDs indicating the relay states?  

In this case, you can not find the LEDs because they use the same pins as the relays. They are hardwired together, the ZT3L does not have enough GPIO pins to individually control an additional 4 LEDs.

The remaining pins for the reset button are C4, B4 and B1.  
But **we can build the firmware without it**. You can reset the device by quickly pressing 10 times on any button.

I added the device to the database, but I want you to **confirm you actually need the custom firmware**.  

Our firmware does not have power on/off timers implemented yet. (I see your device has **On with timed off**, a feature we are currently missing)
Also I see your device has RF control? I don't know if this will work with the custom fw.

So what issues are you having with the factory firmware.

bigroma73, 2025-08-30:
Issue was unpredictable switch behavior when fast switching. I don't need RF and time off for this switch now. I think reset works through SMG-C45 and SWS but I'm not sure.

andrei-lazarov, 2025-08-30:
Then I think we can continue with the configuration I already added (using the red LED as a network indicator instead of the blue LED).

If you want to try it like this, I built the fw on my branch [_TZ3000_imaccztn](https://github.com/andrei-lazarov/tuya-zigbee-switch/tree/refs/heads/_TZ3000_imaccztn).

Get the quirks / converters from that branch and use the corresponding index:

```
https://raw.githubusercontent.com/andrei-lazarov/tuya-zigbee-switch/refs/heads/_TZ3000_imaccztn/zigbee2mqtt/ota/index_router.json
```

andrei-lazarov, 2025-10-06:
Device added to main:
```yaml
MHCOZY_TS0004:
  human_name: MHCOZY TYWB 4ch-RF
  neutral: required
  device_type: router
  tuya_model_name: TS0004
  tuya_manufacturer_name: _TZ3000_imaccztn
  stock_converter_manufacturer: Tuya
  stock_converter_model: TS0004
  override_z2m_device: TYWB 4ch-RF
  zb_module: ZT3L
  config_str: imaccztn;TS0004-MC;LC3;SD7u;RD4;SC0u;RA0;SB5u;RD2;SB7u;RC2;M;
  alt_config_str: null
  old_manufacturer_names: null
  old_zb_models: null
  tuya_manufacturer_id: 4417
  tuya_image_type: 54179
  firmware_image_type: 43551
  build: yes
  status: In progress
  info: Untested? Unknown pins for reset button and blue LED? (C4, B4, B1)
  github_issue: https://github.com/romasku/tuya-zigbee-switch/issues/130
  store: https://www.aliexpress.com/item/1005005457296054.html
```

closed 2025-11-15 by romasku

Ctrl-F4, 2025-11-23:
Firmware works great (also for version without RF).
Network led must be inversed btw.
`imaccztn;TS0004-MC;LC3i;SD7u;RD4;SC0u;RA0;SB5u;RD2;SB7u;RC2;M;`

One question: at the moment all buttons can set the device in pairing mode.
Is it possible to assign **only one** of the four buttons for this task?

andrei-lazarov, 2025-11-23:
Hi! Currently no. I'll take this into consideration for upcoming releases. 

Please help me clarify: 
- Do you have a different device (no RF variant) that shares the Zigbee part?
  ⮡ Give me the full name and store link, so I can add it to the database

- Is the network LED, currently controlled by the firmware, blue or red?

- Can you find the button labeled RESET / Mode Switch?
  ⮡ We can currently use it in the custom fw for pairing, with more possibilities in the future.
  - If you are on version v1.0.21 (current main), it's safe to test.
  ⮡ The available pins are B1, B4, C4. Please test them in the config string like this:
  Apply `imaccztn;TS0004-MC;BB1u;LC3i;SD7u;RD4;SC0u;RA0;SB5u;RD2;SB7u;RC2;M;` and hold 3s to reset.
  Repeat for the other pins.

Thank you.