romasku/tuya-zigbee-switch #135: Support fot Avatto LZWSM16-W3

soosp, 2025-08-30:
Hi,

I just bought an Avatto LZWSM16-W3 no neutral switch. It introduce itself as _TZ3000_avotanj3 TS0013. It based on the ZTU module. After disassembe and same reverse engineering I recognised the following pinout:

Button: B4
LED: B5
S1: A1
S2: C3
S3: C0
R1: C2
R2: C4
R3: D2

The voltage on the unconnected switch inputs is 3.3V, so I assumed that they are pull-up ones, but measuring the resistance to the Vcc and GND it found 100kOhm to both directions. The onboard button pin has ~1MOhm resistance to the both directions.

The voltage on LED output pin is 0V if the LED is off and 3.3V if it is on.

So a the configuration string should be similar to this:
`"avotanj3;TS0013-avatto;BB4;LB5;SA1u;SC3u;SC0u;RC2;RD2;RB5;"`

soosp, 2025-08-30:
Here is some photos about the device:

[image]

[image]

andrei-lazarov, 2025-08-30:
I think you wrote the switches in reverse order.  
It should be like this to match the pattern:
```
hbxsdd6k;TS0011-AVB;BB4u;LB5;SC0u;RC2;
ljhbw1c9;TS0012-AVB;BB4f;LB5;SC0f;RC2;SC3f;RC4;
avotanj3;TS0013-AVB;BB4u;LB5;SC0u;RC2;SC3u;RC4;SA1u;RD2;
```

I will create the PR and wait for romasku to double check the pull-up/down resistors. I see both u and f are used by the other variants.

soosp, 2025-08-30:
A will re-check the switch order soon.
The f (float) setting may cause problem, as per some of my TS0012-AVB switches sometimes randomly turns on

soosp, 2025-09-02:
The switch inputs has been tested again with the original Tuya firmware:
The wire on top side of strip cable, connected to A1operates the L1 output (C2).
The middle one (C3) operates the L2 output (C4).
The wire nearest to the terminal block (C0) operates the L3 output (D2).

andrei-lazarov, 2025-09-02:
I'm glad you got it working! I will update the PR.
If the device is stable, you can close this issue.

soosp, 2025-09-02:
For now I tested the OTA forced update with success. So it seems that everything is working fine.

Regarding to the pull-up configuration, I updated all of my Avatto no neutral devices to u (10k pull-up) configuration, and since that the random turn on events gone.