romasku/tuya-zigbee-switch #145: Add support BSEED TS011F in‑wall socket (FK86ZEUSK1W, _TZ3000_b28wrpvx)

analienx, 2025-09-10:
### Context
This request is to add support for the **BSEED in‑wall FR socket model FK86ZEUSK1W** identified on Zigbee as **TS011F** with manufacturer **_TZ3000_b28wrpvx**, using Tuya’s **ZTU module**.  
The same fingerprint/device is discussed in upstream Zigbee2MQTT issue threads, which helps with identification and labeling:  
- https://github.com/Koenkk/zigbee2mqtt/issues/28256  
- https://github.com/Koenkk/zigbee2mqtt/issues/6190  

**Priority:** Reliable routing and binding (energy monitoring can be integrated afterwards).  

---

### Hardware / Fingerprint
- **Product:** BSEED wall‑mounted FR/EU socket FK86ZEUSK1W (variants with/without energy monitoring).  
- **Zigbee:** `modelId: TS011F`, `manufacturer: _TZ3000_b28wrpvx` (matches prior BSEED wall socket reports).  
- **Radio module:** Tuya **ZTU**, PCB label `Model: ZTU, P/N: 2.01.03.00300`.  

References:  
- ZTU datasheet: https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp  
- FCC filing: https://fcc.report/FCC-ID/2ANDL-ZTU/5037964.pdf  

---

### References (Product Pages)
- AliExpress listing (Zigbee FR socket, BSEED): https://www.aliexpress.com/item/1005004402438527.html  
- AliExpress listing (FR socket selection, BSEED): https://www.aliexpress.com/i/1005005835434423.html  
- AliExpress listing (power‑monitoring variant): https://www.aliexpress.com/item/1005003350696939.html  

---

### Behavior and Ask
- On stock firmware, these sockets perform poorly as routers:
  - Rarely accept end‑device children  
  - Bindings unreliable  
- This behavior is consistent with other TS01xx / TS011F devices.  
- A **custom firmware image** that:
  - Ensures stable routing and binding, or  
  - Provides a fallback *end‑device join mode*  
  would resolve the issue.  

**Initial support:** Reliable on/off switching and binding.  
**Future extension:** Energy monitoring (if hardware SKU supports it), using the Tuya datapoint mapping process:  
- https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html  
- https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html  

---

### What Can Be Provided for Development
- High‑resolution PCB photos (with ZTU label) and external part markings:
- https://smarthomescene.com/reviews/bseed-16a-zigbee-energy-monitoring-outlet-review/?utm_source=perplexity

[image]
  
- Pairing interview data and logs for fingerprint/cluster validation  
- Willingness to test experimental images across multiple installed units and provide feedback

andrei-lazarov, 2025-09-10:
Hello, thanks for your interest in the project.   

Are all your units installed in the wall?
Supporting new devices consists of tearing down the device and tracing the board.  
Would you be comfortable with that?  

We need to know the pinout (map ZTU pins to relay, button, LED) to build the firmware.

andrei-lazarov, 2025-09-17:
@analienx was able to tear-down 2 variants: with and without PM. Thankfully the pins were labeled so porting was easy.  
Updating was successful.

The 2 LEDs (blue and red) are in the same spot, so the user can swap them around in the config string.  
Note that Power Monitoring is not implemented yet.  
Also the converters might need updating to remove unnecessary options for sockets.

## Non-PM
- `o1jzcxou;TS011F-BS;LC2;SB4u;RC3;ID2;M;`
[image]
[image]
[image]
[image]

## PM
- `b28wrpvx;TS011F-BS-PM;LC3;SB5u;RD2;IB4;M;`

[image]
[image]

pawelwo, 2026-04-08:
Hi, what about energy monitoring? Is this feature planned? When will it be released?

andrei-lazarov, 2026-04-08:
- it's in progress #314